SpikeInterface/spikeinterface #3239: error loading sorting_analyzer to phy : channel position is 3D but must be 2D

borrepp, 2024-07-20:
Hello,

I have exported sorting_analyzer (with all extensions) into phy ("export_to_phy"). But when loading it into phy, there is an error about the dimensions of the channel_locations. The original recording came from a NWB file, which has 3D locations. These 3D locations are carried through all the preprocessing and sorting. 

The way I "fixed" was to force channel_positions into a 2D vector:

`if channel_positions.shape[1]>2:
        channel_positions = channel_positions[:, 0:2] #Assume first to locations are [X, Y] in the probe`

I'm not sure if it is the best practice, but I can load the sorting_analyzer on phy. 

Thanks

zm711, 2024-07-22:
That's a problem with Phy not with NWB/SpikeInterface. Phy is only designed for 2D probes. I used to have to flatten my probes all the time for phy. Maybe we could do something in our export function to help with that, but there isn't a straight forward way to do this because if you're truly 3d and flattened then shanks would be overlapping and wouldn't display properly. so we would have to do a lateral translation, but then it could require variable translations depending on the specific probe. So in this case it would likely be more understandable for the end-user to decide how they want to flatten their probe so they know what they did rather than have it be magical behind the scenes. What I personally do is I split my shanks into probegroups and then translate them myself because spikeinterface lets us sort probes/shanks individually the positions of the shanks (as long as they are far enough apart) don't matter, what matters are the positions of the contacts on a shank.

Does that make sense? Or do you need some more info?

borrepp, 2024-07-22:
Hello @zm711 
Thanks for your response. Yes, it makes sense. I will flatten the probe on my end. And I think that will solve another problem that I encountered with the sorting_analyzer extension "unit_locations":

```
spikeinterface\src\spikeinterface\postprocessing\localization_tools.py:336, in make_initial_guess_and_bounds(wf_data, local_contact_locations, max_distance_um, initial_z)
    334 com = np.sum(wf_data[:, np.newaxis] * local_contact_locations, axis=0) / np.sum(wf_data)
    335 x0 = np.zeros(4, dtype="float32")
--> 336 x0[:2] = com
    337 x0[2] = initial_z
    338 initial_alpha = np.sqrt(np.sum((com - local_contact_locations[ind_max, :]) ** 2) + initial_z**2) * max_ptp

ValueError: could not broadcast input array from shape (3,) into shape (2,)
```

However, this localization shouldn't have a problem even with a 3D probe, isn't it? I'm using monopolar triangulation, which runs fine when I use peaks_localization to estimate motion correction.

I'm sure after I flatten the probe I will not get that problem, but let me know if I should raise an issue for unit_locations to follow up.

Thanks again for your help and advice. 
Best 
Pepe

zm711, 2024-07-23:
It shouldn't. which version of spikeinterace are you using? Could you post the snippet of your script where you set up your probe and then run unit localizations?

borrepp, 2024-07-23:
I didn't set up a probe, but the recording object has the locations imported from the original NWB (X, Y, Z).

I'm using version "0.101.0". 

When I run unit localizations, it displays a message that it didn't find a Probe and it will create a dummy one. 

Here is the snippet of my code
```

sorting_analyzer = si.create_sorting_analyzer(
        sorting=sorting, 
        recording=si_recording, 
        format='binary_folder', 
        folder=folder_analyzer, 
        overwrite=True,
        sparse=True, 
        **estimate_sparsity_params,
        **job_kwargs
        )

estimate_sparsity_params = {
    'num_spikes_for_sparsity': 100, # How many spikes per units to compute the sparsity (default: int = 100) 
    'ms_before': 0.6, # Cut out in ms before spike time (default: float = 1.0)
    'ms_after':  1.5, # Cut out in ms after spike time (default: float'= 2.5)
    'method': 'radius', # (radius' | 'best_channels # (default: str = 'radius')
    'peak_sign': 'both', # Sign of the template to compute best channels (“neg” | “pos” | “both”, default: str = 'neg') 
    'radius_um': step_chan*3.0, # (default: float = 100.0)
    'num_channels': 3, # Used for “best_channels” method (default: int = 5)
}

sorting_analyzer_compute_opts = {
        'random_spikes': {},
        'noise_levels': {},
        'correlograms': {},
        'isi_histograms': {},
        'waveforms': {},
        'principal_components': {'n_components': 5, 'mode': 'by_channel_local'},
        'templates': {'operators': ["average"]},
        'template_metrics': {},
        'template_similarity': {},
        'amplitude_scalings': {},
        'spike_amplitudes': {},
        'spike_locations': {},
        'unit_locations': {'method':"monopolar_triangulation", 'radius_um': step_chan*3.0}
    }
    sorting_analyzer.compute(sorting_analyzer_compute_params, save=True, **job_kwargs)
```

zm711, 2024-07-23:
For this type of thing you need the probe. My guess is that it is making a dummy probe which is expecting the 2D but your data is 3D. You could set up a probegroup or make a 3d probe. I don't remember the probeinterface reading for nwb to create a probe for you. (Basically the probe/probegroup is necessary) so we need to make a probe out of your nwb file and attach it to the recording. @h-mayorquin you would be the best to comment on creating a probe from an nwb file. Maybe @samuelgarcia would also be able to help with that part.

borrepp, 2024-07-23:
If I remove the Z coordinates from the "location" property immediately after loading the NWB file, the "unit_localization" runs fine. Without the need to setup a probe.

zm711, 2024-07-24:
yeah, what I'm saying is that our code for people not using probes attaches a dummy probe that is likely expecting 2d data. Best case scenario (for accuracy) would be for you to take your real probe information from the nwb file and make a probe so that the function uses your real probe rather than making a dummy probe. I don't know the geometry of our dummy probe so it may or may not approximate your probe well. 

so you removing the z dimension allows the dummy probe to be attached but might not be the most accurate. 

Does that make sense?

borrepp, 2024-07-24:
Yes, it makes sense. I tried some time ago to set up the probe but I saw that some indexes were rearranged, but at that point, I didn't fully understand the recording object. I will try it again and let you know if I encounter any problems with the unit localization.

By the way, one quick question about the probes. Separately, I have constructed a 3D probe based on my NWB file. When setting up the probe in the spikeinterface recording object, should I set up the probe as 2D [probe..to_2d()], or it is fine to keep it as 3D?

Thanks again for all your help.

zm711, 2024-07-24:
Spikeinterface in general can handle 3d probes the problem is that spike sorters can't. So if you're using spikeinterface and not doing any spike sorting (for example using motion detection or something like that). But if you're using spikeinterface to spike sort then you need to find a way to make your probe 2d or you make a probe group where you each probe of the group is 2d, but put together they are 3d (I do this all the time since I use a 3d probe :) )

borrepp, 2024-07-24:
I'm using spikeinterface to run sorters too. I will do it in 2D, which is not too difficult to do it on my end when handling the NWB recording. 

I will let you know if there is any problem. It may take some time (days) for me to update my code while I'm running some experiments. I don't know if you want to close this issue and If there is any error later I can open another one specifically for unit_localization

Thanks again¡¡

zm711, 2024-07-24:
yep. We can close this for now, but if you run into the unit_localization issue after attaching a probe/probegroup with the correct dimensionality definitely open a new issue :)

closed 2024-07-24 by zm711