SpikeInterface/spikeinterface #3240: About metric

bxy666666, 2024-07-21:
Traceback (most recent call last):
  File "G:\bxy\spikeinterface-main\examples\get_started\quickstart.py", line 102, in <module>
    metric = spost.compute_template_metrics(we_KS2, include_multi_channel_metrics=True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python312\Lib\site-packages\spikeinterface\postprocessing\template_metrics.py", line 320, in compute_template_metrics
    tmc.run()
  File "C:\Python312\Lib\site-packages\spikeinterface\core\waveform_extractor.py", line 2025, in run
    self._run(**kwargs)
  File "C:\Python312\Lib\site-packages\spikeinterface\postprocessing\template_metrics.py", line 172, in _run
    value = func(
            ^^^^^
  File "C:\Python312\Lib\site-packages\spikeinterface\postprocessing\template_metrics.py", line 935, in get_spread
    MM = gaussian_filter1d(MM, spread_sigma)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python312\Lib\site-packages\scipy\ndimage\_filters.py", line 276, in gaussian_filter1d
    lw = int(truncate * sd + 0.5)
         ^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: cannot convert float NaN to integer

JoeZiminski, 2024-07-22:
Hi @bxy666666, thanks for the bug report. Did you make any changes to the quickstart.py script? What version of spikeinterface are you running?

MilagrosMarin, 2024-07-23:
Hi @JoeZiminski,

I've encountered a similar error today while running the latest version of SpikeInterface `0.101.0` through a DataJoint pipeline. The error stack is copied below. Any help you can provide would be greatly appreciated. Thanks!

`Traceback (most recent call last):
  File "/opt/conda/lib/python3.10/site-packages/datajoint/autopopulate.py", line 315, in _populate1
    make(dict(key), **(make_kwargs or {}))
  File "/opt/conda/lib/python3.10/site-packages/element_array_ephys/spike_sorting/si_spike_sorting.py", line 364, in make
    _sorting_analyzer_compute()
  File "/opt/conda/lib/python3.10/site-packages/element_interface/utils.py", line 235, in wrapped
    results = func(*args, **kwargs)
  File "/opt/conda/lib/python3.10/site-packages/element_array_ephys/spike_sorting/si_spike_sorting.py", line 346, in _sorting_analyzer_compute
    sorting_analyzer.compute(extensions_to_compute, **job_kwargs)
  File "/opt/conda/lib/python3.10/site-packages/spikeinterface/core/sortinganalyzer.py", line 890, in compute
    self.compute_several_extensions(extensions=input, save=save, verbose=verbose, **job_kwargs)
  File "/opt/conda/lib/python3.10/site-packages/spikeinterface/core/sortinganalyzer.py", line 1029, in compute_several_extensions
    self.compute_one_extension(extension_name, save=save, verbose=verbose, **extension_params)
  File "/opt/conda/lib/python3.10/site-packages/spikeinterface/core/sortinganalyzer.py", line 967, in compute_one_extension
    extension_instance.run(save=save, verbose=verbose)
  File "/opt/conda/lib/python3.10/site-packages/spikeinterface/core/sortinganalyzer.py", line 1640, in run
    self._run(**kwargs)
  File "/opt/conda/lib/python3.10/site-packages/spikeinterface/postprocessing/template_metrics.py", line 251, in _run
    value = func(
  File "/opt/conda/lib/python3.10/site-packages/spikeinterface/postprocessing/template_metrics.py", line 891, in get_spread
    MM = gaussian_filter1d(MM, spread_sigma)
  File "/opt/conda/lib/python3.10/site-packages/scipy/ndimage/_filters.py", line 276, in gaussian_filter1d
    lw = int(truncate * sd + 0.5)
ValueError: cannot convert float NaN to integer`

MilagrosMarin, 2024-07-30:
@zm711 , of course, and thank you for your reply and time. Here are some more details, please let me know if you need any other info:

For PreProcessing: 

```    
recording = si.preprocessing.bandpass_filter(
        recording=recording, freq_min=300, freq_max=4500
    )
    recording = si.preprocessing.common_reference(
        recording=recording, operator="median"
```

For Processing and PostProcessing:

`include_multi_channel_metrics=True` seems to be the cause for the mentioned error `ValueError: cannot convert float NaN to integer`. Here is the specific paramset used for the processing and postprocessing, which corresponds to the default parameter set for spyking circus2 using SpikeInterface v0.101:

```
{'SI_SORTING_PARAMS': {'general': {'ms_before': 2,
    'ms_after': 2,
    'radius_um': 100},
   'filtering': {'freq_min': 150},
   'detection': {'peak_sign': 'neg', 'detect_threshold': 4},
   'selection': {'method': 'smart_sampling_amplitudes',
    'n_peaks_per_channel': 5000,
    'min_n_peaks': 20000,
    'select_per_channel': False},
   'clustering': {'legacy': False},
   'matching': {'method': 'circus-omp-svd', 'method_kwargs': {}},
   'apply_preprocessing': True,
   'cache_preprocessing': {'mode': 'memory',
    'memory_limit': 0.5,
    'delete_cache': True},
   'multi_units_only': False,
   'job_kwargs': {'n_jobs': 0.8},
   'debug': False},
  'SI_POSTPROCESSING_PARAMS': {'extensions': {'random_spikes': {},
    'waveforms': {},
    'templates': {},
    'noise_levels': {},
    'correlograms': {},
    'isi_histograms': {},
    'principal_components': {'n_components': 5, 'mode': 'by_channel_local'},
    'spike_amplitudes': {},
    'spike_locations': {},
    'template_metrics': {'include_multi_channel_metrics': True},
    'template_similarity': {},
    'unit_locations': {},
    'quality_metrics': {}},
   'job_kwargs': {'n_jobs': -1, 'chunk_duration': '1s'},
   'export_to_phy': True,
   'export_report': True}}}
```
I've debugged further into the error and here are the outputs for your reference: 

This is the code snippet computing the template metrics without including the multichannel metrics (setting it to `False`), and a successful run:
```
`In [11]: extensions_to_compute
Out[11]:
{'random_spikes': {},
 'waveforms': {},
 'templates': {},
 'noise_levels': {},
 'correlograms': {},
 'isi_histograms': {},
 'principal_components': {'n_components': 5, 'mode': 'by_channel_local'},
 'spike_amplitudes': {},
 'spike_locations': {},
 'template_metrics': {'include_multi_channel_metrics': False},
 'template_similarity': {},
 'unit_locations': {},
[... 84 lines cut ...]
--> 251         value = func(
    252             template_upsampled,
    253             channel_locations=channel_locations_sparse,
    254             sampling_frequency=sampling_frequency_up,
    255             **self.params["metrics_kwargs"],
    256         )
    257         template_metrics.at[index, metric_name] = value
    258 self.data["metrics"] = template_metrics

File /opt/conda/lib/python3.10/site-packages/spikeinterface/postprocessing/template_metrics.py:891, in get_spread(template, channel_locations, sampling_frequency, **kwargs)
    888     from scipy.ndimage import gaussian_filter1d
    890     spread_sigma = spread_smooth_um / np.median(np.diff(np.unique(channel_depths)))
--> 891     MM = gaussian_filter1d(MM, spread_sigma)
    893 MM = MM / np.max(MM)
    895 channel_locations_above_threshold = channel_locations[MM > spread_threshold]

File /opt/conda/lib/python3.10/site-packages/scipy/ndimage/_filters.py:276, in gaussian_filter1d(input, sigma, axis, order, output, mode, cval, truncate, radius)
    274 sd = float(sigma)
    275 # make the radius of the filter equal to truncate standard deviations
--> 276 lw = int(truncate * sd + 0.5)
    277 if radius is not None:
    278     lw = radius

ValueError: cannot convert float NaN to integer`
```

zm711, 2024-08-07:
Let me actually tag @alejoe91, since he knows this better than I do. I think he may have tried to work on this to protect for nans. So this may be fixed on main. It might be worth installing from main/source and see if it is fixed @MilagrosMarin.

zm711, 2024-12-03:
Were you able to test main @MilagrosMarin ?

MilagrosMarin, 2024-12-04:
Yes, I was installing directly from the main source code. The error no longer appeared, so I will proceed to close this issue. I will reach out if I encounter the same issue again. Thank you so much, @zm711!

closed 2024-12-04 by zm711