SpikeInterface/spikeinterface #3241: motion_interpolation recording is not preserved after whitening

borrepp, 2024-07-22:
Hello,

I'm getting a new recording object from "interpolate_motion" which I can explore with widgets.plot_traces and the motion correction is reflected on the traces. However, when I apply whitening (lazy), the resulting recording object is no longer motion-corrected, is the original recording whitened. The original recording came from a binary folder (created with spikeInterface).

The way I made it work was by saving the interpolate_motion-recording (binary folder), and after loading it, I applied the whitening (lazy) and the motion correction was there. 

Just wanted to know if I did something wrong or whether I should always save the motion-corrected recording in case of applying whitening. 

I haven't tried to run any sorter yet, but in case I didn't apply any whitening, I assume that it is ok to run any sorter on the recording object that came directly from "interpolate_motion". Am I right?

Thanks in advance for any comments about the best practices to run a sorter after motion correction.

Best
Pepe

JoeZiminski, 2024-07-22:
Hi @borrepp that's interesting 🤔, might you be able to post the code you are using to preprocess / inspect the data? 

It depends a little on the sorter you use, in genereal is a sorter requires the whitening step it will implement it itself under the hood (e.g. kilosort, mountainsort). What sorter are you sing?

borrepp, 2024-07-22:
Sure. 
Here is the code, it is not exactly the one I'm using because the original one is nested in other variables to keep track of directories when doing multiple motion correction methods, but the steps are like this:

NOTES: During the preprocessing I created a binary file with the filtered CMR etc, recording.  From the Binary recording, I created a whitened recording and dumped to pickle

```
# MOTION CORRECTION :
# Load the preprocessed recording from a Binary file (saved with spikeinterface)
rec_prepro = si.load_extractor(rec_prepro_path) 

# Do "motion_estimation" on rec_prepro with certain parameters
......

# Use the resulting "motion"-object to interpolate
rec_motion = interpolate_motion(rec_prepro, motion.....)

# Save the motion-corrected as lazy object 
rec_motion.dump_to_pickle(rec_motion_path) 

# After inspecting different motion correction parameters, load the "best" interpolation recording to inspect it vs. the original recording
recMotion1 = si.load_extractor(rec_motion_path) # Load the recording from pickle

%matplotlib ipympl
sw.plot_traces({
    'Original' : rec_prepro,
    'Motion_corrected' : recMotion1,
    }, 
    segment_index=0,
    time_range=[0, 10], 
    backend="ipywidgets"
    )
```

Up to here everything looks fine, the motion-corrected recording shows the motion correction in comparison with the original
Then I apply the whitening

```
# Apply whitening
step_chan = distance_between_channels
recMotion_whiten = spre.whiten(recording=recMotion1, mode='local', radius_um = step_chan*4)

# Load original whitened
rec_prepro_whitened = si.load_extractor(rec_prepro_whitened_path)  # pickle file
rec_prepro_whitened_match = rec_prepro_whitened.channel_slice(channel_ids=[ch_i for ch_i in recMotion_whiten.channel_ids])

%matplotlib ipympl
sw.plot_traces({
    'Whiten' : rec_prepro_whitened_match,
    'Whiten_motion': recMotion_whiten
    }, 
    segment_index=0,
    time_range=[0, 10], 
    backend="ipywidgets"
    )
```

Here is where the "recMotion_whiten" looks the same as "rec_prepro_whitened_match".

I have also compared {"recMotion_whiten" "recMotion1",  "rec_prepro"} and despite the difference in scale "recMotion_whiten" looks similar to "rec_prepro"

Hope it makes sense what I wrote. If not let me know and I can explain anything in further detail.

Thanks @JoeZiminski

borrepp, 2024-07-23:
Hi @JoeZiminski 

The whitening is working fine on the motion-corrected recording, without the need to save any traces beforehand. 

I don't know what happened, I ran it at least 3 times yesterday to be sure before posting any comment, but today I did it again and now is working fine. 

I'm sorry, but it seems that I might have had some error in my code.

I will close this issue.

Thanks for your response.

borrepp, 2024-07-23:
Hello @JoeZiminski.

It is me again following the comments about whitening after motion correction. Now I have an error when I try to run the sorter. I have tried kilosort4, mountainsort5, and spikingcircus2. All gave the same error:

```
SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "M:\Monkey_Python\SIenv\Lib\site-packages\spikeinterface\src\spikeinterface\sorters\basesorter.py", line 261, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "M:\Monkey_Python\SIenv\Lib\site-packages\spikeinterface\src\spikeinterface\sorters\external\mountainsort5.py", line 124, in _run_from_folder
    recording = cls.load_recording_from_folder(sorter_output_folder.parent, with_warnings=False)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "M:\Monkey_Python\SIenv\Lib\site-packages\spikeinterface\src\spikeinterface\sorters\basesorter.py", line 212, in load_recording_from_folder
    recording = load_extractor(json_file, base_folder=output_folder)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "M:\Monkey_Python\SIenv\Lib\site-packages\spikeinterface\src\spikeinterface\core\base.py", line 1184, in load_extractor
    return BaseExtractor.load(file_or_folder_or_dict, base_folder=base_folder)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "M:\Monkey_Python\SIenv\Lib\site-packages\spikeinterface\src\spikeinterface\core\base.py", line 769, in load
    extractor = BaseExtractor.from_dict(d, base_folder=base_folder)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "M:\Monkey_Python\SIenv\Lib\site-packages\spikeinterface\src\spikeinterface\core\base.py", line 515, in from_dict
    extractor = _load_extractor_from_dict(dictionary)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "M:\Monkey_Python\SIenv\Lib\site-packages\spikeinterface\src\spikeinterface\core\base.py", line 1104, in _load_extractor_from_dict
    new_kwargs[name] = _load_extractor_from_dict(value)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "M:\Monkey_Python\SIenv\Lib\site-packages\spikeinterface\src\spikeinterface\core\base.py", line 1123, in _load_extractor_from_dict
    extractor = extractor_class(**new_kwargs)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "M:\Monkey_Python\SIenv\Lib\site-packages\spikeinterface\src\spikeinterface\sortingcomponents\motion\motion_interpolation.py", line 302, in __init__
    assert channel_locations.ndim >= motion.dim, (
                                     ^^^^^^^^^^
AttributeError: 'dict' object has no attribute 'dim'
```

When I explore the recording object that I'm giving to the sorters (sw.plot_traces()) I can see the raw traces (whitened and motion corrected). 

When I saved the recording with traces after motion correction and later applied the whitening on that one, the sorters ran fine.

I know that the sorters that I'm using have the whitening within, but I haven't tried to run the sorters after the motion correction without the whitening.  I was trying to use spikeinterface whitening before the sorting to have a "better" comparison between the sorter output results. 

I think that I can follow one of these two strategies: either save the traces after motion correction and apply whitening afterward, or feed the motion-corrected recording ("lazy" object) and use the whitening functions provided by the sorters. 

Which one would you recommend?

Thanks in advance for your help and comments.