SpikeInterface/spikeinterface #3248: Channel ids are mixed up when loading recording

Antho2422, 2024-07-24:
Hi all, 

I have noticed a strange behavior with channel_ids that might be due to an update because I never noticed this in the past. 

I am working with Neuralynx data and using `si.extractors.read_neuralynx()` function. 

In the past I had the feeling that all the channel_ids were sorted, but now when I'm loading my data everything is mixed up (see example below).

[image]

Is this an expected behavior ? And if yes, is there a method to order the channels and channel ids in the recording object ? 

Thank you, 
Anthony

zm711, 2024-07-24:
Have you updated Neo at all? We have had some work on Neuralynx (mostly @samuelgarcia so I don't know if a change occurred during that work). Also @h-mayorquin changed how displaying this info works as you can see from the nice html display in your notebook. So I think we need to know version numbers of Neo and spikeinterface to figure this out.

Are you just doing a frame slice or did you do a channel slice before as well?

Antho2422, 2024-07-24:
[image]
 Here are the versions I'm using

Also, I frame sliced the recording just before this yes but no channel slice. I have a folder with 2 stream ids and I loaded a single stream id by giving it as arg in `read_neuralynx()`

zm711, 2024-07-24:
I think we will have to wait for @samuelgarcia to check this since I don't know neuralynx that well. 

If you do 
```python
probe_from_rec = recording.get_probe()
probe_from_rec.to_dataframe(complete=True).loc[:, ["contact_ids", "device_channel_indices"]]
```
and then print that dataframe out what does it look like?

zm711, 2024-07-24:
Wait a tic.

So the idea is the contact_ids are numbers indicating which contact we are referring to on the probe itself. So it sounds like contacts 0-8 should be on probe 1, 9-15 on probe 2 and 16-23 on probe 3. Then the device_channel_indices should be numbers between 0-24 that rearrange the contact_ids such that they are linked to the correct channel_id. So because I don't know the device correct me if I'm wrong, but the `mHaT2_4` is related to the channel_id of the recording equipment and not the actual probe itself right? 

I think the confusion is you are trying to attach the channel_id to the probe identity when what you really need to do is set the `device_channel_indices` to do that operation instead. I'm wondering if the sorting of the channel_ids is weird because of that.

Antho2422, 2024-07-25:
@zm711 

I don't think my recording equipment maps the probe. As far as I know, the only thing in the recording data are the ch_ids (the ones going from "128" to "151". Those ids were always ordered in ascending way and my contact names such as `mHaT2_1`, `mHaT2_2` ... where following the ch_ids since each contact is linked to this specific id. 

The thing is that before, the contact_names and ids where ordered in a way that each of my contacts where group by probe. 

Here is a dataframe representing how I expect things to be ordered : 
[image]

If I understand correctly what you said above, since the ch_ids are mixed up, I should find for each of my channel, the index where it's corresponding appear in ch_ids and then give this list to `set_global_device_channel_indices()`

I did that and it changed the group as you can see below : 
[image]

The issue it that the probes and channel does not correspond to what I expected :

[image]

In the example above I sorted by channel names but you can see that the position does not correspond at all to the example I mentionned above. 

So I'm a bit confused on how I should proceed, I feel like the mix of channel_ids is not expected and may not be due to my recording system as it was not like that a few weeks ago (at least never noticed and never had problem with it)

Thank you 👍 ,

Cheers, 
Anthony

samuelgarcia, 2024-07-25:
The problem is from https://github.com/NeuralEnsemble/python-neo/pull/1440

We need to add back the sorted(os.listdir) which was important for channel ordering

zm711, 2024-07-29:
Is this fixed @Antho2422 ? Can we close?

Antho2422, 2024-07-29:
Yes ! It is fixed @zm711 ! Thank you

zm711, 2024-07-29:
Great! Closing then :)

closed 2024-07-29 by zm711