SpikeInterface/spikeinterface #3252: impact of n_jobs and chunk_size on writing speed

MarinManuel, 2024-07-24:
I wanted to test how `n_jobs` and `chunk_size` affects writing speed, since the behavior didn't seem linear to me.

I ran a series of tests with a computer with 64 cores and 128GB of RAM. The test consists in writing a 31.54GB file (384 channels - 30.0kHz - 1 segments - 44,096,989 samples).
the command I used was:
```python
si.write_binary_recording(recording=rec, 
                              file_paths=str(target.resolve()), 
                              add_file_extension=False, 
                              dtype="int16", 
                              verbose=True, 
                              n_jobs=n_jobs, 
                              chunk_size=chunk_size, 
                              progress_bar=True)
```

The results are summarized in the graph below, and clearly show that there write times initially decrease quickly with adding more CPUs, but the effect then dissipate, probably because of increase overhead past ~24 CPUs.
[image]

The fastest write time was obtained for
```
n_jobs            24.000000
chunk_size    651000.000000
time              27.924926
```

I'm sure there are tons of factor that influence the write time, but maybe this could be informative for your users.

zm711, 2024-07-24:
Thanks so much! We've been meaning to do this. And honestly people should do this for their own computer to optimize for themselves as well. People can also control the max number of threads they generate so there another big factor people can change. The one other piece of information that would be super-valuable for users looking at this is the OS you are using.

Since Linux multiprocessing is different than Mac and Windows OS is super important in thinking about this :) 

thanks again this is really great :)

samuelgarcia, 2024-07-24:
Hi.
Thank for this graph, this is very very interesting.
I guess this is also depedant of the `rec` and how much `rec` has many preprocessing and heavy steps.

Apparently the system has an optimum to balance between context/process switch and number of workers doing computations.

We should add something like on the doc.

zm711, 2024-07-24:
Also tagging @h-mayorquin because he loves this stuff.

MarinManuel, 2024-07-25:
There was a glitch with the tests using 24 cores, I reran and updated the graph.

The recording that was being written was a zarr file, that had pre-processed previously. no lazy operations were done before writing to binary

MarinManuel, 2024-07-26:
I wanted to do a new graph with more sensible values for `chunk_size` so I ran a new test overnight, but the results did not exactly reproduce, so YMMV ::shrug::
[image]

The part that remains true, however, is that there is not a strong gain after `n_jobs>24`

MarinManuel, 2024-07-30:
Happy to run more tests with different parameters if that could be useful to integrate in the documentation. Let me know what range of parameters you would like me to run.

> I think using memory of the chunk instead of the chunk size would be more general because not all recordings have the same number channels. I also think that the real relevant parameter is how you divde your free ram among your chunks and processess which is not clear in the graph above.

This raises a point of potential improvement in the documentation. [This page](https://spikeinterface.readthedocs.io/en/stable/modules/core.html#parallel-processing-and-job-kwargs) talks about the job-kwargs but I am not sure what the difference is between `chunk_memory` and `total_memory` (and why I played with `chunk_size` instead). Is `chunk_memory` == `total_memory` / `n_jobs`? Or something else? This could be explained better.

> 
> In my own experiments larger chunks are always better as long as you can fit all of the min ram by maximising process.

Should I run the tests with `total_memory` == total RAM then?

> 
> I also don't expect a difference between real cores and hyperthreading here but how many real cores do you have?

The machine I ran these tests has 64 real cores and 128GB of RAM, AFAIK

h-mayorquin, 2024-07-30:
> This raises a point of potential improvement in the documentation. [This page](https://spikeinterface.readthedocs.io/en/stable/modules/core.html#parallel-processing-and-job-kwargs) talks about the [...]

Yes, this is lacking. Hope this helps:
[image]

Pending for us to improve the documentation in that regards. You are correct.

> Should I run the tests with total_memory == total RAM then?

So I think the important parameter is `memory_resources_used = (n_jobs x chunk_memory) / system_ram`

Now, this quantity can be varied in two ways, by changing `n_jobs` or by changing `chunk_memory`. I would do two plots with time in the y-axis and memory_resources_used in the x axis each with either `n_jobs` varying or `chunk_memory` varying till you sature to 80 % of you total resources. You can also do this in a single plot in a grid with imshow but I am not sure what will be more clear.

Importantly, we want to get out this any reading side effects so I suggest your use or lazy machinery which will also allow you to define arbitrarly large recordings that can be tested in your machine. I suggest something along these lines:

```python
from spikeinterface.core import generate_recording_by_size, write_binary_recording
import psutil

recording = generate_recording_by_size(full_traces_size_GiB=60.0)

# Get total system ram
system_ram_GiB = psutil.virtual_memory().total / 1024**3

# Need a loop here to test different combination of n_jobs and chunk_memory keeping memory_resources_used < system_ram_GiB
n_jobs = 3
chunk_memory = "1G"
job_kwargs = {"n_jobs": n_jobs, "chunk_memory": chunk_memory, "progress_bar": True} 
memory_resources_used = n_jobs * float(chunk_memory[:-2]) 

file_path = Path("./a_speed_test.dat")
if file_path.exists():
    file_path.unlink()
    
write_binary_recording(recording=recording, file_paths=[file_path], **job_kwargs)
```

Let me know if I am making sense.

h-mayorquin, 2024-07-30:
Be aware of the base for memory specification:
https://en.wikipedia.org/wiki/Binary_prefix