SpikeInterface/spikeinterface #3253: Functools "cache" + compatibility with python 3.8

jonahpearl, 2024-07-24:
Hi all — I tried upgrading to the new 0.101 version to get the latest SortingAnalyzer improvements. I have python 3.8 in my environment. When trying to import I get `AttributeError: module 'functools' has no attribute 'cache'`, which appears to be because `functools.cache` only exists in python 3.9 and up — before that it's [lru_cache](https://docs.python.org/3.8/library/functools.html#functools.lru_cache). So either needs to use the lru cache, unless you know it's not going to work, in which case you might want to require python 3.9.

Thanks!

Full error trace:
```python
AttributeError                            Traceback (most recent call last)
Cell In[3], line 1
----> 1 import spikeinterface.full as si

File ~/miniconda3/envs/spikeinterface/lib/python3.8/site-packages/spikeinterface/full.py:27
     25 from .widgets import *
     26 from .exporters import *
---> 27 from .generation import *

File ~/miniconda3/envs/spikeinterface/lib/python3.8/site-packages/spikeinterface/generation/__init__.py:24
     16 from .noise_tools import generate_noise
     18 from .drifting_generator import (
     19     make_one_displacement_vector,
     20     generate_displacement_vector,
     21     generate_drifting_recording,
     22 )
---> 24 from .template_database import (
     25     fetch_template_object_from_database,
     26     fetch_templates_database_info,
     27     list_available_datasets_in_template_database,
     28     query_templates_from_database,
     29 )
     31 # expose the core generate functions
     32 from ..core.generate import (
     33     generate_recording,
     34     generate_sorting,
   (...)
     47     inject_templates,
     48 )

File ~/miniconda3/envs/spikeinterface/lib/python3.8/site-packages/spikeinterface/generation/template_database.py:8
      3 import numpy as np
      5 from spikeinterface.core.template import Templates
----> 8 @functools.cache
      9 def fetch_template_object_from_database(dataset="test_templates.zarr") -> Templates:
     10     """
     11     Fetch a template dataset from the spikeinterface template database.
     12     A dataset is a collection of templates with associated metadata for one specific recording.
   (...)
     23         The templates object.
     24     """
     25     s3_path = f"s3://spikeinterface-template-database/{dataset}/"

AttributeError: module 'functools' has no attribute 'cache'
```

zm711, 2024-07-24:
I think 3.8 goes end of life in October so the plan was to try to hold out until then. But I guess maybe we should switch sooner @h-mayorquin, this was your work on the template fetching right? Thanks for letting us know @jonahpearl.

h-mayorquin, 2024-07-29:
Yeah, we should bump the minimal version to python 3.9 I think.

h-mayorquin, 2024-07-29:
See #3267 but this is a good opportunit to learn more and ask @jonahpearl why are they using python 3.8 though

jonahpearl, 2024-07-29:
I don't think there was a particular reason. I have some cookie cutter conda envs and I made it ~4 years ago so...it was just a handy default. Time to update I suppose :)

h-mayorquin, 2024-07-29:
Good, thanks for sharing. I am weary of leaving people without working workflows and is always useful to hear user stories form our user base. Sorry that this got in your way, it was me being careless and not removing the python=3.8 in pyproject.toml when we don't test on it.

zm711, 2024-07-30:
That's a good point @h-mayorquin. For neo we test 3.8 in the core (although will drop soon), but we don't test for the io's (Sam preferring to be as environmentally friendly as possible).  So we straddle the line of testing 3.8 over there, but I think this is a good discussion point in general that I will bring up at our next meeting :)

h-mayorquin, 2024-07-30:
Uh, I forgot the core tests. I have to remove them as they are being double tested now. But got stuck with some other things.

closed 2024-08-28 by h-mayorquin