SpikeInterface/spikeinterface #3254: help with creating custom probe using probeinterface

kshtjkumar, 2024-07-24:
Hi we are using a custom 32 channel probe which looks like this:
[image]

I want to map this with intan probes:
[image]

jakeswann1, 2024-07-25:
Attached is a script I've used to generate a probeinterface file with some unusual geometry in the past, hopefully it can be useful!

[generate_5x12-16_buz.txt](https://github.com/user-attachments/files/16375678/generate_5x12-16_buz.txt)

kshtjkumar, 2024-07-25:
I will try but I still get stuck on how to get it mapped based on the dimensions, here is a closer look at the probe:
[image]

jakeswann1, 2024-07-25:
Can you just treat it as a linear probe and generate using 
`probeinterface.generator.generate_linear_probe(num_elec=32, ypitch=100, contact_shapes='circle', contact_shape_params: dict = {'radius': 20})`, then set indices as described here: https://probeinterface.readthedocs.io/en/latest/examples/ex_05_device_channel_indices.html ?

kshtjkumar, 2024-08-09:
how do i set electrode ids to custom channels ? for example the electrode 1 which is at the tip is channel 0. Just like the table i pasted above. @jakeswann1

jakeswann1, 2024-08-11:
You can do this by following the instructions in the link in my previous comment. The key function is `probe.set_device_channel_indices(channel_indices)`, where `channel_indices` is a list of the channels in the order in your table. 

They do need to match the order of the electrode IDs assigned to your generated probe, which you can check with `plot_probe(probe, with_contact_id=True, with_device_index=True)`. The order of contact IDs just tells you which electrode is which for the purposes of assigning the device index. The device index is the thing which should match the order of your channels in the recording file

If you want to set the contact_ids as well, to match the order in your 'ele_no' column, you can do that with `probe.set_contact_ids`

Hope this makes sense!