SpikeInterface/spikeinterface #3255: job kwargs conflict in sorting analyzer

jonahpearl, 2024-07-25:
Hi all — in the sorting analyzer's parsing of job kwargs in `compute_one_extension`, it seems to make a decision to parse or not based on the class of the requested extension: https://github.com/SpikeInterface/spikeinterface/blob/be7ce8d21354863dadb6fd6f06d4010f5da28c1e/src/spikeinterface/core/sortinganalyzer.py#L1212

This fails in the case where the user is looping through a bunch of metrics and computing one at a time, and just passing job kwargs in (which is what I was doing), because then if the job_kwargs are present in a class that doesn't need it, `_set_params` gets unexpected kwargs. I suspect that the intended use-case here is to run a bunch at once (and `compute_several_extensions` seems to work as expected), or to set global job kwargs, but I don't see any reason why a for-loop shouldn't also be allowed.

Any reason that
```python
if extension_class.need_job_kwargs:
    params, job_kwargs = split_job_kwargs(kwargs)
else:
    params = kwargs
    job_kwargs = {}
```

shouldn't just be

```python
params, job_kwargs = split_job_kwargs(kwargs)
```
? I guess it could spit a warning that XYZ extension won't use the job kwargs, but if it doesn't use them, it's presumably fast enough that no one will really mind.

Thanks!

samuelgarcia, 2024-07-25:
I thinnk I did this to protect against pushing job_kwwargs when not needed.
You proposal do not hurt even without warning.

zm711, 2024-08-28:
Closed by #3259.

closed 2024-08-28 by zm711