SpikeInterface/spikeinterface #3257: SortingAnalyzer extension names "pca" vs "principal_components"

jonahpearl, 2024-07-25:
The docs appear to have a small bug, in the bottom snippet on [this page](https://spikeinterface.readthedocs.io/en/stable/modules/qualitymetrics.html#quality-metrics-module), there is `sorting_analyzer.compute(input={"pca": dict(...)...` but using "pca" raises a [ValueError](https://github.com/SpikeInterface/spikeinterface/blob/0a60f7c37b3a0be7cafff46c32d30e828dc5f3a9/src/spikeinterface/core/sortinganalyzer.py#L1633), and it looks like the correct name is "principal_components", based on [here](https://github.com/SpikeInterface/spikeinterface/blob/0a60f7c37b3a0be7cafff46c32d30e828dc5f3a9/src/spikeinterface/core/sortinganalyzer.py#L2124). 

Thanks!

zm711, 2024-07-25:
Eagle eye :) 

Any interest in submitting the patch :) you're right it is `"principal_components"`. If not we will open a PR shortly.

jonahpearl, 2024-07-25:
Sure, which branch should I submit the PR to?

zm711, 2024-07-25:
Thanks for asking. Our preference is for you to always make a branch on your fork to allow us to push changes if necessary and then submit the PR to main on the repo!

PR #3261 (2024-07-29): Fix name of principal_components ext in qm docs (take 2)
Fixes #3257
closed 2024-07-29 by alejoe91