SpikeInterface/spikeinterface #3266: spike_amplitudes extantion is broken

rat-h, 2024-07-29:
```python
import sys, json
import spikeinterface.full as si
print(f'Version: {si.__version__}')
si.set_global_job_kwargs(n_jobs=-1)

rec = si.load_extractor("runthepipe3-test/preprocessed")
srt = si.load_extractor("runthepipe3-test/sorting-saved")

analyzer = si.create_sorting_analyzer(
    recording=rec,
    sorting=srt,
    folder='runthepipe3-test/analyzer',
    format="binary_folder",
    overwrite=True
)

analyzer.compute("random_spikes")
analyzer.compute("waveforms")
analyzer.compute("templates")
amplitudes = analyzer.compute(input="spike_amplitudes",
                         peak_sign="neg",
                         outputs="concatenated")
print(amplitudes)
```
returns
```
Version: 0.101.0
estimate_sparsity: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1015/1015 [00:00<00:00, 5733.27it/s]
create_sorting_analyzer: recording does not have scaling to uV, forcing return_scaled=False
compute_waveforms: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1015/1015 [00:14<00:00, 69.85it/s]
Traceback (most recent call last):
  File "/home/spikeinterface/spikeinterface/test-assessment-modules.py", line 22, in <module>
    amplitudes = analyzer.compute(input="spike_amplitudes",
  File "/home/spikeinterface/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/core/sortinganalyzer.py", line 1150, in compute
    return self.compute_one_extension(extension_name=input, save=save, verbose=verbose, **kwargs)
  File "/home/spikeinterface/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/core/sortinganalyzer.py", line 1231, in compute_one_extension
    extension_instance.set_params(save=save, **params)
  File "/home/spikeinterface/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/core/sortinganalyzer.py", line 2054, in set_params
    params = self._set_params(**params)
TypeError: ComputeSpikeAmplitudes._set_params() got an unexpected keyword argument 'outputs'
```

alejoe91, 2024-07-29:
@rat-h the `outputs` parameter is not used anymore, since now it is always concatenated ;)

You can get the default extension params with:
```
import spikeinterface as si

si.get_default_analyzer_extension_params("spike_amplitudes")

>>> {'peak_sign': 'neg'}
```

rat-h, 2024-07-29:
what is I need amplitudes by units?

alejoe91, 2024-07-29:
you can still get the output by unit, but from the `get_data` function. So here's what you would do:

```
...

_ = analyzer.compute(input="spike_amplitudes", peak_sign="neg")
amplitude_ext = analyzer.get_extension("spike_amplitudes")

amplitudes_by_unit = amplitude_ext.get_data(outputs="by_unit")
```

rat-h, 2024-07-29:
[documentation](https://spikeinterface.readthedocs.io/en/latest/modules/postprocessing.html#spike-amplitudes) says I can. This constant change in postprocessing interface makes me very upset.

alejoe91, 2024-07-29:
Note that 0.101.0 has been a major release with known breaking changes with the introduction of the sorting analyzer! Hopefully the last of such major ones!

Thanks for the pointed, will fix the docs!

rat-h, 2024-07-29:
```python
import sys, json
import spikeinterface.full as si

print(f'Version: {si.__version__}')
si.set_global_job_kwargs(n_jobs=-1)

rec = si.load_extractor("runthepipe3-test/preprocessed")
srt = si.load_extractor("runthepipe3-test/sorting-saved")

analyzer = si.create_sorting_analyzer(
    recording=rec,
    sorting=srt,
    folder='runthepipe3-test/analyzer',
    format="binary_folder",
    overwrite=True
)

#DB>> spike_amplitudes
analyzer.compute("random_spikes")
analyzer.compute("waveforms")
analyzer.compute("templates")
_ = analyzer.compute(input="spike_amplitudes",peak_sign="neg")
amplitude_ext = analyzer.get_extension("spike_amplitudes")
amplitudes_by_unit = amplitude_ext.get_data(outputs="concatenated")
print(amplitudes)
```
returns
```
Version: 0.101.0
estimate_sparsity: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1015/1015 [00:00<00:00, 5596.38it/s]
create_sorting_analyzer: recording does not have scaling to uV, forcing return_scaled=False
compute_waveforms: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1015/1015 [00:09<00:00, 111.23it/s]
spike_amplitudes: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1015/1015 [00:00<00:00, 2255.79it/s]
Traceback (most recent call last):
  File "/home/spikeinterface/spikeinterface/test-assessment-modules.py", line 24, in <module>
    amplitudes_by_unit = amplitude_ext.get_data(outputs="concatenated")
  File "/home/spikeinterface/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/core/sortinganalyzer.py", line 2108, in get_data
    return self._get_data(*args, **kwargs)
  File "/home/spikeinterface/.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/postprocessing/spike_amplitudes.py", line 151, in _get_data
    raise ValueError(f"Wrong .get_data(outputs={outputs})")
ValueError: Wrong .get_data(outputs=concatenated)
```

zm711, 2024-07-29:
You want `outputs="by_unit"`. The other option is "numpy".

I think we should have a better error message though!

rat-h, 2024-07-29:
@zm711 old version accepted '`by_unit` or `concatinated`. There are modules for optimization which use both mode. You cannot just drop one option.

zm711, 2024-07-29:
concatenated is now `numpy`. It will just return the array. I honestly don't know why the argument changed.

PR #3268 (2024-07-29): Fix postprocessing docs
Fixes #3266
closed 2024-07-29 by alejoe91