SpikeInterface/spikeinterface #3271: Spykingcirus2 not detecting probe

developersrkl, 2024-07-29:
Greetings community,

I had this issue for a while, where my internal sorter of choice (spykingcircus2) doesn't seem to detect my probes, even though I have a probe object instantiated and have set the probe to the recordings, my code is here below, can someone tell me if this is a syntax error? When I debug and do a print of the probes object, it prints out all the right numbers, even though it's not recognized by the sorter. Thanks!

[image]

yger, 2024-07-29:
This is not an issue. SC2 is handling the probe properly despite this warning that can sometimes appears. It is due to the creation of an internal recording (without probes) just before template matching. Not really sure why this message is appearing to be honest, seems to be mostly on windows. However as said, you can just ignore it

developersrkl, 2024-08-02:
I wanted to ask why this code isn't discovering all of the units of the recording? It should be getting 170 ish units, but this code seems to only be getting one electrode, hence why it's only 2 units, thanks!

yger, 2024-08-26:
Can you show print(recording) and print(probe), just to be sure data/probe are correctly loaded?

zm711, 2024-12-04:
Hey @developersrkl is this still a problem or have you figured this out? Can we close this issue?

zm711, 2025-08-22:
As we have not received an answer in 9 months I'll close this. But feel free to re-open if something comes up or open a new issue!

closed 2025-08-22 by zm711