SpikeInterface/spikeinterface #3272: Sorting in docker with kilosort 4 fails

jonpedros, 2024-07-30:
Sorting in docker fails with Kilosort 4when running it on tetrode grouped, bandpassed, rereferenced, whitened data, although it works fine with other sorters, including tridesclous and mountainsort4. A 5 minutes slice of the data is used with `BaseRecording.get_traces()`.

```python
    params = {'skip_kilosort_preprocessing': True, 
              'do_CAR': False,
              'do_correction': True}
    sorting = ss.run_sorter_by_property(sorter_name=sorter,
                                        recording=recording_preprocessed,
                                        grouping_property='group',
                                        folder=sorter_fp,
                                        docker_image=True,
                                        verbose=True,
                                        **params)
```
Runtime trace:

```python
{
    "sorter_name": "kilosort4",
    "sorter_version": "4.0.12",
    "datetime": "2024-07-29T08:22:40.976628",
    "runtime_trace": [],
    "error": true,
    "error_trace": [
        "Traceback (most recent call last):",
        "  File \"/root/.local/lib/python3.11/site-packages/spikeinterface/sorters/basesorter.py\", line 261, in run_from_folder",
        "    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)",
        "  File \"/root/.local/lib/python3.11/site-packages/spikeinterface/sorters/external/kilosort4.py\", line 261, in _run_from_folder",
        "    st, tF, _, _ = detect_spikes(ops, device, bfile, tic0=tic0, progress_bar=progress_bar)",
        "                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^",
        "  File \"/home/miniconda3/lib/python3.11/site-packages/kilosort/run_kilosort.py\", line 481, in detect_spikes",
        "    st0, tF, ops = spikedetect.run(ops, bfile, device=device, progress_bar=progress_bar)",
        "                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^",
        "  File \"/home/miniconda3/lib/python3.11/site-packages/kilosort/spikedetect.py\", line 204, in run",
        "    ops['wPCA'], ops['wTEMP'] = extract_wPCA_wTEMP(",
        "                                ^^^^^^^^^^^^^^^^^^^",
        "  File \"/home/miniconda3/lib/python3.11/site-packages/kilosort/spikedetect.py\", line 72, in extract_wPCA_wTEMP",
        "    model = TruncatedSVD(n_components=ops['settings']['n_pcs']).fit(clips)",
        "            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^",
        "  File \"/home/miniconda3/lib/python3.11/site-packages/sklearn/decomposition/_truncated_svd.py\", line 208, in fit",
        "    self.fit_transform(X)",
        "  File \"/home/miniconda3/lib/python3.11/site-packages/sklearn/utils/_set_output.py\", line 313, in wrapped",
        "    data_to_wrap = f(self, X, *args, **kwargs)",
        "                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^",
        "  File \"/home/miniconda3/lib/python3.11/site-packages/sklearn/base.py\", line 1473, in wrapper",
        "    return fit_method(estimator, *args, **kwargs)",
        "           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^",
        "  File \"/home/miniconda3/lib/python3.11/site-packages/sklearn/decomposition/_truncated_svd.py\", line 228, in fit_transform",
        "    X = self._validate_data(X, accept_sparse=[\"csr\", \"csc\"], ensure_min_features=2)",
        "        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^",
        "  File \"/home/miniconda3/lib/python3.11/site-packages/sklearn/base.py\", line 633, in _validate_data",
        "    out = check_array(X, input_name=\"X\", **check_params)",
        "          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^",
        "  File \"/home/miniconda3/lib/python3.11/site-packages/sklearn/utils/validation.py\", line 1082, in check_array",
        "    raise ValueError(",
        "  ValueError: Found array with 0 sample(s) (shape=(0, 61)) while a minimum of 1 is required by TruncatedSVD."
    ],
    "run_time": null
}
```

I get a different error if I skip preprocessing with SpikeInterface and do it while spike sorting with KS 4 instead.

```python
{
    "sorter_name": "kilosort4",
    "sorter_version": "4.0.12",
    "datetime": "2024-07-30T01:31:07.906966",
    "runtime_trace": [],
    "error": true,
    "error_trace": [
        "Traceback (most recent call last):",
        "  File \"/root/.local/lib/python3.11/site-packages/spikeinterface/sorters/basesorter.py\", line 261, in run_from_folder",
        "    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)",
        "  File \"/root/.local/lib/python3.11/site-packages/spikeinterface/sorters/external/kilosort4.py\", line 261, in _run_from_folder",
        "    st, tF, _, _ = detect_spikes(ops, device, bfile, tic0=tic0, progress_bar=progress_bar)",
        "                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^",
        "  File \"/home/miniconda3/lib/python3.11/site-packages/kilosort/run_kilosort.py\", line 481, in detect_spikes",
        "    st0, tF, ops = spikedetect.run(ops, bfile, device=device, progress_bar=progress_bar)",
        "                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^",
        "  File \"/home/miniconda3/lib/python3.11/site-packages/kilosort/spikedetect.py\", line 262, in run",
        "    tF[k:k+nsp] = xfeat.transpose(0,1).cpu().numpy()",
        "    ~~^^^^^^^^^",
        "  ValueError: could not broadcast input array from shape (686,4,6) into shape (686,10,6)"
    ],
    "run_time": null
}
```

alejoe91, 2024-07-30:
Same reply as #3273 ! You might be able to fix it by setting `min_template_size=4` as a parameter, but still KS4 is not designed for tetrodes

zm711, 2024-08-05:
I think we can close this one at this point. If anyone searches in the future, a reminder that KS4 (and KS2.5 and KS3) were designed for neuropixels. On KS's own repo they recommend KS2 for "low-channel count" electrodes.

closed 2024-08-05 by zm711