SpikeInterface/spikeinterface #3273: non-zero exit code with KS 2/2.5/3 on grouped data

jonpedros, 2024-07-30:
Sorting in docker fails with Kilosort 2/2.5/3 when running it on tetrode-grouped data, although it works fine with other sorters, including tridesclous2 and mountainsort4. Raw data was sliced with `BaseRecording.get_traces()` do get a 5 snippet. Also trying fiddling with CAR and file-deleting parameters but didn't get different errors.

```python
if sorter=="kilosort2":
    verbose = True
    params = {'skip_kilosort_preprocessing': False, 
              'car': True,
              'delete_recording_dat': True,
              'delete_tmp_files': True}
elif sorter=="kilosort2_5":
    verbose = True
    params = {'skip_kilosort_preprocessing': False, 
              'car': True,
              'do_correction': False,
              'delete_recording_dat': True,
              'delete_tmp_files': True}
elif sorter=="kilosort3":
    verbose = True
    params = {'skip_kilosort_preprocessing': False, 
              'car': False,
              'do_correction': False,
              'delete_recording_dat': True,
              'delete_tmp_files': True}
    sorting = ss.run_sorter_by_property(sorter_name=sorter,
                                        recording=recording_preprocessed,
                                        grouping_property='group',
                                        folder=sorter_fp,
                                        docker_image=True,
                                        verbose=True,
                                        **params)
```

Sorting in KS2 succeeds for first channel group but fails on the following one (apparently after all four channels are detected as bad channels):

```python
{
    "sorter_name": "kilosort2",
    "sorter_version": "compiled",
    "datetime": "2024-07-30T01:36:42.625851",
    "runtime_trace": [
        "Warning: X does not support locale C.UTF-8",
        "Time   0s. Determining good channels..",
        "found 4 threshold crossings in 150.88 seconds of data",
        "found 4 bad channels",
        "Time   3s. Computing whitening matrix..",
        "Getting channel whitening matrix...",
        "Channel-whitening matrix computed.",
        "Time   3s. Loading raw data and applying filters...",
        "Time   3s. Finished preprocessing 138 batches.",
        "random seed for clusterSingleBatches: 1",
        "----------------------------------------Error using indexing",
        "Subscript indices must either be real positive integers or logicals."
    ],
    "error": true,
    "error_trace": [
        "Traceback (most recent call last):",
        "  File \"/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py\", line 261, in run_from_folder",
        "    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)",
        "  File \"/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/external/kilosortbase.py\", line 217, in _run_from_folder",
        "    raise Exception(f\"{cls.sorter_name} returned a non-zero exit code\")",
        "  Exception: kilosort2 returned a non-zero exit code"
    ],
    "run_time": null
}
```

For KS2_5:

```python
{
    "sorter_name": "kilosort2_5",
    "sorter_version": "compiled",
    "datetime": "2024-07-30T02:32:27.635808",
    "runtime_trace": [
        "Warning: X does not support locale C.UTF-8",
        "Time   0s. Computing whitening matrix..",
        "Getting channel whitening matrix...",
        "Channel-whitening matrix computed.",
        "Time   3s. Loading raw data and applying filters...",
        "Time   4s. Finished preprocessing 138 batches.",
        "Drift correction DISABLED",
        "pitch is 1.000000e+01 um",
        "----------------------------------------Index in position 1 exceeds array bounds. Index must not exceed 4."
    ],
    "error": true,
    "error_trace": [
        "Traceback (most recent call last):",
        "  File \"/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py\", line 261, in run_from_folder",
        "    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)",
        "  File \"/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/external/kilosortbase.py\", line 217, in _run_from_folder",
        "    raise Exception(f\"{cls.sorter_name} returned a non-zero exit code\")",
        "  Exception: kilosort2_5 returned a non-zero exit code"
    ],
    "run_time": null
}
```
For KS3:

```python
{
    "sorter_name": "kilosort3",
    "sorter_version": "compiled",
    "datetime": "2024-07-30T02:11:54.852403",
    "runtime_trace": [
        "Warning: X does not support locale C.UTF-8",
        "Time   0s. Computing whitening matrix..",
        "Getting channel whitening matrix...",
        "Channel-whitening matrix computed.",
        "Time   2s. Loading raw data and applying filters...",
        "Time   3s. Finished preprocessing 138 batches.",
        "Drift correction DISABLED",
        "vertical pitch size is 1.000000e+01",
        "horizontal pitch size is 5",
        "Columns 1 through 7",
        "",
        "-10.0000   -7.5000   -5.0000   -2.5000         0    2.5000    5.0000",
        "",
        "Columns 8 through 9",
        "",
        "7.5000   10.0000",
        "",
        "6",
        "",
        "----------------------------------------Index in position 1 exceeds array bounds. Index must not exceed 4."
    ],
    "error": true,
    "error_trace": [
        "Traceback (most recent call last):",
        "  File \"/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py\", line 261, in run_from_folder",
        "    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)",
        "  File \"/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/external/kilosortbase.py\", line 217, in _run_from_folder",
        "    raise Exception(f\"{cls.sorter_name} returned a non-zero exit code\")",
        "  Exception: kilosort3 returned a non-zero exit code"
    ],
    "run_time": null
}
```

alejoe91, 2024-07-30:
@jonpedros Kilosort needs at least 32 channels to work properly, so it will fail for individual tetrodes. My suggestion is to either spike sort the entire recording and split later, or to use another sorter for tetrodes ;)

closed 2024-07-30 by alejoe91

jonpedros, 2024-07-31:
@alejoe91 Thank you very much for the support! So, if I can ask for a piece of advice for spike sorting tetrode recordings from cortical regions, would it be better to pursue this sorting approach with Kilosort, or would it be more recommended to give it up altogether and use algorithms more suited to low channel count recordings?

alejoe91, 2024-07-31:
You can try mountainsort5!

I recommended splitting the units after sorting if you need to do tetrode-wise analysis. I guess you need to know downstream on which tetrode unit X was found ;)