SpikeInterface/spikeinterface #3274: set times propagation

RobertoDF, 2024-07-31:
Hi,
There is something not entirely clear in how time is managed:
if i set the times to a recording object with `recording.set_times(correct_times)` and I sort that object the sorting object will not have the correct_times, so I have to do again
```
recording.set_times(correct_times)
sorting.register_recording(recording)
```
I find this quite unintuitive, is there an internal reason for this to be necessary?

Thanks!!

zm711, 2024-07-31:
Hey @RobertoDF, there a currently a bunch of discussions about this: https://github.com/SpikeInterface/spikeinterface/issues/3157, https://github.com/SpikeInterface/spikeinterface/issues/3029, (and others I can't find at the moment). Basically we are trying to get times to work better and be more intuitive, but it is still a big discussion to figure out the best way to do this.

Maybe @alejoe91 has a better work-around for now.

RobertoDF, 2024-08-27:
Times are propagated and saved to disk if I write to binary folder (`save(format="binary"`)! didn´t know that!