SpikeInterface/spikeinterface #3277: IndexError when running SortingAnalyzer.compute('quality_metrics')

MarinManuel, 2024-08-01:
When computing extensions for a `SortingAnalyzer`, I encountered the following error. This has happened for several sortings, but not all.

I can see that the error is caused by the fact that `spikes` and `all_amplitudes` do not have the same size, but that's as far as I got.

```python
> analyzer.compute('quality_metrics')
╭─────────────────────────────── Traceback (most recent call last) ────────────────────────────────╮
│ /tmp/ipykernel_1826945/2892321906.py:3 in <module>                                               │
│                                                                                                  │
│   1 a.delete_extension('quality_metrics')                                                        │
│   2                                                                                              │
│ ❱ 3 a.compute('quality_metrics')                                                                 │
│   4                                                                                              │
│                                                                                                  │
│ /work/pi_mmanuel_uri_edu/src/spikeinterface/src/spikeinterface/core/sortinganalyzer.py:1150 in   │
│ compute                                                                                          │
│                                                                                                  │
│   1147 │   │                                                                                     │
│   1148 │   │   """                                                                               │
│   1149 │   │   if isinstance(input, str):                                                        │
│ ❱ 1150 │   │   │   return self.compute_one_extension(extension_name=input, save=save, verbose=v  │
│   1151 │   │   elif isinstance(input, dict):                                                     │
│   1152 │   │   │   params_, job_kwargs = split_job_kwargs(kwargs)                                │
│   1153 │   │   │   assert len(params_) == 0, "Too many arguments for SortingAnalyzer.compute_se  │
│                                                                                                  │
│ /work/pi_mmanuel_uri_edu/src/spikeinterface/src/spikeinterface/core/sortinganalyzer.py:1233 in   │
│ compute_one_extension                                                                            │
│                                                                                                  │
│   1230 │   │   extension_instance = extension_class(self)                                        │
│   1231 │   │   extension_instance.set_params(save=save, **params)                                │
│   1232 │   │   if extension_class.need_job_kwargs:                                               │
│ ❱ 1233 │   │   │   extension_instance.run(save=save, verbose=verbose, **job_kwargs)              │
│   1234 │   │   else:                                                                             │
│   1235 │   │   │   extension_instance.run(save=save, verbose=verbose)                            │
│   1236                                                                                           │
│                                                                                                  │
│ /work/pi_mmanuel_uri_edu/src/spikeinterface/src/spikeinterface/core/sortinganalyzer.py:1936 in   │
│ run                                                                                              │
│                                                                                                  │
│   1933 │   │   │   self._save_params()                                                           │
│   1934 │   │   │   self._save_importing_provenance()                                             │
│   1935 │   │                                                                                     │
│ ❱ 1936 │   │   self._run(**kwargs)                                                               │
│   1937 │   │                                                                                     │
│   1938 │   │   if save and not self.sorting_analyzer.is_read_only():                             │
│   1939 │   │   │   self._save_data(**kwargs)                                                     │
│                                                                                                  │
│ /work/pi_mmanuel_uri_edu/src/spikeinterface/src/spikeinterface/qualitymetrics/quality_metric_cal │
│ culator.py:188 in _run                                                                           │
│                                                                                                  │
│   185 │   │   return metrics                                                                     │
│   186 │                                                                                          │
│   187 │   def _run(self, verbose=False, **job_kwargs):                                           │
│ ❱ 188 │   │   self.data["metrics"] = self._compute_metrics(                                      │
│   189 │   │   │   sorting_analyzer=self.sorting_analyzer, unit_ids=None, verbose=verbose, **jo   │
│   190 │   │   )                                                                                  │
│   191                                                                                            │
│                                                                                                  │
│ /work/pi_mmanuel_uri_edu/src/spikeinterface/src/spikeinterface/qualitymetrics/quality_metric_cal │
│ culator.py:151 in _compute_metrics                                                               │
│                                                                                                  │
│   148 │   │   │   func = _misc_metric_name_to_func[metric_name]                                  │
│   149 │   │   │                                                                                  │
│   150 │   │   │   params = qm_params[metric_name] if metric_name in qm_params else {}            │
│ ❱ 151 │   │   │   res = func(sorting_analyzer, unit_ids=non_empty_unit_ids, **params)            │
│   152 │   │   │   # QM with uninstall dependencies might return None                             │
│   153 │   │   │   if res is not None:                                                            │
│   154 │   │   │   │   if isinstance(res, dict):                                                  │
│                                                                                                  │
│ /work/pi_mmanuel_uri_edu/src/spikeinterface/src/spikeinterface/qualitymetrics/misc_metrics.py:89 │
│ 0 in compute_amplitude_cutoffs                                                                   │
│                                                                                                  │
│    887 │   │   elif sorting_analyzer.has_extension("waveforms") and peak_sign == "pos":          │
│    888 │   │   │   invert_amplitudes = True                                                      │
│    889 │   │                                                                                     │
│ ❱  890 │   │   amplitudes_by_units = _get_amplitudes_by_units(sorting_analyzer, unit_ids, peak_  │
│    891 │   │                                                                                     │
│    892 │   │   for unit_id in unit_ids:                                                          │
│    893 │   │   │   amplitudes = amplitudes_by_units[unit_id]                                     │
│                                                                                                  │
│ /work/pi_mmanuel_uri_edu/src/spikeinterface/src/spikeinterface/qualitymetrics/misc_metrics.py:80 │
│ 8 in _get_amplitudes_by_units                                                                    │
│                                                                                                  │
│    805 │   │   for unit_id in unit_ids:                                                          │
│    806 │   │   │   unit_index = sorting_analyzer.sorting.id_to_index(unit_id)                    │
│    807 │   │   │   spike_mask = spikes["unit_index"] == unit_index                               │
[... cut ...]
│    809 │                                                                                         │
│    810 │   elif sorting_analyzer.has_extension("waveforms"):                                     │
│    811 │   │   waveforms_ext = sorting_analyzer.get_extension("waveforms")                       │
╰──────────────────────────────────────────────────────────────────────────────────────────────────╯
IndexError: boolean index did not match indexed array along dimension 0; dimension is 2190473 but corresponding 
boolean dimension is 2190477

```

zm711, 2024-08-01:
Did you sort with Kilosort? (Any of the versions) We often find that kilosort generates spikes after the recording which causes some issues. I would try 

```python

sorting_cleaned = spikeinterface.curation.remove_excessive_spikes(sorting)
```

and use that for making your sorting analyzer.

MarinManuel, 2024-08-01:
Yes, you are correct, I didn't notice the warning at the end of the sorting.
I added the `remove_excessive_spikes` to my workflow.

zm711, 2024-08-01:
Cool this comes up quite a lot. We still need to figure out why this happens, but since it occurs in the sorters (and thus is external to us) it is low-priority and so using `remove_excessive_spikes` is definitely the best bet!

MarinManuel, 2024-08-02:
Sorry, I actually have a question about this.

I am doing:
```python
rec = si.read_zarr('recording.zarr')
sort = si.read_sorter_folder('kilosort_sorting_folder')
    UserWarning: The registered recording will not be persistent on disk, but only available in memory
sort2 = si.remove_excess_spikes(recording=rec, sorting=sort)
```
This works, I get a `RemoveExcessSpikesSorting` object, with no warning, and I can use that to create a SortingAnalyzer and run all the quality metrics.

However, I am having an issue when I try to save `sort2` to disk to reload it later (just in case)

I have tried:
```python
sort2.save(format='numpy_folder', folder='kilosort_sorting_noexcess')
```
which seems to work (albeit with a warning "UserWarning: The registered recording will not be persistent on disk, but only available in memory"), but if I try to load it:
```python
s2 = si.read_sorter_folder(kilosort_sorting_noexcess')
---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
Cell In[18], line 1
----> 1 s2 = si.read_sorter_folder(kilosort_sorting_noexcess)

File /work/pi_mmanuel_uri_edu/src/spikeinterface/src/spikeinterface/sorters/runsorter.py:700, in read_sorter_folder(folder, register_recording, sorting_info, raise_error)
    697 log_file = folder / "spikeinterface_log.json"
    699 if not log_file.is_file():
--> 700     raise Exception(f"This folder {folder} does not have spikeinterface_log.json")
    702 with log_file.open("r", encoding="utf8") as f:
    703     log = json.load(f)

Exception: This folder kilosort_sorting_noexcess does not have spikeinterface_log.json
```
I have tried with `register_recording=False` with the same result.

alejoe91, 2024-08-02:
@MarinManuel 

The `read_sorter_folder` is supposed to load a folder created by the `run_sorter` function.

If you use the built-in `save(folder=...)` function, you can simply use `si.load_extractor` to load it back!

closed 2024-08-02 by MarinManuel