SpikeInterface/spikeinterface #3278: Plotting function for average/pooled waveforms?

taningh86, 2024-08-01:
Hi,
I believe there used to be a function (early versions) to plot pooled waveforms which looked something like this below:
[image]

Is there a similar plotting function available currently? get_unit_waveforms() or get_unit_templates() are just getting the templates/waveforms on neighboring channels. I am looking for something that could be used for publication. Thanks!

taningh86, 2024-08-01:
No, I do use `plot_unit_waveforms` but for publication it will be nice to have an average waveform (not necessarily the best) to show. If not possible, then I can certainly use the best channel waveform. Is that possible through one of the arguments in `plot_unit_waveforms()`?

zm711, 2024-08-01:
plot_unit_waveforms takes in a `channel_ids` argument. You could set that to only one channel in order to only do one channel. Not sure how you want to do it for a pub. Like the same channel for all the units or always pick the best channel. Honestly moving forward I would assume that journals would accept multi-channel displays of this information, but I understand wanting a simple representation.

taningh86, 2024-08-01:
I agree that journals should accept multi-channel waveforms but if they are doing it now, that I am not sure about. I guess an averaged waveform has been sort of the norm and common so I am shooting for the same.

zm711, 2024-08-02:
That's totally fair. If you want the best channel for each template (the waveforms you're talking about), then you could use

`get_template_extremum_channel`. Docstring is [here](https://github.com/SpikeInterface/spikeinterface/blob/b273d0807bcb362afce7c645a91df6b897dc6de4/src/spikeinterface/core/template_tools.py#L122-L151). It provides the options to fine tune how you define "best" channel and then you could use that as the channel_id in teh plot_unit_waveforms function :)

taningh86, 2024-08-02:
Ah, ok. Thats a nice workaround. Thanks!

zm711, 2024-08-02:
Cool, I close this then and you can reopen if something goes wrong.

closed 2024-08-02 by zm711

taningh86, 2024-08-02:
Sounds good! Thanks!