SpikeInterface/spikeinterface #3282: Calculate mean wavrforms

nadine-saluti, 2024-08-05:
Not an issue, more of a question.
I have data that I preprocessed and sorted with Kilosort 2.0 using the spike interface pipeline. I then curated the data with phy, and now I want to extract the waveforms in the new units after the curation. However, I only want the templates_avergade.npy file—and preferably only for the units I labeled as "good." Extracting all the waveforms again using the extract_waveforms function seems too time-consuming. How can I do it?

zm711, 2024-08-05:
Great question!

you can use the `read_phy` function. (docstring [here](https://github.com/SpikeInterface/spikeinterface/blob/b273d0807bcb362afce7c645a91df6b897dc6de4/src/spikeinterface/extractors/phykilosortextractors.py#L37) ) which has a `keep_good_only=True` option you can set that will only load the units you labeled as "good". Unfortunately after that you would still have to run the `extract_waveforms` (it seems like you are using spikeinterface < 0.101.0 since we've had an api change (and some nice speed-ups too).  But in this case you would only extract on the good units and not waste time on all the "non-good" units :)

nadine-saluti, 2024-08-05:
Thank you! 
Is there a way to use extract_waveforms without re-loading the recording (this is pretty time-consuming as well, in my case)?

zm711, 2024-08-05:
Unfortunately not :( 

The only way you can get waveforms is with the raw data (and it is recommended to preprocess your recording as well to get accurate waveforms) . Our new version 0.101.0 let's you calculate the templates (waveform averages) on the fly so that you don't have to save a bunch of waveforms that you don't want, but it still requires the raw data. Although it takes some time this method will be more accurate in the end. 

Our new version has greater compatibility for merging inside of spikeinterface, which will automatically just pull out the waveforms and let you get new averages etc without having to re-extract. So if your workflow is really this time-consuming it might be beneficial to update and test out either `sortingview` (a web version of general phy functionality) or `spikeinterface-gui` ( a local gui with phy-like functionality). The api changes will have a little bit of a learning curve, but might fit better with your workflow and save you time in the long-run. I would suggest reading our docs on the update and seeing if it might be better for you long-term [docs here](https://spikeinterface.readthedocs.io/en/stable/tutorials/waveform_extractor_to_sorting_analyzer.html)

But for your current issue you have to re-extract!

nadine-saluti, 2024-08-05:
Alright, thanks a lot!

I will re-extract for now, but when I'm done with my current analysis, I will check out the updates. Maybe for future experiments, I will also use your GUI for the curation.

zm711, 2024-08-05:
Cool, I'll close this then, but feel free to open new issues if needed :)

nadine-saluti, 2024-08-05:
I tried to extract waveforms again but I'm running into issues.

This is my code:

import spikeinterface.full as si
import pathlib as pl
import probeinterface as pi

rhd_file_path = pl.Path(r"path\to\file\recording.rhd")
rhd_file_path = rhd_file_path.as_posix()
rec = si.read_intan(rhd_file_path, stream_id='0')
probe = pi.get_probe(manufacturer='cambridgeneurotech', probe_name='ASSY-236-H9')
probe.wiring_to_device('cambridgeneurotech_mini-amp-64')
rec = rec.set_probe(probe)

rec_filtered = si.bandpass_filter(recording=rec,freq_min=300, freq_max=6000)

phy_path = pl.Path(r"path\to\phy\folder")
phy_path = phy_path.as_posix()
phy_sorting = si.read_phy(folder_path= phy_path, exclude_cluster_groups= "noise")
phy_sorting = si.remove_excess_spikes(sorting=phy_sorting, recording=rec_filtered) 

we_folder = "/mean_waveforms"
we_folder = phy_path + we_folder
we = si.extract_waveforms(recording=rec_filtered, sorting=phy_sorting, folder=we_folder, overwrite=True)

And this is the error I get:

Traceback (most recent call last):

  Cell In[192], line 1
    we = si.extract_waveforms(recording=rec_filtered, sorting=phy_sorting, folder=we_folder, overwrite=True)

  File ~\.conda\envs\si_env\Lib\site-packages\spikeinterface\core\waveform_extractor.py:1655 in extract_waveforms
    sparsity = precompute_sparsity(

  File ~\.conda\envs\si_env\Lib\site-packages\spikeinterface\core\waveform_extractor.py:1799 in precompute_sparsity
    local_sparsity = compute_sparsity(local_we, **sparse_kwargs)

  File ~\.conda\envs\si_env\Lib\site-packages\spikeinterface\core\sparsity.py:428 in compute_sparsity
    sparsity = ChannelSparsity.from_radius(templates_or_waveform_extractor, radius_um, peak_sign=peak_sign)

  File ~\.conda\envs\si_env\Lib\site-packages\spikeinterface\core\sparsity.py:297 in from_radius
    best_chan = get_template_extremum_channel(templates_or_we, peak_sign=peak_sign, outputs="index")

  File ~\.conda\envs\si_env\Lib\site-packages\spikeinterface\core\template_tools.py:112 in get_template_extremum_channel
    peak_values = get_template_amplitudes(templates_or_waveform_extractor, peak_sign=peak_sign, mode=mode)

  File ~\.conda\envs\si_env\Lib\site-packages\spikeinterface\core\template_tools.py:52 in get_template_amplitudes
    templates_array = _get_dense_templates_array(templates_or_waveform_extractor)

  File ~\.conda\envs\si_env\Lib\site-packages\spikeinterface\core\template_tools.py:17 in _get_dense_templates_array
    raise ValueError("templates_or_waveform_extractor should be Templates or WaveformExtractor")

ValueError: templates_or_waveform_extractor should be Templates or WaveformExtractor