SpikeInterface/spikeinterface #3283: why recomputing extensions is required even after all extensions have been computed in an analyzer folder and saved in the system

taningh86, 2024-08-05:
Hi,
Not sure if this has been discussed earlier but I could not find this in open or closed issues. For me, this issue is new. When I had the beta (unreleased) version of 101 this worked fine but with the released version this is a new issue.
Basically, the issue is that it won't allow plotting or other functions on extensions (spike_amplitudes, templates, correlograms etc.) even after they have been extracted and computed in an earlier session and saved in folder in a system. This happens when the notebook was closed and get back to it. I did it with no issues in the past
What I did previously was to just load the sorting analyzer folder with 
`sorting_analyzer = si.load_sorting_analyzer(base_folder/"sorting_analyzer")
print(sorting_analyzer)` which will give the following output stating that all the extensions that have been computed and available
`SortingAnalyzer: 378 channels - 384 units - 1 segments - binary_folder - sparse - has recording
Loaded 7 extensions: correlograms, noise_levels, random_spikes, spike_amplitudes, templates, unit_locations, waveforms`
But, when I try to plot: `sw.plot_amplitudes(sa, backend="ipywidgets")`, it would through the error saying `AssertionError: You must run the extension spike_amplitudes before retrieving data`. 
Therefore, I am wondering if with the officially released 101 version if loading the analyzer folder don't work anymore and all extensions have to be recomputed if the notebook was closed or if the kernel was stopped etc.? Please let me know.

Thanks

taningh86, 2024-08-05:
Actually, never mind! I was able to do it this time and it worked with no issues. I think the last time I computed the extensions it did not compute it fully and may be the notebook was shut prematurely. My apologies. We can close this issue.

zm711, 2024-08-05:
Good. No worries, sometimes there are kernel issues (especially in notebooks) which requires closing everything and reopening. Hard to know why the problems happen. I'll close but if something else comes up let us know :)

closed 2024-08-05 by zm711