SpikeInterface/spikeinterface #3285: Neuralynx sorting extractor

HesTheMan, 2024-08-06:
I would like to use the se.read_neuralynx_sorting function on nse files
When I call the function there are more segments than expected.  sw.plot_raster does not show any rasters.
Here is a code snippet:
  
sorting   = se.read_neuralynx_sorting(nldirectory,32000,stream_id='0')
print("num_units:",sorting.get_num_units(), "num_segments:",sorting.get_num_segments())
print("num spikes per unit:",sorting.count_num_spikes_per_unit())
w = sw.plot_rasters(sorting,color='b',time_range=(0,200),segment_index=0)

num_units: 43 num_segments: 84
num spikes per unit: {0: 23364, 1: 3, 2: 77, 3: 0, 4: 15, 5: 483, 6: 57, 7: 2236, 8: 212, 9: 5850, 10: 6878, 11: 2936, 12: 0, 13: 0, 14: 4, 15: 41, 16: 2, 17: 465, 18: 27, 19: 1842, 20: 4204, 21: 12100, 22: 23364, 23: 3, 24: 77, 25: 15, 26: 483, 27: 57, 28: 2236, 29: 212, 30: 5850, 31: 6878, 32: 2936, 33: 4, 34: 1, 35: 41, 36: 2, 37: 465, 38: 27, 39: 1842, 40: 4204, 41: 12100, 42: 1}

zm711, 2024-08-06:
Just to double-check which version of Neo is in your environment? There have been some updates to the neuralynx reading capabilities so just want to double check if this has been fixed with recent versions or is still a problem.

HesTheMan, 2024-08-08:
neo==0.13.1

zm711, 2024-08-08:
Could you try 0.13.2 and see if there is any change. If not we may need to check at the Neo level (ie open a issue on the python-neo repo)

HesTheMan, 2024-08-08:
No difference.   Are there any neo functions you would recommend I try to test at the neo level?

zm711, 2024-08-08:
No. We should just take the segment structure from Neo. So if there are too many segments it means the problem is happening inside of Neo. It is probably best to open this issue on the neo repo so we can troubleshoot over there. Basically if there are gaps in timestamps or other issues with the data then it just get hives into new segments. Maybe the reader/gui you use accounts for irregularities different. Want to open an issue over there?

HesTheMan, 2024-08-08:
Sure.  What should I do?

zm711, 2024-08-08:
Neo is here: https://github.com/NeuralEnsemble/python-neo/issues. You can open the issue there. Typically it helps us most if you can share the file so we can open it locally and try to track down the exact issue causing the extra segments.

HesTheMan, 2024-08-09:
I opened an issue and uploaded the files.

samuelgarcia, 2024-08-27:
I made a pacth a few month ago you can now use strict_gap_mode=False then this multi segment error for neuralynx should be fixed. This is clearly dependant of which system you have with neuralynx.

zm711, 2024-12-03:
I'll close this since this is super active on Neo. We should focus on the neuralynx issues over there!

closed 2024-12-03 by zm711