SpikeInterface/spikeinterface #3286: error when trying to extract waveforms from data curtated with phy

nadine-saluti, 2024-08-06:
Hi, I'm trying to extract waveforms on data I curated with Phy, by loading the recording and using the read_phy function. But when I try to use the extract_waveforms function, I get an error.

This is my code:

import spikeinterface.full as si
import pathlib as pl
import probeinterface as pi

rhd_file_path = pl.Path(r"path\to\file\recording.rhd")
rhd_file_path = rhd_file_path.as_posix()
rec = si.read_intan(rhd_file_path, stream_id='0')
probe = pi.get_probe(manufacturer='cambridgeneurotech', probe_name='ASSY-236-H9')
probe.wiring_to_device('cambridgeneurotech_mini-amp-64')
rec = rec.set_probe(probe)
rec_filtered = si.bandpass_filter(recording=rec,freq_min=300, freq_max=6000)

phy_path = pl.Path(r"path\to\phy\folder")
phy_path = phy_path.as_posix()
phy_sorting = si.read_phy(folder_path= phy_path, exclude_cluster_groups= "noise")
phy_sorting = si.remove_excess_spikes(sorting=phy_sorting, recording=rec_filtered)

we_folder = "/mean_waveforms"
we_folder = phy_path + we_folder
we = si.extract_waveforms(recording=rec_filtered, sorting=phy_sorting, folder=we_folder, overwrite=True)

And this is the error I get:

Traceback (most recent call last):

Cell In[192], line 1
we = si.extract_waveforms(recording=rec_filtered, sorting=phy_sorting, folder=we_folder, overwrite=True)

File ~.conda\envs\si_env\Lib\site-packages\spikeinterface\core\waveform_extractor.py:1655 in extract_waveforms
sparsity = precompute_sparsity(

File ~.conda\envs\si_env\Lib\site-packages\spikeinterface\core\waveform_extractor.py:1799 in precompute_sparsity
local_sparsity = compute_sparsity(local_we, **sparse_kwargs)

File ~.conda\envs\si_env\Lib\site-packages\spikeinterface\core\sparsity.py:428 in compute_sparsity
sparsity = ChannelSparsity.from_radius(templates_or_waveform_extractor, radius_um, peak_sign=peak_sign)

File ~.conda\envs\si_env\Lib\site-packages\spikeinterface\core\sparsity.py:297 in from_radius
best_chan = get_template_extremum_channel(templates_or_we, peak_sign=peak_sign, outputs="index")

File ~.conda\envs\si_env\Lib\site-packages\spikeinterface\core\template_tools.py:112 in get_template_extremum_channel
peak_values = get_template_amplitudes(templates_or_waveform_extractor, peak_sign=peak_sign, mode=mode)

File ~.conda\envs\si_env\Lib\site-packages\spikeinterface\core\template_tools.py:52 in get_template_amplitudes
templates_array = _get_dense_templates_array(templates_or_waveform_extractor)

File ~.conda\envs\si_env\Lib\site-packages\spikeinterface\core\template_tools.py:17 in _get_dense_templates_array
raise ValueError("templates_or_waveform_extractor should be Templates or WaveformExtractor")

ValueError: templates_or_waveform_extractor should be Templates or WaveformExtractor

zm711, 2024-08-06:
Hey @nadine-saluti,

what version of spikeinterface do you have?

nadine-saluti, 2024-08-06:
0.100.2

zm711, 2024-08-06:
@alejoe91, do you think there was a check added here when `Templates` were accepted that now breaks the function when creating the waveform_extractor? I don't see an obvious mistake in the script, but maybe you do?

alejoe91, 2024-08-06:
The script looks fine! @nadine-saluti can you try to upgrade to 0.100.8?

nadine-saluti, 2024-08-06:
All right, the update has solved this problem.
Thanks guys!

alejoe91, 2024-08-06:
Great! closing the issue then!