SpikeInterface/spikeinterface #3288: Erorr in `set_property` with missing values for string

h-mayorquin, 2024-08-06:
The following is a bug when setting missing properties with unicode strings in `base.set_property`:

```python
from spikeinterface.core import generate_recording

num_channels = 5
recording = generate_recording(num_channels=5, durations=[1.0])
channel_ids = ["a", "b", "c", "d", "e"]
recording = recording.rename_channels(new_channel_ids=channel_ids)

incomplete_values = ["value"] * (num_channels - 1)
# Passs a missing value
recording.set_property(
    key="missing_property",
    ids=channel_ids[:-1],
    values=incomplete_values,
    missing_value="missing",
)

property_in_recording = recording.get_property("missing_property")
expected_array = np.array(incomplete_values + ["missing"])
assert np.array_equal(property_in_recording, expected_array)

```

The output in property_in_recording is:
```python
array(['value', 'value', 'value', 'value', 'missi'], dtype='<U5')
```

I found the error while working on https://github.com/SpikeInterface/spikeinterface/pull/3287. I can fix it after we are done with that one.

zm711, 2024-08-28:
Is this fixed with #3287 @h-mayorquin ?

h-mayorquin, 2024-08-28:
No. Still to be fixed.

chrishalcrow, 2025-03-20:
lol

closed 2025-03-25 by alejoe91