SpikeInterface/spikeinterface #3291: about templates_average.npy

ohkoibln, 2024-08-07:
Hi , I used kilosort2 and got my results. However, my templates_average.npy  has lots of zero, which I think whether I set something wrong?

chrishalcrow, 2024-08-07:
Hi @ohkoibln  , it would be helpful to see more details of what you did. Could you post some of your code? It would be great to see any preprocessing steps you applied and the line where you call kilosort.

zm711, 2024-08-07:
Also what is `templates_averages.npy`. That file is not written by Kilosort2 and is not exported by spikeinterface. And with version 0.101.0 of spikeinterface you would have an `average.npy` file in the templates folder of extensions. So I agree with Chris. Without seeing what you did and what version of spikeinterface you were using we can't really help you.

ohkoibln, 2024-08-09:
> 还有什么`templates_averages.npy`。该文件不是由 Kilosort2 编写的，也不是由 spikeinterface 导出的。使用 0.101.0 版本的 spikeinterface，您会`average.npy`在扩展的 templates 文件夹中看到一个文件。所以我同意 Chris 的观点。如果看不到您做了什么以及您使用的是哪个版本的 spikeinterface，我们真的无法帮助您。

Yes, It's average.npy. I think I set some pars too strictly.

alejoe91, 2024-08-09:
@ohkoibln still unsure where you are getting the `average.npy` file. Kilosort2 should not produce such file.. can you paste the script that you're running?

zm711, 2024-12-04:
Just want to follow up @ohkoibln is this solved or can we help more?

zm711, 2025-02-05:
@ohkoibln I'm going to close this as stale, but feel free to reopen if we can help you more with things! :)

closed 2025-02-05 by zm711