SpikeInterface/spikeinterface #3293: about unit number

bxy666666, 2024-08-08:
I tried kilosort2 on different files, and I got different number of units. I want to compare the result of these two files, so can I  get the same number of units when solving different files? Thanks!

bxy666666, 2024-08-08:
> Each spike sorting run will generate a different number of files, because the probe will record from a different number of neurons. So you cannot compare the output of spike sorting from different files.

I think I didn't say accurately. I mean that I have 2 files, (day1 and day2), when I used kilosort2, h5(day1) got 50units,  h5(day2) got 70units, the number of the units is not equal. Could I make them same?

zm711, 2024-08-08:
That's a big problem and something that plenty of people are actively working on (trying to track units/neurons over multiple days).  The repo had a big discussion on this problem [here](https://github.com/SpikeInterface/spikeinterface/issues/2626), but you would likely need to seek out other tools that might help you make these type of alignments. It is not easy to do though!

bxy666666, 2024-08-09:
> 这是一个大问题，许多人正在积极解决它（尝试在多天内跟踪单元/神经元）。repo 对此问题进行了深入讨论[，](https://github.com/SpikeInterface/spikeinterface/issues/2626)但您可能需要寻找其他工具来帮助您进行此类对齐。但这并不容易！

Thanks for your reply!

zm711, 2024-08-09:
I'll close this but the issue I linked will remain active. So my advice would be to check that link every once in a while to see if any tools have been incorporated here.

closed 2024-08-09 by zm711