SpikeInterface/spikeinterface #3297: Using phy on sorted multirecording

danielpollak, 2024-08-09:
Hello, I followed the following steps to concatenate and sort my Neuropixel 1,0 openephys recording with SI version: '0.101.0':

```
#%%
import spikeinterface.full as si
import spikeinterface.extractors as se
import spikeinterface.sorters as ss
import numpy as np 
import matplotlib.pyplot as plt
from pathlib import Path
import superiorcolliculus.processing as processing

#%%
recording_paths = [    path1,    path2]

multirecording_AP = si.concatenate_recordings([se.read_openephys(path, stream_id="0") for path in recording_paths])

#%%
NP1 = processing.get_NP1()
NP1.set_device_channel_indices(np.arange(384))
# Not in place
multirecording_AP = multirecording_AP.set_probe(NP1)

# sorting = si.read_sorter_folder(base_folder / 'kilosort4')
sorting = si.run_sorter(
    'kilosort4', multirecording_AP,
    folder=output_folder,
    verbose=True, **si.get_default_sorter_params("kilosort4"),
)
#%% Phy
sorting_analyzer = si.create_sorting_analyzer(sorting=sorting, recording=multirecording_AP,)

sorting_analyzer.compute("random_spikes", method="uniform", max_spikes_per_unit=500)
job_kwargs = dict(n_jobs=40, chunk_duration='1s', progress_bar=True)

sorting_analyzer.compute("waveforms",  ms_before=1.5,ms_after=2., **job_kwargs)
sorting_analyzer.compute("templates", operators=["average", "median", "std"])
sorting_analyzer.compute("noise_levels")

sorting_analyzer.compute("correlograms")
sorting_analyzer.compute("unit_locations")
sorting_analyzer.compute("spike_amplitudes", **job_kwargs)
sorting_analyzer.compute("template_similarity")
sorting_analyzer

# some computations are done before to control all options
_ = sorting_analyzer.compute('principal_components', n_components = 5, mode="by_channel_local")

#%%
from spikeinterface.exporters import export_to_phy

# the export process is fast because everything is pre-computed
export_to_phy(sorting_analyzer=sorting_analyzer, output_folder=phy_output_folder, n_jobs=3)
```

which works great, but I open phy to curate it, I get this:
[image]

which suggests I have two shanks instead of 1. I think I must have concatenated the recordings incorrectly, but I couldn't figure out another way to do it. Can you help me understand what happened here?

alejoe91, 2024-08-09:
Hi @danielpollak 

A couple of comments:

1. you shouldn't need the `multirecording_AP = multirecording_AP.set_probe(NP1)`, since the probe is loaded automatically by the `read_openephys`
2. Why do you think it's 2 shanks from the Phy output? all channel groups are 0 and the 4 columns you see seem to correspond to the 4 NP1 columns of the probe

alirza67, 2024-08-09:
Does your probe have a staggered layout? If yes, then it should show four columns.

Also, instead of this :
```
#%%
NP1 = processing.get_NP1()
NP1.set_device_channel_indices(np.arange(384))
# Not in place
multirecording_AP = multirecording_AP.set_probe(NP1)
```
You can get the probe info from one single recording and assign it to the concatenated multi-recording.

danielpollak, 2024-08-09:
Hi @alejoe91, thanks for your reply.
1. That worked. Thanks!
2. Whoops, I did forget that there are four columns. Thanks for the reminder.

closed 2024-08-09 by danielpollak