SpikeInterface/spikeinterface #3300: Proper way to align OpenEphys recording and stimulations

MarinManuel, 2024-08-13:
Hi,
I am trying to use events stored in an OpenEphys Binary file to remove artifacts in a recording, however, I am having troubles aligning the two.

For instance:
```
rec = si.read_openephys(trial1_folder / '2024-08-13_11-24-37', block_index=0, stream_name='Record Node 103#NI-DAQmx-102.PXIe-6341')
si.plot_traces(recording=rec, segment_index=0, channel_ids=['AI1'], show_channel_ids=True, time_range=[0.,15])
```
shows stimulation artifacts around 1.04 sec
[image]

However, when I load the events using
```
evts = si.read_openephys_event(trial1_folder / '2024-08-13_11-24-37', block_index=0)
evts = evts.get_events(channel_id='PXIe-6341Digital Input Line', segment_index=0)
stim_times = evts[evts['label']=='1']
stim_times[:5]

array([(2.85033246, 0.00099999, '1'), (2.86133295, 0.00099999, '1'),
       (2.87233289, 0.00099999, '1'), (2.88333282, 0.00099999, '1'),
       (2.89433276, 0.00099999, '1')],
      dtype=[('time', '<f8'), ('duration', '<f8'), ('label', '<U100')])
```
We can see that the first stim is at time 2.85. This is because in OpenEphys, the file does not necessarily starts at 0s, since acquisition can start earlier than recording.

This is especially an issue when I concatenate recordings.

I would be happy to work on making `si.read_openephys()` and `si.read_openephys_event` be more consistent, but first I wanted to double check that I am not missing something obvious.

MarinManuel, 2024-08-14:
Nevermind, just had the realization that I need to load the file with `load_sync_timestamps=True`

closed 2024-08-14 by MarinManuel

alejoe91, 2024-08-14:
Thanks for figuring out! Yes it gets lost in concatenation on purpose. Concatenation makes a single segment, which conceptually is a piece of contiguous data. So you are not formally allowed to have gaps in the timestamps.

What you can do is build the concatenated timestamps externally and set it to the entire concatenated object OR concatenate just for spike sorting and then split back into segments (check the `si.split_sorting()` function) and continue with multi-segment objects