SpikeInterface/spikeinterface #3305: about time slice

bxy666666, 2024-08-15:
I want to analyze some time slice, for example, 100ms after time point 10.34s, how could I solve it, Thanks!

zm711, 2024-08-15:
If you want to time slice your recording you can just `time_slice` explained [here](https://github.com/SpikeInterface/spikeinterface/blob/9d6ad5b29e21cc6822612beae924a285ea0e61d9/src/spikeinterface/core/baserecording.py#L701-L723)

so in your case you would do 

```python
sliced_recording = recording.times_slice(start_time=10.34, end_time = 10.44)
```

If that's not what you want then I think we will need a bit more explanation for what you are hoping to do.

bxy666666, 2024-08-15:
> 如果你想对录音进行时间分割，你可以[在这里](https://github.com/SpikeInterface/spikeinterface/blob/9d6ad5b29e21cc6822612beae924a285ea0e61d9/src/spikeinterface/core/baserecording.py#L701-L723)`time_slice`解释[](https://github.com/SpikeInterface/spikeinterface/blob/9d6ad5b29e21cc6822612beae924a285ea0e61d9/src/spikeinterface/core/baserecording.py#L701-L723)
> 
> 所以你的情况是你应该这样做
> 
> ```python
> sliced_recording = recording.times_slice(start_time=10.34, end_time = 10.44)
> ```
> 
> 如果这不是您想要的，那么我认为我们需要对您希望做的事情进行更多的解释。

thanks, futher, I had a serise time slice, should I need to sliced_recording1 = recording.times_slice(start_time=10.34, end_time = 10.44),   sliced_recording2 = recording.times_slice(start_time=11.34, end_time = 11.44)，.....or may I solve them in a serise?

bxy666666, 2024-08-15:
> So is your goal to actual delete chunks of time? Like in the end do you want to stitch all those slices back together? Or you want to analyze slice separately?

I want to analyse both the whole h5 and the  slices separately?

bxy666666, 2024-08-15:
> > So is your goal to actual delete chunks of time? Like in the end do you want to stitch all those slices back together? Or you want to analyze slice separately?
> 
> I want to analyse both the whole h5 and the slices separately?

I want to first analyse the whole h5, then extract the data in my slices

zm711, 2024-08-15:
I think I need to understand what you mean by "extract" the data in your slices. Could you share a script of pseudo code of what you want to do? Then we might be able to tell you how to actually do it. :)

zm711, 2024-08-16:
In that case you might want to `remove_artifacts` instead. See docstring [here](https://github.com/SpikeInterface/spikeinterface/blob/9d6ad5b29e21cc6822612beae924a285ea0e61d9/src/spikeinterface/preprocessing/remove_artifacts.py#L13-L91)

Basically you would do:
```python
list_of_artifacts = [1.33, 3.34]
cleaned_recording = spre.remove_artifacts(recording, list_triggers=list_of_artifacts, ms_before=0, ms_after=100)
```

then you could look for the other arguments to decide if you need to change them.

bxy666666, 2024-08-20:
> In that case you might want to `remove_artifacts` instead. See docstring [...]

I think that this is to delete, however I want to save recordings in my slices

alejoe91, 2024-08-20:
@bxy666666 the `remove_artifacts` doesn't delete the "bad" slices, but it replaces them with 0s (@zm711 note that the `list_triggers` need to be in samples, not seconds :) ). The only caveat is that artifacts are assumed to have the same durations.

If you have a list of periods you want to remove that have different durations, you can use the [`silence_periods`](https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.preprocessing.silence_periods) function.

This is how it works:
```
list_periods_seconds = [[1.23, 1.33], [2,34,3,34]] # you can add more

# we first need to convert the seconds to samples
list_periods_samples = []
for period_s in list_periods_seconds:
    period_samples = [recording.time_to_sample_index(period_s[0]), recording.time_to_sample_index(period_s[1])]
    list_periods_samples.append(period_samples)

cleaned_recording = spre.silence_periods(recording, list_periodslist=list_periods_samples, mode="zeros")
```

I think this is the best way yo handle bad periods, because if you remove the bad periods and then stitch the good periods back together, you mess up timing information!

You can also use `mode="noise"` to replace the bad intervals with a gaussian noise estimated from each channel.

bxy666666, 2024-08-20:
> @bxy666666不会`remove_artifacts`删除“坏”切片，但会用 0 替换它们（@zm711请注意，`list_triggers`需要以样本为单位，而不是以秒为单位：)。唯一需要注意的是，工件被假定具有相同的持续时间。
> 
> 如果您有一个想要删除的具有不同持续时间的时期列表，则可以使用该[`silence_periods`](https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.preprocessing.silence_periods)函数。
> 
> 工作原理如下：
> 
> ```
> list_periods_seconds = [[1.23, 1.33], [2,34,3,34]] # you can add more
> 
> # we first need to convert the seconds to samples
> list_periods_samples = []
> for period_s in list_periods_seconds:
>     period_samples = [recording.time_to_sample_index(period_s[0]), recording.time_to_sample_index(period_s[1])]
>     list_periods_samples.append(period_samples)
> 
> cleaned_recording = spre.silence_periods(recording, list_periodslist=list_periods_samples, mode="zeros")
> ```
> 
> 我认为这是处理坏时期的最好方法，因为如果您删除坏时期然后将好时期重新​​拼接在一起，您就会搞乱时间信息！
> 
> 您还可以用`mode="noise"`从每个通道估计的高斯噪声来替换坏间隔。

well, my aim is that compare "no stimulation period" with "with stimulation period". What I did is that I have a file, I gave some stimulation at some time point, for example: my recording is 0-30s, I gave stimulaiton at 0.1s,   2.1s,  3.1s......，I want to analyse the period after these timepoint   with those no stimulation such as 0.5s, 2.5s......，I do not have bad things.

closed 2024-08-25 by bxy666666