SpikeInterface/spikeinterface #3306: Errors compiling documentation locally

tabedzki, 2024-08-15:
Steps to reproduce (Python 3.12.4):

1. gh repo clone spikeinterface/spikeinterface && cd spikeinterface
2. uv venv && . .venv/bin/activate && uv pip install -e '.[dev]'
3. sphinx-build -b html doc ./doc/_build/

Error 1:
```
Running Sphinx v8.0.2
loading translations [en]... done

Extension error:
Could not import extension sphinxcontrib.jquery (exception: No module named 'sphinxcontrib.jquery')
```

Steps to remediate:
1. Add `sphinxcontrib.jquery` to the `docs` list within `pyproject.toml` around line 185
2. uv pip install -e '.[dev]' --no-cache

Re-running `sphinx-build` 

Error/Warning Stack:

```
Running Sphinx v8.0.2
loading translations [en]... done
Using Sphinx-Gallery to convert rst text blocks to markdown for .ipynb files.
[autosummary] generating autosummary for: api.rst, authors.rst, development/development.rst, get_started/import.rst, get_started/index.rst, get_started/install_sorters.rst, get_started/installation.rst, get_started/quickstart.rst, how_to/analyze_neuropixels.rst, how_to/benchmark_with_hybrid_recordings.rst, ..., tutorials/qualitymetrics/plot_4_curation.rst, tutorials/qualitymetrics/sg_execution_times.rst, tutorials/sg_execution_times.rst, tutorials/waveform_extractor_to_sorting_analyzer.rst, tutorials/widgets/plot_1_rec_gallery.rst, tutorials/widgets/plot_2_sort_gallery.rst, tutorials/widgets/plot_3_waveforms_gallery.rst, tutorials/widgets/plot_4_peaks_gallery.rst, tutorials/widgets/sg_execution_times.rst, whatisnew.rst
/Users/ct5868/code/spikeinterface/.venv/lib/python3.12/site-packages/hdbscan/plots.py:448: SyntaxWarning: invalid escape sequence '\l'
  axis.set_ylabel('$\lambda$ value')
/Users/ct5868/code/spikeinterface/.venv/lib/python3.12/site-packages/hdbscan/robust_single_linkage_.py:154: SyntaxWarning: invalid escape sequence '\{'
  """Perform robust single linkage clustering from a vector array
/Users/ct5868/code/spikeinterface/.venv/lib/python3.12/site-packages/pymde/functions/losses.py:1: SyntaxWarning: invalid escape sequence '\d'
  """Losses: distortion functions derived from original deviations.
/Users/ct5868/code/spikeinterface/.venv/lib/python3.12/site-packages/pymde/functions/losses.py:102: SyntaxWarning: invalid escape sequence '\e'
  """
/Users/ct5868/code/spikeinterface/.venv/lib/python3.12/site-packages/pymde/functions/penalties.py:311: SyntaxWarning: invalid escape sequence '\l'
  """:math:`p(d) = \log(1 + d^{\\text{exponent}})`"""
/Users/ct5868/code/spikeinterface/.venv/lib/python3.12/site-packages/pymde/functions/penalties.py:325: SyntaxWarning: invalid escape sequence '\l'
  """:math:`p(d) = \log(1 - \\exp(-d^\\text{exponent}))`"""
/Users/ct5868/code/spikeinterface/.venv/lib/python3.12/site-packages/pymde/quadratic.py:126: SyntaxWarning: invalid escape sequence '\q'
  """Compute a spectral embedding
generating gallery...
generating gallery for tutorials...
generating gallery for tutorials/core... [ 17%] plot_1_recording_extractor.py
generating gallery for tutorials/core... [ 33%] plot_2_sorting_extractor.py
generating gallery for tutorials/core... [ 50%] plot_3_handle_probe_info.py
generating gallery for tutorials/core... [ 67%] plot_4_sorting_analyzer.py
WARNING:
../examples/tutorials/core/plot_4_sorting_analyzer.py unexpectedly failed to execute correctly:

    Traceback (most recent call last):
      File "/Users/ct5868/code/spikeinterface/examples/tutorials/core/plot_4_sorting_analyzer.py", line 41, in <module>
        local_path = download_dataset(repo=repo, remote_path=remote_path, local_folder=None)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "/Users/ct5868/code/spikeinterface/src/spikeinterface/core/datasets.py", line 66, in download_dataset
        dataset = datalad.api.install(path=local_folder, source=repo)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "/Users/ct5868/code/spikeinterface/.venv/lib/python3.12/site-packages/datalad/interface/base.py", line 773, in eval_func
        return return_func(*args, **kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
      File "/Users/ct5868/code/spikeinterface/.venv/lib/python3.12/site-packages/datalad/interface/base.py", line 763, in return_func
        results = list(results)
                  ^^^^^^^^^^^^^
      File "/Users/ct5868/code/spikeinterface/.venv/lib/python3.12/site-packages/datalad/interface/base.py", line 940, in _execute_command_
        raise IncompleteResultsError(
    datalad.support.exceptions.IncompleteResultsError: Command did not complete successfully. 1 failed:
    [{'action': 'install',
      'error_message': 'No working git-annex installation of version >= '
                       '8.20200309. Visit '
[... 254 lines cut ...]
reading sources... [100%] whatisnew

/Users/ct5868/code/spikeinterface/src/spikeinterface/core/sortinganalyzer.py:docstring of spikeinterface.core.sortinganalyzer.SortingAnalyzer.merge_units:2: WARNING: Inline interpreted text or phrase reference start-string without end-string. [docutils]
/Users/ct5868/code/spikeinterface/src/spikeinterface/curation/auto_merge.py:docstring of spikeinterface.curation.auto_merge.get_potential_auto_merge:44: WARNING: Bullet list ends without a blank line; unexpected unindent. [docutils]
/Users/ct5868/code/spikeinterface/src/spikeinterface/sortingcomponents/motion/motion_utils.py:docstring of spikeinterface.sortingcomponents.motion.motion_utils.Motion:11: ERROR: Unexpected indentation. [docutils]
/Users/ct5868/code/spikeinterface/doc/modules/sorters.rst:534: WARNING: duplicate label modules/sorters:internal sorters, other instance in /Users/ct5868/code/spikeinterface/doc/modules/sorters.rst [autosectionlabel.modules/sorters]
looking for now-outdated files... none found
pickling environment... WARNING: cannot cache unpickable configuration value: 'sphinx_gallery_conf' (because it contains a function, class, or module object) [config.cache]
done
checking consistency... done
preparing documents... done
copying assets...
copying downloadable files... [  2%] tutorials/comparison/generate_erroneous_sorting.ipynb
[...]
copying downloadable files... [100%] tutorials/widgets/plot_4_peaks_gallery.zip

copying static files... done
copying extra files... done
copying assets: done
writing output... [  1%] api

Theme error:
An error happened in rendering the page api.
Reason: UndefinedError("'style' is undefined")
```

zm711, 2024-08-15:
https://spikeinterface.readthedocs.io/en/latest/development/development.html

The development docs explain you have to have datalad and git-annex. They are so OS specific we can't automate it at this point,

tabedzki, 2024-08-15:
Quite possibly a PEBKAC situation issue on my end: when reading the "[How to build the Documentation](https://spikeinterface.readthedocs.io/en/latest/development/development.html#how-to-build-the-documentation)", it didn't specify that I needed the other installations after doing `pip install -e '.[dev]'` to build the documentation, so I just did a naive installation. Thanks for pointing out what I missed.

zm711, 2024-08-15:
I think it is useful for us to know though if people are missing it. We likely need to guide the reader to all the sections in a better way. Thanks for letting us know that the flow isn't necessarily intuitive!

tabedzki, 2024-08-15:
Since datalad has a [universal installer](https://www.datalad.org/#install), is it safe to use that as part of the installation? What might be some reasons not to include it?

```
pip install datalad-installer
datalad-installer git-annex -m datalad/packages
pip install datalad
```

zm711, 2024-08-15:
It's not quite that simple. You can look at our CI action that does this for all OSes. (@h-mayorquin got it working so he can comment more!) He originally had all of this in the docs, but we had some versioning issues before the hackathon, so I removed it, but I think he found a new way to install so maybe we put it back into the docs? Not sure.

https://github.com/SpikeInterface/spikeinterface/blob/9d6ad5b29e21cc6822612beae924a285ea0e61d9/.github/workflows/all-tests.yml#L102-L121

chrishalcrow, 2024-08-16:
Hello,

I can't reproduce the error with `sphinxcontrib.jquery`. It's a dependency of `sphinx_rtd_theme` so, for me, gets installed with that. What OS are you using? And what version of `sphinx_rtd_theme` is installed?

I do agree with the PR though - seems like a bad idea to rely on a theme package to install a package required for the docs search.

tabedzki, 2024-08-16:
I am using Python 3.12.4 and the neofetch results are below. I did not install the package separately from the spikeinterface installation process. I created a new venv to install spikeinterface.

```
(base)  ➜  spikeinterface git:(main) ✗ git remote --verbose
origin	git@github.com:SpikeInterface/spikeinterface.git (fetch)
origin	git@github.com:SpikeInterface/spikeinterface.git (push)
(base)  ➜  spikeinterface git:(main) ✗ git pull
Already up to date.
 ➜  spikeinterface git:(main) ✗ uv pip list
Package                       Version      Editable project location
----------------------------- ------------ ---------------------------------
aiobotocore                   2.13.2
aiohappyeyeballs              2.3.5
aiohttp                       3.10.3
aioitertools                  0.11.0
aiosignal                     1.3.1
alabaster                     1.0.0
altair                        5.4.0
annexremote                   1.6.5
asciitree                     0.3.3
asttokens                     2.4.1
attrs                         24.2.0
babel                         2.16.0
black                         24.8.0
boto3                         1.34.131
botocore                      1.34.131
cbor2                         5.6.4
certifi                       2024.7.4
cffi                          1.17.0
cfgv                          3.4.0
chardet                       5.2.0
charset-normalizer            3.3.2
click                         8.1.7
cloudpickle                   3.0.0
colorcet                      3.1.0
colorlog                      6.8.2
contourpy                     1.2.1
coverage                      7.6.1
cryptography                  43.0.0
cycler                        0.12.1
dask                          2024.8.0
datalad                       1.1.3
decorator                     5.1.1
distlib                       0.3.8
distributed                   2024.8.0
distro                        1.9.0
docutils                      0.21.2
elephant                      1.1.0
et-xmlfile                    1.1.0
executing                     2.0.1
fasteners                     0.19
figurl                        0.2.22
filelock                      3.15.4
flake8                        7.1.1
fonttools                     4.53.1
frozenlist                    1.4.1
fsspec                        2024.6.1
globus-sdk                    3.44.0
graphviz                      0.20.3
greenlet                      3.0.3
h5py                          3.11.0
hdbscan                       0.8.38.post1
hdmf                          3.14.3
huggingface-hub               0.24.5
humanize                      4.10.0
ibl-neuropixel                1.2.0
iblatlas                      0.5.3
ibllib                        2.38.0
iblutil                       1.12.1
identify                      2.6.0
[... 152 lines cut ...]
zarr                          2.17.2
zict
```

```
(base)  ➜  spikeinterface git:(main) ✗ neofetch
                    'c.         
                 ,xNMM.          -----------------------
               .OMMMMo           OS: macOS 14.6.1 23G93 arm64
               OMMM0,            Host: Mac15,6
     .;loddo:' loolloddol;.      Kernel: 23.6.0
   cKMMMMMMMMMMNWMMMMMMMMMM0:    Uptime: 2 days, 16 hours, 24 mins
 .KMMMMMMMMMMMMMMMMMMMMMMMWd.    Packages: 149 (brew)
 XMMMMMMMMMMMMMMMMMMMMMMMX.      Shell: zsh 5.9
;MMMMMMMMMMMMMMMMMMMMMMMM:       Resolution: 1920x1080, 1512x982
:MMMMMMMMMMMMMMMMMMMMMMMM:       DE: Aqua
.MMMMMMMMMMMMMMMMMMMMMMMMX.      WM: Rectangle
 kMMMMMMMMMMMMMMMMMMMMMMMMWd.    Terminal: iTerm2
 .XMMMMMMMMMMMMMMMMMMMMMMMMMMk   Terminal Font: Monaco 18
  .XMMMMMMMMMMMMMMMMMMMMMMMMK.   CPU: Apple M3 Pro
    kMMMMMMMMMMMMMMMMMMMMMMd     GPU: Apple M3 Pro
     ;KMMMMMMMWXXWMMMMMMMk.      Memory: 5096MiB / 36864MiB
       .cooc,.    .,coo:.

```

tabedzki, 2024-08-16:
If we need a minimum of 1.2 for `sphinx-rtd-theme`, I think we should pin a minimum version into `pyproject.toml` to ensure that it gets install along with the `jquery` module. 

After specifying the package minimum version locally, I was able to build the documentation but I got the following errors:

```
Running Sphinx v7.4.7
loading translations [en]... done
Using Sphinx-Gallery to convert rst text blocks to markdown for .ipynb files.
[autosummary] generating autosummary for: api.rst, authors.rst, development/development.rst, get_started/import.rst, get_started/index.rst, get_started/install_sorters.rst, get_started/installation.rst, get_started/quickstart.rst, how_to/analyze_neuropixels.rst, how_to/benchmark_with_hybrid_recordings.rst, ..., tutorials/qualitymetrics/plot_4_curation.rst, tutorials/qualitymetrics/sg_execution_times.rst, tutorials/sg_execution_times.rst, tutorials/waveform_extractor_to_sorting_analyzer.rst, tutorials/widgets/plot_1_rec_gallery.rst, tutorials/widgets/plot_2_sort_gallery.rst, tutorials/widgets/plot_3_waveforms_gallery.rst, tutorials/widgets/plot_4_peaks_gallery.rst, tutorials/widgets/sg_execution_times.rst, whatisnew.rst
generating gallery...
generating gallery for tutorials...
generating gallery for tutorials/core... [100%] plot_6_handle_times.py
generating gallery for tutorials/extractors... [100%] plot_2_working_with_unscaled_traces.py
generating gallery for tutorials/qualitymetrics... [100%] plot_4_curation.py
generating gallery for tutorials/comparison... [100%] plot_5_comparison_sorter_weaknesses.py
generating gallery for tutorials/widgets... [100%] plot_4_peaks_gallery.py
computation time summary:
    - ../examples/tutorials/core/plot_4_sorting_analyzer.py:                     17.28 sec   0.0 MB
    - ../examples/tutorials/qualitymetrics/plot_3_quality_mertics.py:             2.17 sec   0.0 MB
    - ../examples/tutorials/widgets/plot_3_waveforms_gallery.py:                  1.88 sec   0.0 MB
    - ../examples/tutorials/qualitymetrics/plot_4_curation.py:                    1.25 sec   0.0 MB
    - ../examples/tutorials/widgets/plot_4_peaks_gallery.py:                      1.15 sec   0.0 MB
    - ../examples/tutorials/extractors/plot_1_read_various_formats.py:            1.04 sec   0.0 MB
    - ../examples/tutorials/comparison/generate_erroneous_sorting.py:             0.00 sec   0.0 MB
    - ../examples/tutorials/comparison/plot_5_comparison_sorter_weaknesses.py:    0.00 sec   0.0 MB
    - ../examples/tutorials/core/plot_1_recording_extractor.py:                   0.00 sec   0.0 MB
    - ../examples/tutorials/core/plot_2_sorting_extractor.py:                     0.00 sec   0.0 MB
    - ../examples/tutorials/core/plot_3_handle_probe_info.py:                     0.00 sec   0.0 MB
    - ../examples/tutorials/core/plot_5_append_concatenate_segments.py:           0.00 sec   0.0 MB
    - ../examples/tutorials/core/plot_6_handle_times.py:                          0.00 sec   0.0 MB
    - ../examples/tutorials/extractors/plot_2_working_with_unscaled_traces.py:    0.00 sec   0.0 MB
    - ../examples/tutorials/widgets/plot_1_rec_gallery.py:                        0.00 sec   0.0 MB
    - ../examples/tutorials/widgets/plot_2_sort_gallery.py:                       0.00 sec   0.0 MB
loading intersphinx inventory 'neo' from https://neo.readthedocs.io/en/latest/objects.inv...
loading intersphinx inventory 'probeinterface' from https://probeinterface.readthedocs.io/en/stable/objects.inv...
building [mo]: targets for 0 po files that are out of date
writing output...
building [html]: targets for 112 source files that are out of date
updating environment: [new config] 112 added, 0 changed, 0 removed
<unknown>:310: SyntaxWarning: invalid escape sequence '\m'
<unknown>:382: SyntaxWarning: invalid escape sequence '\m'
<unknown>:310: SyntaxWarning: invalid escape sequence '\m'
<unknown>:382: SyntaxWarning: invalid escape sequence '\m'
<unknown>:298: SyntaxWarning: invalid escape sequence '\m'
<unknown>:370: SyntaxWarning: invalid escape sequence '\m'
reading sources... [100%] whatisnew
/Users/ct5868/code/spikeinterface/src/spikeinterface/core/sortinganalyzer.py:docstring of spikeinterface.core.sortinganalyzer.SortingAnalyzer.merge_units:2: WARNING: Inline interpreted text or phrase reference start-string without end-string.
/Users/ct5868/code/spikeinterface/src/spikeinterface/curation/auto_merge.py:docstring of spikeinterface.curation.auto_merge.get_potential_auto_merge:44: WARNING: Bullet list ends without a blank line; unexpected unindent.
/Users/ct5868/code/spikeinterface/src/spikeinterface/sortingcomponents/motion/motion_utils.py:docstring of spikeinterface.sortingcomponents.motion.motion_utils.Motion:11: ERROR: Unexpected indentation.
/Users/ct5868/code/spikeinterface/doc/modules/sorters.rst:534: WARNING: duplicate label modules/sorters:internal sorters, other instance in /Users/ct5868/code/spikeinterface/doc/modules/sorters.rst
looking for now-outdated files... none found
pickling environment... WARNING: cannot cache unpickable configuration value: 'sphinx_gallery_conf' (because it contains a function, class, or module object)
done
checking consistency... done
preparing documents... done
copying assets...
copying downloadable files... [100%] tutorials/widgets/plot_4_peaks_gallery.zip
[... cut ...]
copying images... [100%] tutorials/widgets/images/thumb/sphx_glr_plot_4_peaks_gallery_thumb.png
dumping search index in English (code: en)... done
dumping object inventory... done

Sphinx-Gallery successfully executed 6 out of 6 files subselected by:

    gallery_conf["filename_pattern"] = '/plot'
    gallery_conf["ignore_pattern"]   = '/generate_*'

after excluding 9 files that had previously been run (based on MD5).

embedding documentation hyperlinks...
embedding documentation hyperlinks for tutorials... [100%] plot_1_rec_gallery.html
build succeeded, 26 warnings.
```

zm711, 2024-08-16:
I think specifying the minimum sphinx-rtd-theme is the better move. Their official rec is to actually pin your docs to a specific version. So I think adding in minimums is a step in the right direction and will help new developers that want to build locally not run into an issue like this!

zm711, 2024-08-16:
I don't see errors. I see warnings. Unfortunately the warnings typically build up because we are all a bit lax on confirming any code changes don't generate warnings. So Alessio has cleaned up the warnings once. I cleaned up the warnings once (with a big assist from Alessio), and I think Chris did a cleanup once. So it's just something we do every once in a while.

tabedzki, 2024-08-16:
It's one error but not one that prevents compilation.

`/Users/ct5868/code/spikeinterface/src/spikeinterface/sortingcomponents/motion/motion_utils.py:docstring of spikeinterface.sortingcomponents.motion.motion_utils.Motion:11: ERROR: Unexpected indentation.`

I've gone ahead and added the minimum version for `sphinx-rtd-theme` over in #3307

tabedzki, 2024-08-16:
Ahh okay. Thanks for explaining a bit of the history and dependency reliance. Happy to have been able to help on this

PR #3307 (2024-08-28): Added sphinxcontrib-jquery
…query

Fixes #3306  issue of missing the `sphinxcontrib.jquery` to the pyproject.toml.

Still needs to address other components of that issue.
closed 2024-08-28 by samuelgarcia