SpikeInterface/spikeinterface #3310: set_files requires bad_channels positional argument for kilosort v4.0.16

jbowler06, 2024-08-16:
The 'set_files' method in kilosort4 version 4.0.16 now requires a 'bad_channels' positional argument which seems to call an error on line 208 in src/spikeinterface/sorters/external/kilosort4.py (..the issue can be easily mitigated by downgrading kilosort to 4.0.13)

e.g.:
```
Error running kilosort4                                                                                                                
Traceback (most recent call last):                                                                                                     
  File "/home/jack/code/ephys_tools/run_si_pipeline.py", line 303, in <module>                                                         
    main(sys.argv[1:])                                                                                                                 
  File "/home/jack/code/ephys_tools/run_si_pipeline.py", line 295, in main                                                             
    sorting = run_spike_sorting(recording_corrected, ks_path, sorter)                                                                  
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                  
  File "/home/jack/code/ephys_tools/run_si_pipeline.py", line 175, in run_spike_sorting                                                
    sorting = si.run_sorter(                                                                                                           
              ^^^^^^^^^^^^^^                                                                                                           
  File "/home/jack/.miniconda3/envs/si/lib/python3.11/site-packages/spikeinterface/sorters/runsorter.py", line 216, in run_sorter      
    return run_sorter_local(**common_kwargs)                                                                                           
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                                                                           
  File "/home/jack/.miniconda3/envs/si/lib/python3.11/site-packages/spikeinterface/sorters/runsorter.py", line 276, in run_sorter_local
    SorterClass.run_from_folder(folder, raise_error, verbose)
  File "/home/jack/.miniconda3/envs/si/lib/python3.11/site-packages/spikeinterface/sorters/basesorter.py", line 301, in run_from_folder
    raise SpikeSortingError(                                       
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting error trace:                                                        
Traceback (most recent call last):                                  
  File "/home/jack/.miniconda3/envs/si/lib/python3.11/site-packages/spikeinterface/sorters/basesorter.py", line 261, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)                                                          
  File "/home/jack/.miniconda3/envs/si/lib/python3.11/site-packages/spikeinterface/sorters/external/kilosort4.py", line 208, in _run_from_folder
    filename, data_dir, results_dir, probe = set_files(settings, filename, probe, probe_name, data_dir, results_dir)
                                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                   
TypeError: set_files() missing 1 required positional argument: 'bad_channels'                                                           
                                                                                                                                       
Spike sorting failed
```

zm711, 2024-08-16:
Thanks! We will put it on our list to fix for newer versions!

chrishalcrow, 2024-08-20:
Hello, I made a PR. If @jbowler06 or @Hobart10 could give it a try, that'd be great!

JoeZiminski, 2024-09-09:
@alejoe91 @chrishalcrow  I think this has been added now in #3339 and can be closed?

closed 2024-09-09 by chrishalcrow