SpikeInterface/spikeinterface #3311: Error when parallelizing export_to_phy

jbscherrer, 2024-08-16:
I get the following error whenever running `export_to_phy()` using `n_jobs > 1` and `chunk_duration = "100s"` on a 8,000 second dataset:

`BlockingIOError: [Errno 11] Resource temporarily unavailable`

I'm on Linux and have the `nofile` limit set to 1,000,000. I first encountered this error during preprocessing steps, but found a combination of `n_jobs` and `chunk_duration` parameters that kept the number of open files below the limit without filling up RAM. However, the exporter function throws the same error when run on more than one core, which means the PC extraction step can't be parallelized and is extremely slow. Any thoughts as to what's happening?

zm711, 2024-08-16:
It sounds like you've narrowed it down to the PC extraction or am I misunderstanding? 

I usually leave my chunk_duration to be quite a bit smaller ~1s (and my lab's workstation is Windows not linux). 

I think for our PC extraction it uses a slightly different version of parallelization than the other library so if you are confident that you've narrowed it down to PCs then I think this would be a good argument for tweaking that. I believe we discussed this right @alejoe91 @samuelgarcia ? This was also related to #2817 on a Windows computer where setting `n_jobs` was not quite working. 

The other question we need to ask is what version are you on? The waveforms storage was changed in the most recent version to help with the file limit issue. So if you aren't on 0.101.0 then I think the first step would be upgrading (there is an api change that you would need to adjust for). But if you are already at 0.101.0 then yeah it might be worth exploring how PCs are done.

jbscherrer, 2024-08-16:
The error occurs before PC extraction, during the write_binary_recording step. Changing the chunk duration to 1s still results in the error if n_jobs > 1. Version is 0.101.0.

zm711, 2024-08-16:
Cool. So this is an issue with the overall parallelization and not just the PC step. That's really good to know. Are you changing any other job_kwargs (could you just list all the job_kwargs you put in? )

jbscherrer, 2024-08-16:
Currently just `n_jobs=1, chunk_duration="1s", progress_bar=True`

zm711, 2024-08-17:
@h-mayorquin did you have any tests you wanted to suggest to help diagnose this more or just wait for Alessio and Sam.

h-mayorquin, 2024-08-20:
> @h-mayorquin did you have any tests you wanted to suggest to help diagnose this more or just wait for Alessio and Sam.

I don't really know this functionality. I would have to sit this one out.

zm711, 2024-08-21:
What if you don't concatenate? I'm still wondering if something about concatenate can cause problem with the multiprocessing at the borders of the concatenation. I can't remember when but I thought we had another issue with using multiprocessing and writing a binary after concatenating.

jbscherrer, 2024-08-21:
If I run CatGT on the dataset and then try to analyze the single file output by CatGT, I get the same `Resource temporarily unavailable` error during preprocessing instead. Using `n_jobs=8` and `chunk_duration="1s"` with the following preprocessing code:

```
rec = si.read_spikeglx(catgt_output_folder / (run_name+'_imec0'), stream_name='imec0.ap')
rec_hp = si.highpass_filter(rec)
rec_interp = si.interpolate_bad_channels(rec_hp, bad_channel_ids=bad_channel_ids)
rec_car = si.common_reference(rec_interp, operator="median", reference="global")
rec_car.save(folder=preprocessing_output_folder/run_name, overwrite=True)
```

jbscherrer, 2024-08-22:
Here's the traceback for that error:
```

_RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/home/jodie/anaconda3/lib/python3.12/concurrent/futures/process.py", line 263, in _process_worker
    r = call_item.fn(*call_item.args, **call_item.kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jodie/anaconda3/lib/python3.12/concurrent/futures/process.py", line 212, in _process_chunk
    return [fn(*args) for args in chunk]
            ^^^^^^^^^
  File "/home/jodie/anaconda3/lib/python3.12/site-packages/spikeinterface/core/job_tools.py", line 463, in function_wrapper
    return _func(segment_index, start_frame, end_frame, _worker_ctx)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jodie/anaconda3/lib/python3.12/site-packages/spikeinterface/core/recording_tools.py", line 171, in _write_binary_chunk
    memmap_obj = mmap.mmap(file.fileno(), length=length, access=mmap.ACCESS_WRITE, offset=memmap_offset)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
BlockingIOError: [Errno 11] Resource temporarily unavailable
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):

  File ~/anaconda3/lib/python3.12/site-packages/spyder_kernels/py3compat.py:356 in compat_exec
    exec(code, globals, locals)

  File /media/jodie/DATA_FAST/catgt_process.py:56
    rec_car.save(folder=preprocessing_output_folder/run_name, overwrite=True)

  File ~/anaconda3/lib/python3.12/site-packages/spikeinterface/core/base.py:863 in save
    loaded_extractor = self.save_to_folder(**kwargs)

  File ~/anaconda3/lib/python3.12/site-packages/spikeinterface/core/base.py:970 in save_to_folder
    cached = self._save(folder=folder, verbose=verbose, **save_kwargs)

  File ~/anaconda3/lib/python3.12/site-packages/spikeinterface/core/baserecording.py:537 in _save
    write_binary_recording(self, file_paths=file_paths, dtype=dtype, verbose=verbose, **job_kwargs)

  File ~/anaconda3/lib/python3.12/site-packages/spikeinterface/core/recording_tools.py:146 in write_binary_recording
    executor.run()

  File ~/anaconda3/lib/python3.12/site-packages/spikeinterface/core/job_tools.py:425 in run
    for res in results:

  File ~/anaconda3/lib/python3.12/site-packages/tqdm/notebook.py:250 in __iter__
    for obj in it:

  File ~/anaconda3/lib/python3.12/site-packages/tqdm/std.py:1181 in __iter__
    for obj in iterable:

  File ~/anaconda3/lib/python3.12/concurrent/futures/process.py:642 in _chain_from_iterable_of_lists
    for element in iterable:

  File ~/anaconda3/lib/python3.12/concurrent/futures/_base.py:619 in result_iterator
    yield _result_or_cancel(fs.pop())

  File ~/anaconda3/lib/python3.12/concurrent/futures/_base.py:317 in _result_or_cancel
    return fut.result(timeout)

  File ~/anaconda3/lib/python3.12/concurrent/futures/_base.py:449 in result
    return self.__get_result()

  File ~/anaconda3/lib/python3.12/concurrent/futures/_base.py:401 in __get_result
    raise self._exception

BlockingIOError: [Errno 11] Resource temporarily unavailable
```

alejoe91, 2024-08-22:
Is the data on a local drive or a remote server?

jbscherrer, 2024-08-22:
Data is on a local drive.

jbscherrer, 2024-08-22:
Preprocessing runs without error if I set the save location to my OS drive, rather than the data drive I've been saving to. Could the error be connected to mounting the data drive? The error persists if I run the script from the data drive.

alejoe91, 2024-08-22:
If saving locally works then for sure it must be a problem with the drive. I recommend saving locally and moving the data to the drive later!

jbscherrer, 2024-08-22:
Is there a workaround for working with data on auxiliary drives? Our recordings are typically in the 100s of GB, so it's a little unwieldy to keep everything on the OS drive.

zm711, 2024-08-22:
You said that `n_jobs=1` works right? So it seems like the data drive you're using is limited and so causing the failure. Your options are to get a better data drive (maybe check out what the writing speed is and see if you can get a drive that is more optimized--you might need different connection as well-- are you using USB? ) or you can't multiprocess with the data drive. 

For me I have no problem with the OS drive or an auxiliary drive through M.2 (it's spinning disk so it's not even that fast). I haven't tried to an external, but with file transfer over USB I usually get end up getting choking due to the increased temperature. This could be also be contributed.

jbscherrer, 2024-08-22:
Interesting, I'm also using an M.2 SSD as the data drive, with a stated speed of up to 7 GB/s. Not sure why transfer rates would be limited.

jbscherrer, 2024-08-23:
The problem seems to have been fixed by reformatting the data drive from NTFS to exFAT.

zm711, 2024-08-23:
That's good to know. I think the `BlockingIOError` is hard for us as a library to fix, but at least you found a solution. I guess python was trying to tell us something about the drive, but I wouldn't be able to guess that that would be the solution based on what the error was. I'll close this, but feel free to reopen if the error persists after this reformatting. I'm also going to change the title to try to help future users that bump into this to find this issue and test your solution :)

closed 2024-08-23 by zm711

samuelgarcia, 2024-08-27:
tricky case!
@h-mayorquin do you kown if there is a possible limitation to create and delete so many mmap object for some os ?