SpikeInterface/spikeinterface #3315: channels problem

bxy666666, 2024-08-20:
BandpassFilterRecording: 67 channels - 20.0kHz - 1 segments - 13,001,400 samples
                         650.07s (10.83 minutes) - int16 dtype - 1.62 GiB
After common reference:
CommonReferenceRecording: 67 channels - 20.0kHz - 1 segments - 13,001,400 samples
                          650.07s (10.83 minutes) - int16 dtype - 1.62 GiB
Use cache_folder=C:\Users\bxy\AppData\Local\Temp\spikeinterface_cache\tmp410o81a7\K1NDPI1B
write_binary_recording
n_jobs=1 - samples_per_chunk=20,000 - chunk_memory=2.56 MiB - total_memory=2.56 MiB - chunk_duration=1.00s
write_binary_recording:   0%|                                                                  | 0/651 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "D:\sjwlab\bxy\spikeinterface-main\examples\get_started\quickstart.py", line 67, in <module>
    recording_preprocessed = recording_cmr.save(format='binary')
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\sjwlab\bxy\spikeinterface-main\examples\get_started\myenv\Lib\site-packages\spikeinterface\core\base.py", line 863, in save
    loaded_extractor = self.save_to_folder(**kwargs)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\sjwlab\bxy\spikeinterface-main\examples\get_started\myenv\Lib\site-packages\spikeinterface\core\base.py", line 970, in save_to_folder
    cached = self._save(folder=folder, verbose=verbose, **save_kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\sjwlab\bxy\spikeinterface-main\examples\get_started\myenv\Lib\site-packages\spikeinterface\core\baserecording.py", line 537, in _save
    write_binary_recording(self, file_paths=file_paths, dtype=dtype, verbose=verbose, **job_kwargs)
  File "D:\sjwlab\bxy\spikeinterface-main\examples\get_started\myenv\Lib\site-packages\spikeinterface\core\recording_tools.py", line 146, in write_binary_recording
    executor.run()
  File "D:\sjwlab\bxy\spikeinterface-main\examples\get_started\myenv\Lib\site-packages\spikeinterface\core\job_tools.py", line 405, in run
    res = self.func(segment_index, frame_start, frame_stop, worker_ctx)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\sjwlab\bxy\spikeinterface-main\examples\get_started\myenv\Lib\site-packages\spikeinterface\core\recording_tools.py", line 186, in _write_binary_chunk
    memmap_array[...] = traces
    ~~~~~~~~~~~~^^^^^
ValueError: could not broadcast input array from shape (20000,1024) into shape (20000,67)

alejoe91, 2024-08-20:
@bxy666666 Can you paste a minimal script to reproduce your issue? How are you loading the recording? What are you printing? These are all things that we need to know to help you out!

bxy666666, 2024-08-20:
> @bxy666666您能粘贴一个最小脚本来重现您的问题吗？您如何加载录音？您正在打印什么？这些都是我们需要知道的，以便帮助您！

Thanks for your reply! I tried to solve it through

 traces = recording.get_traces(
        start_frame=start_frame,
        end_frame=end_frame,
        segment_index=segment_index,
        channel_ids=channel_ids,  # 添加通道部分
        cast_unsigned=cast_unsigned
    )     I added the channel_ids,  this seems no warning. Now I am solving another question.   about time slice #3305  I am looking forward your reply!!!!!

alejoe91, 2024-08-20:
What was your initial function call? Anyways, if this problem is fixed, feel free to close the issue!

(if you're using SpikeInterface and would like to support us, can you also add a "Star" to the project?)

bxy666666, 2024-08-20:
> 您的初始函数调用是什么？无论如何，如果此问题已解决，请随时关闭该问题！
> 
> （如果您正在使用 SpikeInterface 并且愿意支持我们，您也可以为项目添加“星号”吗？）

OK,  I will close this problem.  I have already add this project!!

closed 2024-08-20 by bxy666666