SpikeInterface/spikeinterface #3317: How can I split sortings

BEGINRX, 2024-08-20:
After used `si.concatenate_recordings` to concatenate recordings and sorting with the concatenated recording, how can I split the sorting results? I have used `si.split_sorting`, and then I get the segmented sorting. But the result is not subscriptable. How can I get the sorting segments?

alejoe91, 2024-08-21:
If you want to access the spike trains form each segment, you can use:
```
spike_train = sorting.get_unit_spike_train(unit_id, segment=i)
```

You can also get a mono-segment sorting out of a multi segment one with this:
```
sorting_mono = si.select_segment_sorting(sorting_multi, segment_indices=[1])
```
In this example only the second segment will be returned

alejoe91, 2024-08-21:
@chrishalcrow you can pass a `sorting` and a `concatemated_recording` to the `split_sorting` function!

chrishalcrow, 2024-08-21:
@alejoe91 ahhh, very nice!

BEGINRX, 2024-08-22:
@alejoe91 Thank you very much! It worked! 
@chrishalcrow Thank you! But the appended recording used `si.append_recordings` can't do the preprocessing `estiment_motion`. So my order of processing is `si.concatenate_recordings`, `set_probe`, preprocess, `run_sorter`, `si.split_sorting`, `si.select_segment_sorting` for each segment.

closed 2024-08-22 by alejoe91

BEGINRX, 2024-08-22:
Now I have new question. After getting each segment, I need to do postprocess using `si.create_sorting_analyzer`. But the input is each sorting and responding preprocessed recording. And the concatenated recording has one segement. How can I continue?