SpikeInterface/spikeinterface #3320: Non default Sparsity kwargs when creating SortingAnalyzer

jonpedros, 2024-08-21:
I'd like to create a `SortingAnalyzer` object with sparsity computed from channel groups rather than the default setting. However, it seems sparsity arguments passed to `spikeinterface.core.create_sorting_analyzer()` these arguments are passed to `spikeinterface.core.estimate_sparsity()`, which does not handle such method, instead of to `spikeinterface.core.compute_sparsity()`. What would be the standard way to use parameters available to `compute_sparsity()` and use the obtained `ChannelSparsity` for the created `SortingAnalyzer`?

At any rate, even with the relatively in-depth documentation specific to explaining sparsity, the many ways to generate it seem redundant/confusing. It seems `compute_sparsity()` is the most full-fledged function for generating it, but the fact it needs `SortingAnalyzer` to do so means one would create `SortingAnalyzer` with `sparse=False`, generate sparsity and then create it again.

alejoe91, 2024-08-21:
Hi @jonpedros 

You're absolutely right! We should support also the `by_property` method! I'll make a PR to patch this!

alejoe91, 2024-08-21:
> Furthermore, it seems that default behavior is for the group property to be referred as `'group'` for recordings, but when a grouped sorting is created from a grouped recording, this key changes automatically to `'channel_group'`, so `compute_sparsity()` will return an error when trying the routine I described above.

Could you pinpoint where this happens?

jonpedros, 2024-08-22:
> Could you pinpoint where this happens?

Sure! You mean for the error when computing sparsity, or the point when the grouping key changes?

alejoe91, 2024-08-22:
The grouping key! Is it from Phy?

jonpedros, 2024-08-22:
> The grouping key! Is it from Phy?

Yes, sorting was curated in Phy and imported into SI, while the recording was loaded from the saved preprocessed folder generated before with SI.

alejoe91, 2024-08-22:
I see. Yeah phy renames the groups as `channel_group`. So you would need something like:
```
sorting.set_property("group", sorting.get_property("channel_group"))
```

jonpedros, 2024-08-23:
I agree! Should I close this one, though?

PR #3369 (2024-09-15): Extend estimate_sparsity, add from_amplitude and deprecate from_ptp
Fixes #3320 

This PR extends the methods supported by the `estimate_sparsity` function. In particular, the `by_property` method is also added and handled differently.

Note that the implementation of the `from_ptp` is deprecated, in favor of `from_snr(..., amplitude_mode="peak_to_peak")`.

A new `from_amplitude` sparsity option is added, to select channels based on uV amplitudes.
closed 2024-09-15 by alejoe91