SpikeInterface/spikeinterface #3326: si.create_sorting_analyzer for concatenate recording?

BEGINRX, 2024-08-22:
Now I have new question. After getting each segment, I need to do postprocess using `si.create_sorting_analyzer`. But the input is each sorting and responding preprocessed recording. And the concatenated recording has one segement. How can I continue?

_Originally posted by @BEGINRX in https://github.com/SpikeInterface/spikeinterface/issues/3317#issuecomment-2303958650_

alejoe91, 2024-08-22:
In that case you can use multi-segment recording and sorting objects.
```
recording_concat = si.concatenate_recording(recording_list)
recording_multi = si.append_recordings(recording_list)

...

sorting_multi = si.split_sorting(sorting, recording_concat)

# create analyzer and proceed with multi-segment objects
analyzer = si.create_sorting_analyzer(sorting_multi, recording_multi)
```

All analyzer operations support multiple segments, so I would recommend keeping them :)

BEGINRX, 2024-08-22:
But I think the input of `si.create_sorting_analyzer` should be sortings and preprocessed recordings. The appended recordings can't do the preprocessing `estimate_motion`.

alejoe91, 2024-08-22:
Ah I see! We will introduce multi-segment motion correction soon.

Are you only estimating or also correxting for motion?

BEGINRX, 2024-08-22:
I find this bug from `estimate_motion` maybe caused by `si.correct_motion`.

DaohanZhang, 2024-08-23:
I wonder in this case, units sorted in each segments will correspond or not? Will it be different from sorting from two recordings (As I know, If i stop one recording and start another, there will be different number of units)?

BEGINRX, 2024-08-24:
@DaohanZhang I think the sorting results in different time are different. Even for the same recording. But we also find when we use the sorting method "mountainsort5", the raster graph has some problem. The raster of one unit is segmented. I'm looking for the cause of the problem. Does this reply answer your question?

closed 2024-08-27 by BEGINRX