SpikeInterface/spikeinterface #3329: Sorting analyzer extensions don't know if they've been interrupted

jonahpearl, 2024-08-23:
Hi all — sometimes a sorting analyzer's computation of an extension is interrupted, either accidentally (i.e. a SLURM job times out), or manually (I want to change a parameter). Normally, the way one would deal with this, in a script, is to check if the extension has been completed with `get_extension`/`has_extension`, but these only check for the presence of the corresponding dir and the file `params.json` (see [here](https://github.com/SpikeInterface/spikeinterface/blob/39218069ca61529575abf70095d1ce7912a410d3/src/spikeinterface/core/sortinganalyzer.py#L1364)), and so if the computation has been interrupted partway through, the code would treat it as completed, but downstream analyses would fail / be wrong. Currently, for me, I'm manually deleting extension folders if this occurs and I need to fix it. Is there an existing better solution? Otherwise, I'd propose simply having a `is_finished` or `exit_status` flag in `info.json` that could be used for `has_extension`, rather than just checking that the folder / params are there. Thanks!

zm711, 2024-08-23:
If you run `get_extension('xx').get_data()` does it return None? That could be a check for a SLURM or similar failure. If it is not completed then I think `get_data` should return None. If it is not returning None then I guess the data would be partially filled in and then a flag of some sort would be good to have I suppose. This is an interesting failure point! Thanks for letting us know.

What do you think @alejoe91 ?

jonahpearl, 2024-08-23:
No but "load_data" appears to correctly return None. I tried to poke around in the code to understand this but got lost in function factories :p

[image]

zm711, 2024-08-23:
In this case you could make a `try-except` using the `get_data()` and if it doesn't exist  but `has_extension=True` then you delete the folder to recalculate. I think `load_data` and `get_data` are going to be similarly expensive. try is basically free so if the fail rate is low I think the try-except will be better. If you have a high fail rate than an if-else would be better and using `load_data` may be what you want.

> but got lost in function factories :p

I get lost there all the time :) That was a Sam special.

jonahpearl, 2024-08-23:
Yeah that sounds like the right solution in this case — it just isn't an intuitive pattern to write off the bat (expecting a key error from a call to get_data()). Also what about in the case where, e.g., the data file somehow gets corrupted — that would give some other error down the line that I would also have to write a custom try/except for. It just makes more sense to be testing for completeness in some consistent internal way when the analyzer is done / when `has_extension` is called. Not gonna be upset if this doesn't change cuz I know it's complex, but I just think `has_extension` should report whether the extension is usable, not whether a small internal file exists.

zm711, 2024-08-23:
I agree. I think that level design choice would go through Sam and Alessio. So I'm just suggesting  a work around until they decide how to deal with this. I've never seen this, but I do my analyses locally right now, so I don't think we realized that the supporting files would be done, but the actual computation would fail without the user necessarily coming to an error when they arrive. But I think having a completion flag works. We will see what they say.

alejoe91, 2024-08-27:
I actually think that adding an `exit_status` to the `info` is a better pattern. Note that this could be easily implemented in the base class with minimal code additions. @samuelgarcia what do you think?

samuelgarcia, 2024-08-27:
Yep.
exit_status and also run time maybe ?

PR #3347 (2024-09-11): Analyzer extension exit status
Fixes #3329

Here's a first pass at adding information about an extension's completion (or not) to its metadata.

This PR adds a `run_info.json` file to each extension's folder. This contains keys:
* `run_completed` (whether or not the call to `AnalyzerExtension._run()` finished),
* `data_loadable` (whether or not `AnalyzerExtension.load_data()` can be called without an error) 
* `runtime_s` (the runtime of `AnalyzerExtension._run()` in seconds). 

The core logic is implemented in `AnalyzerExtension.run()` and wraps the call to the extension-specific `_run()`. Th [... cut]
closed 2024-09-11 by samuelgarcia