SpikeInterface/spikeinterface #3330: Is there a way to import "phy" curation results into sorting/sorting analyzer?

borrepp, 2024-08-23:
Hello, 

I was wondering if there is a way to import the curation results made with "phy" into a sorting_analyzer. 

Thanks
Pepe

BEGINRX, 2024-08-24:
You can use 
`import spikeinterface.extractors as se`
and the function `se.read_phy`

alejoe91, 2024-08-24:
@BEGINRX the question is different: how to load as a sorting analyzer! 

@borrepp I think that with the sorting analyzer we could make such a function! It would load the sorting object, recording attributes, templates, and optionally PCA, but we would need to construct these extensions ad-hoc. I think it would be a great addition though

borrepp, 2024-08-25:
Thank you both for the comments. 

Hi @alejoe91, It will be great to have a function like that. In the meantime, I will use the extractor that @BEGINRX suggested and explore how much information was imported into the sorting object.

Given this information, can I ask for your advice?

We want to import from phy because we used spikeinterface for preprocessing, motion correction, and sorting. But for curation, the lab decided to use phy (by exporting the sorting analyzer). Now, I'm trying to get the final curation results back to spikeinterface because the sorting was done on concatenated recording sessions. Spikeinterface has the tools to split the sorting into the original sessions and keep all the metrics from each sorted cluster. The plan is to use neuroconv to export each sorting/sorting_analyzer into an NWB file.  

Does it make sense? Is this the best approach? 

Thanks in advance for your help
 Best
 Pepe

BEGINRX, 2024-08-26:
My solution is like this.
```
   sorting_curated_phy = se.read_phy(phy_folder, exclude_cluster_groups=["noise"])
    recording_saved = si.load_extractor(preprocessed_folder)

    rec_splited = si.split_recording(recording_saved)
    sorting_splited = si.split_sorting(sorting_curated_phy, rec_splited)

    for r in range(len(rec_splited)):
        sorting = si.select_segment_sorting(sorting_splited, segment_indices=r)
        sorting_analyzer = si.create_sorting_analyzer(sorting=sorting, recording=rec_splited[r])
```
Then I get every segment sorting analyzer. I also hope to get the  sorting analyzer from phy curated result. And if the sorting analyzer can be splitted, it will be better. 
@alejoe91 Thank you!

borrepp, 2024-08-26:
Thank you @BEGINRX ¡¡¡

closed 2024-08-26 by borrepp