SpikeInterface/spikeinterface #3332: Run_sorter speed between loops

RobertoDF, 2024-08-24:
Hi,
Something unexplainable happens when I sort two probes in this way:
```
split_preprocessed_recording = raw_rec.split_by("group")
for group, sub_rec in split_preprocessed_recording.items():
    sorting = run_sorter(sorter_name="kilosort4", 
        recording=sub_rec,
        folder=f"{path_recording_folder}/spike_interface_output/probe{group}",
        verbose=True,
        remove_existing_folder=True,
        torch_device="cuda" )
```
Raw_rec contains a two probes .3 hours long recording. Somehow the second loop is way slower than the first. This does **not** depend on the recording itself because if I sort the second probe alone it is as fast as usual.

first loop: 1.36it/s
```
INFO:kilosort.spikedetect:Re-computing universal templates from data.
100%|¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦5614/5614 [1:08:45<00:00,  1.36it/s]
```

second loop:10.59s/it
```
INFO:kilosort.spikedetect:Re-computing universal templates from data.
 48%|¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦¦   | 2683/5614 [7:53:24<8:37:10, 10.59s/it]
```

Should I add some garbage collection after each loop? clean some variables? torch.cuda.empty_cache()?
I will run a test with torch.cuda.empty_cache() within the loop now.

Thanks!!

zm711, 2024-08-24:
To know if this is at the spikeinterface level or something with the kilosort4 I would try doing the sorting with one of the other sorters and seeing if it always happens (an SI thing) or if it only happens with kilosort4 (more likely a cuda issue). Maybe we could find a way to tweak our KS4 script to try to help with this, but I think we need to diagnose where the problem is happening for you!

RobertoDF, 2024-08-27:
I am confident my problem was due to a known [WIndows / multiprocessing issue ](https://pytorch.org/docs/stable/notes/windows.html#multiprocessing-error-without-if-clause-protection) solved by using ` if __name__ == '__main__':` .

RobertoDF, 2024-08-27:
I am still puzzled by the fact that some kilosort steps ("first clustering" & "final clustering") are often way slower than expected but this is another issue.

closed 2024-08-27 by RobertoDF