SpikeInterface/spikeinterface #3333: Unable to read the meta files from SpikeGLX Release Version 20240129

shijiaaa, 2024-08-24:
When I ran a session collected by the most recent SpikeGLX (version 20240129), I encountered the following error:
`NotImplementedError: This meta file version of spikeglxis not implemented`

Files recorded from a previous SpikeGLX version (20230905) could be recognized just fine.

Could we update the pipeline to recognize the meta files of the most current SpikeGLX version? 

Much appreciated for your help!

alejoe91, 2024-08-27:
Thanks @shijiaaa 

This is a NEO problem. Could you open the same issue here: https://github.com/NeuralEnsemble/python-neo ?

alejoe91, 2024-08-27:
You should also try to upgrade your NEO installation, since `2013` seems supported: 
https://github.com/NeuralEnsemble/python-neo/blob/master/neo/rawio/spikeglxrawio.py#L522

shijiaaa, 2024-08-29:
Thanks. Upgrading the NEO installation solved the issue!

closed 2024-08-29 by shijiaaa