SpikeInterface/spikeinterface #3336: Spike sorting failed

bxy666666, 2024-08-25:
Starting container
Installing spikeinterface with github in container
Installing numpy==1.26 with pypi in container
Installing neo with pypi in container
Installing h5py with pypi in container
Running kilosort2 sorter inside spikeinterface/kilosort2-compiled-base
Stopping container
Traceback (most recent call last):
  File "D:\sjwlab\bxy\spikeinterface-main\examples\get_started\quickstart(2).py", line 158, in <module>
    sorting_KS2 = ss.run_sorter(sorter_name="kilosort2", recording=recording_preprocessed, extra_requirements=["numpy==1.26"],
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\sjwlab\bxy\spikeinterface-main\examples\get_started\myenv\Lib\site-packages\spikeinterface\sorters\runsorter.py", line 210, in run_sorter
    return run_sorter_container(
           ^^^^^^^^^^^^^^^^^^^^^
  File "D:\sjwlab\bxy\spikeinterface-main\examples\get_started\myenv\Lib\site-packages\spikeinterface\sorters\runsorter.py", line 663, in run_sorter_container
    raise SpikeSortingError(f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting in docker failed with the following error:
/sjwlab/bxy/spikeinterface-main/examples/get_started/in_container_sorter_script.py:23: DeprecationWarning: `output_folder` is deprecated and will be removed in version 0.103.0 Please use folder instead
  sorting = run_sorter_local(
RUNNING SHELL SCRIPT: /sjwlab/bxy/spikeinterface-main/examples/get_started/kilosort2_output/sorter_output/run_kilosort2.sh
Warning: X does not support locale C.UTF-8

Time   0s. Determining good channels..

found 91 threshold crossings in 125.00 seconds of data

found 30 bad channels

Time   4s. Computing whitening matrix..

Getting channel whitening matrix...

Channel-whitening matrix computed.

Time   4s. Loading raw data and applying filters...

Time   5s. Finished preprocessing 39 batches.

random seed for clusterSingleBatches: 1

Obtained 7 PC waveforms in 0.13 seconds

----------------------------------------Error using indexing

Assignment has more non-singleton rhs dimensions than non-singleton subscripts.
Error running kilosort2
Traceback (most recent call last):
  File "/sjwlab/bxy/spikeinterface-main/examples/get_started/in_container_sorter_script.py", line 23, in <module>
    sorting = run_sorter_local(
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py", line 276, in run_sorter_local
    SorterClass.run_from_folder(folder, raise_error, verbose)
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py", line 301, in run_from_folder
    raise SpikeSortingError(
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py", line 261, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/external/kilosortbase.py", line 217, in _run_from_folder
    raise Exception(f"{cls.sorter_name} returned a non-zero exit code")
Exception: kilosort2 returned a non-zero exit code

Spike sorting failed. You can inspect the runtime trace in /sjwlab/bxy/spikeinterface-main/examples/get_started/kilosort2_output/spikeinterface_log.json.

alejoe91, 2024-08-27:
Can you share your script and more details on the recoridng?

DaohanZhang, 2024-08-30:
I encountered the same bug and have seen this question reported before https://github.com/SpikeInterface/spikeinterface/issues/3204

bxy666666, 2024-08-30:
> I encountered the same bug and have seen this question reported before #3204

Thanks, I am running on Docker, you mean run KS2 locally?

DaohanZhang, 2024-08-30:
@bxy666666 Yes, I ran in docker using python codes, not locally using matlab.

bxy666666, 2024-08-30:
> @bxy666666 Yes, I ran in docker using python codes, not locally using matlab. But i do not use a imaging, I download the code locally. It should be the same.

Oh, I used the imaging, I will try to download KS2. Thanks again!

JoeZiminski, 2024-09-09:
I will close this for now, please feel free to reopen if problem is not solved!

closed 2024-09-09 by JoeZiminski