SpikeInterface/spikeinterface #3340: loading matlab into SpikeInterface

ajcheng1, 2024-08-26:
I am following this guide to use spikeinterface with matlab data: [https://spikeinterface.readthedocs.io/en/stable/how_to/load_matlab_data.html#exporting-data-from-matlab](url)
I write the matlab data into a bin file 
[image]
I then put the bin file into my python folder 

[image]

During pre-processing, I get the downstream error
[image]

I don't think the code is wrong, but I am also not sure what's wrong with my data file. Any advice?

chrishalcrow, 2024-08-27:
Hi @ajcheng1 , I don't have much MATLAB experience, but I think you're missing the step when you read the data in. At the moment your `data` variable is just your filepath. You should have something like:
```
data_path = 'C:\Users\blahblah'
data = importdata(data_path)
```

At the moment it looks like your `your_data_as_a_binary.bin` file is empty. You can check how big it is in your File Explorer. I think it will be very small. Once you update the code, it will be about the same size of your raw data file.

Let me know if that works :)

alejoe91, 2024-08-27:
Good catch @chrishalcrow 

I think you just need a `load`:
```matlab
data = load("C:/..../unsorted.mat")
```

ajcheng1, 2024-08-27:
yep haha, I figured it out. I have another issue with the file, but I'll just create a new issue for it. Thanks!

closed 2024-08-27 by ajcheng1