SpikeInterface/spikeinterface #3343: Issue with Spike Sorting Data from matlab file

ajcheng1, 2024-08-27:
This is a bit of a continuation of #3340. I was able to successfully load the matlab file into .bin using this simple two line code into matlab and convert it to bin. 
[image]

Loading data into spikeinterface seems to work properly 
[image]

Pre-processsing runs fine 
[image]

But when I sort, I get this error for three different spike sorter tools: 
[image]

1. Mountain sort5 - `numpy.linalg.LinAlgError: SVD did not converge`
2. Spyking Circus & tridesclous2 - `ValueError: Digital filter critical frequencies must be 0 < Wn < fs/2 (fs=12500.0 -> fs/2=6250.0) Spike sorting failed. You can inspect the runtime trace in C:\Users\Shado\SHJ\spykingcircus2_output/spikeinterface_log.json`.

The error message is lengthy, but I can add it here if needed. Is there any advice on this? I can also share the matlab file.

chrishalcrow, 2024-08-28:
Hey @ajcheng1 , if there are problems with several spike sorters, there might be something wrong with the data (probably something silly - it always is!). I'd recommend taking a look at your data and seeing if it looks like you'd expect.

You can use widgets (https://spikeinterface.readthedocs.io/en/latest/modules/widgets.html) to visualise; and `plot_traces` to look at your raw/preprocessed data. There's some more explanation/examples in the `overview_tutorial` which was done at a recent workshop in Edinburgh: https://github.com/SpikeInterface/SpikeInterface-Training-Edinburgh-May24 .

zm711, 2024-08-29:
I would say the problem is that this is ~3 seconds long.

37149/12500 = 2.97 seconds.

I doubt any sorter is going to be able to do much with 3 seconds of data.

zm711, 2024-08-29:
Are you sure that this data should have 384 channels? Is it a neuropixel dataset? Maybe this is something like tetrode data which would by ~5 minutes or even a monotrode dataset which would be ~20 minutes. Neither very long or super easy to sort with. I think @chrishalcrow is right that you likely need to explore your raw data more so that we have a better idea of what you are actually working with.

ajcheng1, 2024-08-29:
Hi @zm711, good insight. It's MEA data that I exported from plexon. I am looking at the matlab files and it contains a bunch of arrays from each channel. 
[image]
and it seems to include the channel number within the matrix and a column of 0 to separate it from the rest of the data point. 
[image]

let me do some more digging into it, but I appreciate the pointers. I believe it's roughly 3 seconds because plexon has already divided up the recording into 3 second intervals of waveforms.

zm711, 2024-08-29:
No sorter that I know works with waveform snippets any more (legacy ones do), so you need to get the raw data from Plexon rather than the waveform snippet data. @h-mayorquin since you've thought the most about this recent for plexon what should this person be looking for to get the raw data rather than waveform snippets?

zm711, 2024-08-29:
As a side note @ajcheng1 we have a plexon reader ie 

```python
import spikeinterface.extractors as se
recording = se.read_plexon('put_plexon_file_here.plx')
```

Do you not have access to the raw data any more? I'm worried that someone may have just left you the snippet data which might not serve you too well.

h-mayorquin, 2024-08-29:
Which plexon data is that? Is it plexon2 or plexon1? Is it wideband?
I would just run the sorting directly in the plexon recording as @zm711 suggested.

ajcheng1, 2024-08-29:
It's plexon1. It's related to the issue of plexon containing different streams. 
[image]

zm711, 2024-08-29:
This seems like an issue with our extractor @h-mayorquin ? Maybe we need to fix that unless you think a stream_id needs to be specified for this reader. Could you try:

```python
from neo.rawio import PlexonRawIO

reader = PlexonRawIO(file_path)
reader.parse_header()
```

and see if that has any errors.

h-mayorquin, 2024-08-29:
Can you update neo to its latest version?  I just fixed this and this was included in a release.

stream_name = "WB-Wideband"
Just do `read_plexon(file_path, stream_name=stream_name)`

ajcheng1, 2024-08-29:
Just to confirm, is it python neo 0.13.3? 
[image]

I get this error in which the file does not have wideband for some reason. 
[image]

h-mayorquin, 2024-08-29:
Yes, for some reason your recording does not have streams. 

Can you run the code that @zm711 suggested and print the header:

```python
from neo.rawio import PlexonRawIO

reader = PlexonRawIO(file_path)
reader.parse_header()
print(reader.header)
```

ajcheng1, 2024-08-29:
A bit messy to look at so I took a screenshot as well. I believe my file only has "SPK channels" and not wideband channels though not entirely sure. 
[image]

`Parsing event channels: 100%|██████████| 1/1 [00:00<00:00, 7463.17it/s]{'nb_block': 1, 'nb_segment': [1], 'signal_streams': array([], dtype=[('name', '<U64'), ('id', '<U64')]), 'signal_channels': array([],
      dtype=[('name', '<U64'), ('id', '<U64'), ('sampling_rate', '<f8'), ('dtype', '<U16'), ('units', '<U64'), ('gain', '<f8'), ('offset', '<f8'), ('stream_id', '<U64')]), 'spike_channels': array([('A1_11', 'ch1#0', '', 9.25663358e-07, 0., -1, 12500.),
       ('A1_12', 'ch2#0', '', 1.39436687e-06, 0., -1, 12500.),
       ('A1_13', 'ch3#0', '', 1.71408549e-06, 0., -1, 12500.),
       ('A1_14', 'ch4#0', '', 6.14320070e-07, 0., -1, 12500.),
       ('A1_21', 'ch5#0', '', 9.15618906e-07, 0., -1, 12500.),
       ('A1_22', 'ch6#0', '', 1.56677165e-06, 0., -1, 12500.),
       ('A1_23', 'ch7#0', '', 3.11344555e-07, 0., -1, 12500.),
       ('A1_24', 'ch8#0', '', 2.06558072e-06, 0., -1, 12500.),
       ('A1_31', 'ch9#0', '', 8.10165341e-07, 0., -1, 12500.),
       ('A1_32', 'ch10#0', '', 1.29558812e-06, 0., -1, 12500.),
       ('A1_33', 'ch11#0', '', 1.30061277e-06, 0., -1, 12500.),
       ('A1_34', 'ch12#0', '', 5.44017674e-07, 0., -1, 12500.),
       ('A1_41', 'ch13#0', '', 4.56973108e-07, 0., -1, 12500.),
       ('A1_42', 'ch14#0', '', 1.13155192e-06, 0., -1, 12500.),
       ('A1_43', 'ch15#0', '', 6.74580633e-07, 0., -1, 12500.),
       ('A1_44', 'ch16#0', '', 1.71910647e-06, 0., -1, 12500.),
       ('A2_11', 'ch17#0', '', 3.19714254e-07, 0., -1, 12500.),
       ('A2_12', 'ch18#0', '', 9.15618906e-07, 0., -1, 12500.),
       ('A2_13', 'ch19#0', '', 1.42783429e-06, 0., -1, 12500.),
       ('A2_14', 'ch20#0', '', 6.37753714e-07, 0., -1, 12500.),
       ('A2_21', 'ch21#0', '', 7.66644903e-07, 0., -1, 12500.),
       ('A2_22', 'ch22#0', '', 1.02442357e-06, 0., -1, 12500.),
       ('A2_23', 'ch23#0', '', 5.03842597e-07, 0., -1, 12500.),
       ('A2_24', 'ch24#0', '', 2.49410842e-07, 0., -1, 12500.),
       ('A2_31', 'ch25#0', '', 6.37753714e-07, 0., -1, 12500.),
       ('A2_32', 'ch26#0', '', 6.05952348e-07, 0., -1, 12500.),
       ('A2_33', 'ch27#0', '', 9.13946217e-07, 0., -1, 12500.),
       ('A2_34', 'ch28#0', '', 2.72845418e-07, 0., -1, 12500.),
       ('A2_41', 'ch29#0', '', 5.54059153e-07, 0., -1, 12500.),
       ('A2_42', 'ch30#0', '', 1.00099204e-06, 0., -1, 12500.),
       ('A2_43', 'ch31#0', '', 8.21882098e-07, 0., -1, 12500.),
       ('A2_44', 'ch32#0', '', 1.77934687e-06, 0., -1, 12500.),
       ('A3_11', 'ch33#0', '', 1.11314375e-06, 0., -1, 12500.),
       ('A3_12', 'ch34#0', '', 1.16001133e-06, 0., -1, 12500.),
       ('A3_13', 'ch35#0', '', 3.71605158e-07, 0., -1, 12500.),
       ('A3_14', 'ch36#0', '', 1.02275274e-06, 0., -1, 12500.),
       ('A3_21', 'ch37#0', '', 3.02975172e-07, 0., -1, 12500.),
       ('A3_22', 'ch38#0', '', 5.47364504e-07, 0., -1, 12500.),
       ('A3_23', 'ch39#0', '', 8.28576523e-07, 0., -1, 12500.),
       ('A3_24', 'ch40#0', '', 3.18040520e-07, 0., -1, 12500.),
       ('A3_31', 'ch41#0', '', 7.83385823e-07, 0., -1, 12500.),
       ('A3_32', 'ch42#0', '', 7.06387266e-07, 0., -1, 12500.),
       ('A3_33', 'ch43#0', '', 4.23496456e-07, 0., -1, 12500.),
       ('A3_34', 'ch44#0', '', 1.04451443e-06, 0., -1, 12500.),
       ('A3_41', 'ch45#0', '', 2.56106609e-07, 0., -1, 12500.),
       ('A3_42', 'ch46#0', '', 7.88404932e-07, 0., -1, 12500.),
       ('A3_43', 'ch47#0', '', 2.42715400e-07, 0., -1, 12500.),
       ('A3_44', 'ch48#0', '', 9.20646136e-07, 0., -1, 12500.),
       ('A4_11', 'ch49#0', '', 2.17941188e-06, 0., -1, 12500.),
       ('A4_12', 'ch50#0', '', 9.03903149e-07, 0., -1, 12500.),
       ('A4_13', 'ch51#0', '', 4.31872892e-06, 0., -1, 12500.),
       ('A4_14', 'ch52#0', '', 8.05142330e-07, 0., -1, 12500.),
       ('A4_21', 'ch53#0', '', 1.23868889e-06, 0., -1, 12500.),
       ('A4_22', 'ch54#0', '', 1.27551631e-06, 0., -1, 12500.),
       ('A4_23', 'ch55#0', '', 1.55337362e-06, 0., -1, 12500.),
       ('A4_24', 'ch56#0', '', 1.83626969e-06, 0., -1, 12500.),
       ('A4_31', 'ch57#0', '', 5.84191469e-07, 0., -1, 12500.),
       ('A4_32', 'ch58#0', '', 5.23930906e-07, 0., -1, 12500.),
       ('A4_33', 'ch59#0', '', 6.66211146e-07, 0., -1, 12500.),
       ('A4_34', 'ch60#0', '', 1.34749326e-06, 0., -1, 12500.),
       ('A4_41', 'ch61#0', '', 2.84562987e-07, 0., -1, 12500.),
       ('A4_42', 'ch62#0', '', 7.43213333e-07, 0., -1, 12500.),
       ('A4_43', 'ch63#0', '', 6.00928996e-07, 0., -1, 12500.),
       ('A4_44', 'ch64#0', '', 1.27551631e-06, 0., -1, 12500.),
       ('A5_11', 'ch65#0', '', 6.81277045e-07, 0., -1, 12500.),
       ('A5_12', 'ch66#0', '', 1.33577575e-06, 0., -1, 12500.),
[... 295 lines cut ...]
       ('D5_32', 'ch362#0', '', 7.04712576e-07, 0., -1, 12500.),
       ('D5_33', 'ch363#0', '', 6.92993327e-07, 0., -1, 12500.),
       ('D5_34', 'ch364#0', '', 7.34843921e-07, 0., -1, 12500.),
       ('D5_41', 'ch365#0', '', 7.04712576e-07, 0., -1, 12500.),
       ('D5_42', 'ch366#0', '', 9.75885886e-07, 0., -1, 12500.),
       ('D5_43', 'ch367#0', '', 1.06961626e-06, 0., -1, 12500.),
       ('D5_44', 'ch368#0', '', 1.04283687e-06, 0., -1, 12500.),
       ('D6_11', 'ch369#0', '', 1.89318916e-06, 0., -1, 12500.),
       ('D6_12', 'ch370#0', '', 1.66386912e-06, 0., -1, 12500.),
       ('D6_13', 'ch371#0', '', 9.32365871e-07, 0., -1, 12500.),
[... cut ...]
       ('D6_41', 'ch381#0', '', 1.69567222e-06, 0., -1, 12500.),
       ('D6_42', 'ch382#0', '', 8.43648492e-07, 0., -1, 12500.),
       ('D6_43', 'ch383#0', '', 8.05142330e-07, 0., -1, 12500.),
       ('D6_44', 'ch384#0', '', 3.84997264e-07, 0., -1, 12500.)],
      dtype=[('name', '<U64'), ('id', '<U64'), ('wf_units', '<U64'), ('wf_gain', '<f8'), ('wf_offset', '<f8'), ('wf_left_sweep', '<i8'), ('wf_sampling_rate', '<f8')]), 'event_channels': array([('WellStatus', '1', b'event')],
      dtype=[('name', '<U64'), ('id', '<U64'), ('type', 'S5')])}`

zm711, 2024-08-29:
It looks to me like this dataset just has waveform snippets in it? I think plexon was trying to do some realtime sorting in the past. I don't think this dataset has raw data any more. When was it recorded?

ajcheng1, 2024-08-29:
Hmm, this may be from Axion end then. This was recorded from an Axion MEA plate that was convert to a .nex file (from .spk), Within Plexon Offline Sorter, you can export the .nex file into .plx.

zm711, 2024-08-29:
We can read `.nex` files. You want to try:

```python
from neo.rawio import NeuroExplorerRawIO

reader = NeuroExplorerRawIO('path_to_nex_file')
reader.parse_header()
print(reader.header)
```

then we can see if the plexon conversion was a problem or the initial conversion.

ajcheng1, 2024-08-29:
Side issue with the code involving the parse_reader() 
[image]

zm711, 2024-08-29:
Unfortunately we haven't touched that rawio for a quite a while so I was hoping it would still work. We need to update it....

Okay and does your Axion system have settings for just waveforms versus wideband recording? Or maybe you're converter to `.nex` got rid of the wideband. What's the axion workflow?

ajcheng1, 2024-08-29:
Okay, no problem. 
Here is the workflow we are using. This is the guide that can explain it better than I can.  https://www.axionbiosystems.com/sites/default/files/resources/Spike%20Sorting%20Protocol%20-%20Axion%20BioSystems.pdf 
The MEA system creates an Axion .raw file and a .spk file. (I believe .raw can be converted into .csv as well.) To think about it now, I believe converting the data into .nex does break-up the continuous data. Let me do some more digging.

zm711, 2024-08-29:
`.spk` is often shorthand for just using the waveform snippets. `.raw` is often just a binary file. So it might be better to just load the binary file itself. From the script above you can see the info we need: the channel number the number of samples. So if you have some way to get that info you could just load the `.raw` file itself!

ajcheng1, 2024-08-29:
Gotcha. Do u suggest I try spikeinterface.core.read_binary () ?

zm711, 2024-08-30:
You can try that but you will need sampling frequency, channels,, and dtype. So as long as you can look that up from your recording equipment or have it written down you should be good to give it a try. We also need to know if axion provides a header or not... I haven't read the axion link you sent yet, but if it does provide a header than we might need to use a different reader, but I would say give it a try. Look at the data and if it seems wrong or throws and error it is likely because the file has a header attached that we need to deal with....

zm711, 2024-09-04:
@samuelgarcia do you have any experience with Axion equipment. I was trying to look at this and it seems like we don't have anything at Neo. This might be a good issue to open on a neo and we could get a reader in general.

samuelgarcia, 2024-09-04:
I think the "0xb0" error is a wrong codec for decoding "°". should be easy to fix but we would need the file.

samuelgarcia, 2024-09-04:
i am not familiar with axion.

zm711, 2024-09-04:
If that's the case then @ajcheng1 would you be willing to open an issue on Neo https://github.com/NeuralEnsemble/python-neo/issues

and share a file with us. Then we can implement a reader for you on the Neo side and this will all happen under the spikeinterface hood for you!

h-mayorquin, 2024-09-04:
He shared a file with me. I can open the issue.

zm711, 2024-12-03:
Let's lose this one. @h-mayorquin if you want to open a Neo issue for the format feel free, but I don't have time so if users post on Neo we will deal with it, otherwise back burner.

closed 2024-12-03 by zm711