SpikeInterface/spikeinterface #3348: Stray .nfs file preventing deletion of waveforms extension folder

jonahpearl, 2024-08-28:
Hi all — I'm on a Linux / NFS system (CentOS 7 specifically, on my school's HPCC), and I encounter an OSError every time I try to re-compute the waveforms extension of a sorting analyzer. The core problem is that there's a stray .nfs temporary file hanging out in the waveforms folder. If I do `ls -a [that folder]` I can see it clearly, it's always something like `.nfs9eb83351caf602ec00005734`. 

I can see that it's python using the file with `lsof`:
```
COMMAND   PID    USER   FD   TYPE DEVICE  SIZE/OFF                 NODE NAME
python  30135 jop9552  mem    REG   0,45 128896688 11436947680097862380 [...]/analyzer_folder/extensions/waveforms/.nfs9eb83351caf602ec00005734
python  30135 jop9552   93u   REG   0,45 128896688 11436947680097862380 [...]/analyzer_folder/extensions/waveforms/.nfs9eb83351caf602ec00005734
```
and if I call `kill` on that process, my notebook's kernel crashes, so I'm pretty sure it's coming from spikeinterface.

The offending function is at `/spikeinterface/core/sortinganalyzer.py:2090`: `shutil.rmtree(extension_folder)`.

### MRE:
```python
recording = si.load_extractor([any recording])
sorting = si.read_sorter_folder([any sorting])
analyzer = si.create_sorting_analyzer(
    sorting=sorting, 
    recording=recording, 
    format='binary_folder', 
    folder="./tmp_analyzer",
)
to_compute = ["random_spikes", "waveforms"]
analyzer.compute(to_compute)
analyzer.compute("waveforms")  # tries to reset folder before recomputing but fails at deletion
```

Not sure how to go about debugging this but any advice is welcome. This is v0.101.0. Thanks!

zm711, 2024-08-28:
What is the `OSError` that is actually given :) I don't see it in the issue.

h-mayorquin, 2024-08-28:
It is probably in the recording and sorting that keeps a reference file handle. Probably related to:

https://github.com/SpikeInterface/spikeinterface/issues/3295

jonahpearl, 2024-08-28:
> What is the `OSError` that is actually given :) I don't see it in the issue.

Added the traceback :)

jonahpearl, 2024-08-28:
Also the same issue happens if I load sorting output from mountainsort5, so I don't think it's related to KS4 as being discussed in #3295.

zm711, 2024-08-28:
What is the actually context of this use (ie I could imagine recomputing waveforms later, but why do compute(waveforms) compute(waveforms) twice in a row?)? Does the error still happen if you do:

```python
analyzer.compute(['random_spikes', 'waveforms', 'templates'])
```
The waveforms file could be left open I guess during writing, so I'm wondering if computing another extensions right after waveforms ensures the file is closed.

zm711, 2024-08-28:
Looking here 
https://github.com/SpikeInterface/spikeinterface/blob/007b6efd0e675e60658199112d0541fba2b432cd/src/spikeinterface/core/waveform_tools.py#L194C45-L194C53

it seems that the memmap for writing the waveforms is open but I haven't found an explicit close yet? But I think @samuelgarcia or @alejoe91 know this part of the code base the best to know if the close is located somewhere else. But maybe this is causing your specific problem.

jonahpearl, 2024-08-28:
The context is, I'm developing code about the analyzer extensions and sometimes just end up running the same code twice :p The MRE just reproduces it for me, it isn't necessarily how one would come across it in the wild. And no, I've encountered this error with a large list of extensions being computed after waveforms.

If I compute the extensions once, then restart the kernel and reload the analyzer, I can get away with an initial re-compute of waveforms with no errors. But then on the second try of a given kernel there's usually an error, although sometimes it also takes until the third run (??).

zm711, 2024-08-28:
Would you be willing to try the same with zarr. the `binary_folder` backend uses the memmap whereas zarr is using the `shared_memory`. It might be worth testing. If zarr works with no problems then at least from my perspective this would point more strongly to the `waveforms.npy` file not being closed.

jonahpearl, 2024-08-28:
Yup, you're exactly right. It works as expected with zarr, no OSError.

zm711, 2024-08-28:
Okay, then I think we would need to close the memmap to fix that, but I think the general idea would be we create the memmap and leave it open so you can use the waveforms for other extensions without needing to reopen it./reload the data So I'm not sure if/when we could do that. We will just have to wait to see what Alessio and Sam think for this.

samuelgarcia, 2024-08-29:
Thanls for reporting and investigating.
I will have a look into this.

zm711, 2024-12-03:
I'll ping @samuelgarcia again so we can think about this for the memmap.

samuelgarcia, 2025-01-06:
Thanks for ping.
I put this again in my todo list.