SpikeInterface/spikeinterface #3353: Unable to sort from container

ebach23, 2024-08-29:
I have been trying to use sorters within the container (tried a few different sorters). I keep getting the same error message that docker is not installed despite docker being installed. Can you help me understand? TIA

[image]

alejoe91, 2024-08-29:
How did you install docker? Can you try a `docker run hello-world` from terminal?

DaohanZhang, 2024-08-30:
I have encountered the same question, maybe it is the problem of jupyter notebook (I m not sure). Try using python script in terminal after this command:
````
newgrp docker
````

JoeZiminski, 2024-09-09:
Are you running on linux or macOS / Windows?

zm711, 2025-02-21:
Since we haven't received responses from either people in the thread I'll close as stale. Feel free to open new issues if anything comes up!

closed 2025-02-21 by zm711