SpikeInterface/spikeinterface #3355: 'remove_excess_spikes': 'ValueError: need at least one array to concatenate'

shijiaaa, 2024-08-29:
Hello,

When I run 'remove_excess_spikes', I get 'ValueError: need at least one array to concatenate'. The 'rec' and 'sorting' objects were not empty (see attached screenshot).

This problem happened only recently, on Probe types NP2010 and NP2014 and multiple versions of SGLX.

I also just updated NEO and probeinterface installations (because of another issue #3333) which did not help with anything.

Any thoughts are much appreciated!

[image]

zm711, 2024-08-29:
I'm not sure off the top of my head, but your Traceback is cutoff. could you post the full traceback? 

Also the first easy check is: you've confirmed that this recording and sorting go together? 

Which KS were you using? 

And finally could you do:

```
spike_vector = sorting.to_spike_vector()
```
and let us know how many spikes your data actually has?

zm711, 2024-08-29:
One last question. Which version of spikeinterface are you using? It looks like your `has_exceeding_spikes` is before a refactor. So it might be worth a quick update to 0.101.0 or to main/source just to make sure this wasn't a bug that got fixed.

shijiaaa, 2024-08-29:
I was using 0.100.dev0. Updating to 0.101.0 did solve my problem. Thanks!!

closed 2024-08-29 by shijiaaa