SpikeInterface/spikeinterface #3357: Get motion correction info from saved binary preprocessed recording file in the past

atombysx, 2024-08-30:
Hi,

I've run 'non-rigid' motion correction on my recordings and saved the corrected recordings as binary files before spike sorting with kilosort 10 months ago before motion information output functionality was out.

Now I want to check the motion correction information from these recordings. Is it possible to compute/plot the drifts and other information from the saved binary recordings without re-running the motion correction again (which takes a long time)?

samuelgarcia, 2024-09-02:
Hi.
With release 101, the way we store the motion has changed a bit.
We now have an explicit motion object which is also saved.

You should be able to reconstruct this motion object by loading `temporal_bins.npy`, `spatial_bins.npy`, `motion.npy` using `np.load()` and then reconstruct a Motion object.

atombysx, 2024-09-02:
Ah ok,
I meant if I could reverse engineer the motion estimates from loading the original recording and motion-corrected recording bin files. I guess it's not that possible...

alejoe91, 2024-09-05:
> Ah ok, I meant if I could reverse engineer the motion estimates from loading the original recording and motion-corrected recording bin files. I guess it's not that possible...

Yeah I think that's not possible/straightforward. Do you have a `provenance.json` in the folder?

In that case, you should be able to di it!

atombysx, 2024-09-05:
> Yeah I think that's not possible/straightforward. Do you have a `provenance.json` in the folder?
>
> In that case, you should be able to di it!

Yeah I do have them and they do have motion part in them. But it's probably too difficult to do as I don't see any high level API can manage. We're rerunning motion correction and save the motion info now. Thanks!

closed 2024-09-05 by atombysx