SpikeInterface/spikeinterface #3366: Open the sorting reasult by phy

BEGINRX, 2024-09-05:
When I finish sorting, try to fix manually. But there are some problem.
`phy template-gui params.py --debug`

````
14:07:20.546 [D] __init__:68          Start capturing exceptions.
14:07:20.553 [D] model:619            Loading spike clusters.
14:07:20.556 [D] model:569            No channel shank file found.
14:07:20.556 [D] model:692            Loading templates.
14:07:20.559 [D] model:720            Templates are sparse.
14:07:20.559 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
14:07:20.560 [D] model:730            Loading the whitening matrix.
14:07:20.560 [D] model:434            Whitening matrix file not found.
14:07:20.560 [D] model:737            Loading the inverse of the whitening matrix.
14:07:20.562 [D] model:766            Loading features.
14:07:20.569 [E] __init__:62          An error has occurred (RuntimeError): ndarray subclass __array_wrap__ method returned an object which was not an instance of an ndarray subclass
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\admin\miniconda3\envs\phy2\Scripts\phy.exe\__main__.py", line 7, in <module>
    sys.exit(phycli())
             ^^^^^^^^
  File "C:\Users\admin\miniconda3\envs\phy2\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\admin\miniconda3\envs\phy2\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\admin\miniconda3\envs\phy2\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\admin\miniconda3\envs\phy2\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\admin\miniconda3\envs\phy2\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\admin\miniconda3\envs\phy2\Lib\site-packages\click\decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\admin\miniconda3\envs\phy2\Lib\site-packages\phy\apps\__init__.py", line 159, in cli_template_gui
    template_gui(params_path, **kwargs)
  File "C:\Users\admin\miniconda3\envs\phy2\Lib\site-packages\phy\apps\template\gui.py", line 209, in template_gui
    model = load_model(params_path)
            ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\admin\miniconda3\envs\phy2\Lib\site-packages\phylib\io\model.py", line 1433, in load_model
    return TemplateModel(**get_template_params(params_path))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\admin\miniconda3\envs\phy2\Lib\site-packages\phylib\io\model.py", line 339, in __init__
    self._load_data()
  File "C:\Users\admin\miniconda3\envs\phy2\Lib\site-packages\phylib\io\model.py", line 460, in _load_data
    self.sparse_features = self._load_features()
                           ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\admin\miniconda3\envs\phy2\Lib\site-packages\phylib\io\model.py", line 780, in _load_features
    cols = self._read_array(self._find_path('pc_feature_ind.npy'), mmap_mode='r')
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\admin\miniconda3\envs\phy2\Lib\site-packages\phylib\io\model.py", line 488, in _read_array
    return read_array(path, mmap_mode=mmap_mode).squeeze()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: ndarray subclass __array_wrap__ method returned an object which was not an instance of an ndarray subclass

QWidget: Must construct a QApplication before a QWidget
````

But other results can be imported into phy successfully. I find the number of cluster is one. Does the single cluster affect the process?

zm711, 2024-09-05:
> But other results can be imported into phy successfully. I find the number of cluster is one. Does the single cluster affect the process?

By this you mean you've loaded other exported datasets into phy? I think I'm trying to make sure whether this is a phy issue or a spikeinterface export issue. Just to doublecheck what version of numpy do you have in this environment?

BEGINRX, 2024-09-06:
Thanks for your reply. Yes, other exported datasets can be loaded. I think the version of numpy is available. I can't tell if the problem is caused by spikeinterface or phy. I will ask phy's team. Thank you.

zm711, 2024-09-06:
I meant if you do conda list or pip list what version of numpy do you have.

JoeZiminski, 2024-09-09:
Will close as moved to [here](https://github.com/SpikeInterface/spikeinterface/issues/3366) on the Phy repo

closed 2024-09-09 by JoeZiminski