SpikeInterface/spikeinterface #3375: using read_intan with timestamp gaps

nadine-saluti, 2024-09-06:
Hi,

I have two .rhd recordings with small timestamp gaps, so I cannot load them using the function read_intan. 

The gaps are tiny, so it's not an issue for me. Also, I have already sorted them with Kilosort using this function to load them in spike interface. I don't remember which version I had, but I have recently updated to version 0.100.8. Since the update, I have been unable to load these two recordings. In the documentation it seems like there should be a parameter called "ignore_integrity_checks", but I cannot use it.

zm711, 2024-09-06:
Hey @nadine-saluti,

yes that was a bit of debate to add that, but @h-mayorquin convinced me! What version of Neo do you have. I also can't remember when we added that to spikeinterface. 0.100.8 will be the latest before the sortinganalyzer so I understand not wanting to change that. So the first thing to try is to update your Neo install. Could you try that and let us know if `ignore_integrity_checks` works then? 

Also Heberto and I are constantly curious what your exact intern set up is. Do you use the standard one .rhd file or one of the other formats that includes .dat files. And are you splitting every minute or doing longer splits. He and I are trying to improve the reader for edge cases so knowing how you use intan could help us improve things for everyone!

h-mayorquin, 2024-09-06:
You probably do not have the latest version of python-neo because we added information on the trace on how to do this.

Update spikeitnerface and neo:

```python
from spikeinterface.extractor import read_intan

recording = read_intan(file_path=file_paht, ignore_integrity_checks=True)
```

nadine-saluti, 2024-09-06:
I use the standard .rhd file and don't split the recording.

By the way, updating Neo didn't help. 
Here is a screenshot:
[image]

zm711, 2024-09-06:
The issue is that that will force a switch to the SortingAnalyzer. We have done some docs to help with the transition but seems like you'll need to update to get this feature for the timestamps. I'll link the docs [here] (https://spikeinterface.readthedocs.io/en/stable/tutorials/waveform_extractor_to_sorting_analyzer.html)

nadine-saluti, 2024-09-08:
If I load the curated data using the read_phy function, will I be able to extract the waveforms with the sorting analyzer, even if the data was saved before using the waveform extractor?

I use this line of code to extract the waveforms:
si.extract_waveforms(recording=rec_filtered, sorting=phy_sorting, folder=wvf_folder, overwrite=True)

Can I convert it to this (after the update):
# Create sorting analyzer:
sorting_analyzer = si.create_sorting_analyzer(sorting=phy_sorting, recording= rec_filtered, folder=wvf_folder)
# Compute extensions:
sorting_analyzer.compute("random_spikes")
sorting_analyzer.compute("waveforms")

zm711, 2024-09-08:
Hey @nadine-saluti,

So you have two options. If you've already done the waveforms then we have a conversion process so you'll do:

```python
 waveform_folder_path = "path_to/my_waveform_extractor_folder"
new_sorting_analyzer_path = "path_to/my_new_sorting_analyzer_folder"
# On Windows
# new_sorting_analyzer_path = r"path_to\my_new_sorting_analyzer_folder"
wvf_extractor = si.load_waveform(folder=waveform_folder_path)
sorting_analyzer = wvf_extractor.sorting_analyzer
sorting_analyzer.save_as(folder=new_sorting_analyzer_path, format="binary_folder")
```

That way all of the work you've already done isn't lost. If this is a brand new dataset you want to analyzer then you go straight to the `sorting_analyzer` instead so:

```python
analyzer = si.create_sorting_analyzer(sorting=phy_sorting, recording=rec_filtered, format='binary_folder', folder=wvf_folder)
```
Note you're missing one thing. If you don't give a format then you create an in memory analyzer that is not saved. If you give format binary_folder or zarr it automatically saves everything it does to the folder you specify. 

binary_folder is just a series of .npy files and metadata, zarr is a format that allows for compression to save space. So you can choose either format to save to disk.

Does that make sense?

nadine-saluti, 2024-09-08:
It's not a new dataset, but I do want to re-extract the waveforms so that I have them only for the clusters I labeled as good (which are the ones I export with read_phy).
So, I guess the second option should work.

zm711, 2024-09-08:
Cool. Try it out and we are happy to help with the transition process, but in general it is smoother and faster to work with the analyzer. And like you saw you can change on computations. So something like:
```python
analyzer.compute(['random_spikes', 'waveforms', 'templates', 'spike_locations', 'unit_locations'])
``` 
etc. and it will sort out things that need to go to the raw data and work on them together. But since there are a couple changes you might have a couple hiccups. Just post here and we will work through them!

h-mayorquin, 2024-09-09:
Hi, @nadine-saluti, it seems that you were able to load the data even with the timestamp gaps, is that correct? 

I am closing this issue, feel free to re-open this issue if that was not the case and also feel free to open other issues with your questions about the sortinganalyzer.

closed 2024-09-09 by h-mayorquin

nadine-saluti, 2024-11-18:
Hi again,

I want to re-open this issue because since I last posted I got distracted with other experiments, and didn’t update spike-interface or try to work with the sorting analyzer. 
But I have now updated it to the latest version and would like your advice.
I have two recordings that I already sorted and curated with phy (using the spike-interface pipeline), and now I want to re-extract the mean waveforms of the clusters I labeled as good. 
I load the recording using “read_intan” and then “read_phy” to load the good clusters.
Next, I created the sorting analyzer using this command:
analyzer = si.create_sorting_analyzer(sorting=phy_sorting, recording=rec_filtered, format='binary_folder', folder=sorting_analyzer_folder)
And then I tried to compute the average waveforms using this line you suggested:
“analyzer.compute(['random_spikes', 'waveforms', 'templates'])”
But I don’t know where (or if) it saved the average waveforms.

Thanks for your help

zm711, 2024-11-18:
The `templates` default is to be the average waveforms. So the data is stored in the `templates` extension. What did you want to do with the average waveforms?

nadine-saluti, 2024-11-18:
Great, I found it! I want to load them in Matlab and do some analysis on them, and I am able to load them.

Thanks a lot!

nadine-saluti, 2025-03-31:
I tried it on a different recording that doesn't have gaps in the timestamps file, and the templates.npy looks exactly like the old templates_average.npy file. 

I assume the issue is with the fact that my .rhd file has gaps.

Any ideas what I can do to compute the average waveforms?

zm711, 2025-03-31:
@nadine-saluti, this is the problem with having gaps in the recording. We can't guarantee that the data is good after the discontinuity. My honest answer is that if the jump in timestamps occurred near the beginning this is probably a dataset worth tossing. If it occurred later in the recording, my rec would be to do a `recording.time_slice()` and cut the recording before the jump and then only analyze the data before the jump (you can save a new binary of the good data with a 

```python
recording = se.read_intan(xx)
recording_sliced = recording.time_slice(0, 100xx)
bin_recording = recording_sliced.save(xx)
```

This will get rid of the potentially corrupted data.  We have plans to add a feature in the future to let you only choose spikes to analyze from a subset of your recording (ie you could do the time slice idea inside of the analyzer to get templates only from the good data), but that isn't done yet...

zm711, 2025-03-31:
I don't think we can guarantee that. Once a packet is corrupted it could be possible that all packets are conveying junk.. I'm not sure if we know for sure that x number of perfect packets are lost such that we just ignore the gaps. But I guess if @nadine-saluti is willing to share the corrupted data we could explore more. You also have a corrupted rhd/rhs file no? Could we explore that one as well. I'm thinking that if the packets become too distorted the memmap shape itself will be wrong, but we would need examples of data to know.

zm711, 2025-03-31:
There are two potential issues I could see in that case. 1) the code has just drifted from that point and there's not much we can do (hopefully this isn't the case since we haven't changed that much) 2) both the waveform extractor and the analyzer randomly sample spike times for later analysis. It is possible that for purely sampling reasons you previously had "good" spikes and this time you've sampled from some of the "bad" spikes due to the missing timestamps. Maybe @h-mayorquin has other ideas for why things would change. 

If you plot the raw data with:
```sw.plot_traces(time_range(put the time after the missing timestamps here, return_scaled=True)``` does the data still look reasonable or distorted?

nadine-saluti, 2025-03-31:
I will try to plot it tomorrow and update you. 
Thanks.

nadine-saluti, 2025-04-01:
I plotted the traces as you suggested before and after the gap (I filtered the data and plotted a few channels only).

First, this is the error I get when I try to read the .rhd file without setting "ignore_integrity_checks=True):

[image]

And these are the traces of the recording before and after the gap:

[image]

Any ideas?

Thanks :)

nadine-saluti, 2025-04-12:
Hi, are there any updates?

alejoe91, 2025-04-13:
The plots look ok, right? The error is expected since you have to explicitly tell the reader to ignore timestamps integrity.

nadine-saluti, 2025-04-14:
The problem is that when I generate mean waveforms from this file (I set ignore_integrity_checks=True), I get nonsensical waveforms, probably because of this tiny gap, because it works perfectly with the rest of my recordings.

h-mayorquin, 2025-04-16:
You can slice the recording before the gap and after the gap (`recording.time_slice`) and do the waveform means separated. If the gap is the only problem and your recording is long enough that there are enough samples in both sides that should do it, right? Basically, I think you want to get rid of the samples that fall within the gap. Increasing the number of samples should get rid of this phenomena...