SpikeInterface/spikeinterface #3377: about events

bxy666666, 2024-09-06:
hello, I had some events happened at some timepoint, I want to see if these events have effect. How can I do?

alejoe91, 2024-09-06:
Can you be a bit more specific? :)

bxy666666, 2024-09-06:
> Can you be a bit more specific? :)

OK ，for example, I gave stimulation at 0.1s, 0,2s, 0.3s.....，and  my record is about the whole 10minutes. how can I campare the recation between with stimulation and without stimulation in the whole period.

alejoe91, 2024-09-06:
What would you like to compare?

bxy666666, 2024-09-06:
> What would you like to compare?

I want to compare the units in files with these events and that without any events

bxy666666, 2024-09-06:
> > What would you like to compare?
> 
> I want to compare the units in files with these events and that without any events

like waveforms or other metrics.

bxy666666, 2024-09-06:
> > > What would you like to compare?
> > 
> > 
> > I want to compare the units in files with these events and that without any events
> 
> like waveforms or other metrics.

Also I want to delete some recording slices , I think remove_artifactsis not  suitable. I want to delete, no to  replace with 0.

JoeZiminski, 2024-09-09:
Hi @bxy666666, you can slice up your recordings with [`frame_slice`](https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.core.BaseRecording.frame_slice) or [`time_slice`](https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.core.BaseRecording.time_slice). If you want to put them back together (e.g. with parts dropped) you can use the [concatenation machinery](https://spikeinterface.readthedocs.io/en/0.100.1/modules_gallery/core/plot_5_append_concatenate_segments.html).

You can use `get_unit_spike_train()` to get an array of all spikes for a given unit (indices on which they fire). You can then slice this up based on the indices of your events using numpy slicing to generate things like peri-stimulus time histogram. As a first step I would first suggesting looking at the different units, and different events to see whether you are neurons selectively responding to events. Typically I would expect the actual waveforms of the unit to be stable whether responding to an event vs. responding when there is no stimulus presented (unless these events are some kind of drug application). But, of course anything's possible!

zm711, 2024-12-03:
I'll close this since slicing and concatenating explain the SI machinery for this type of thing. Let us know if you have other questions.

closed 2024-12-03 by zm711