SpikeInterface/spikeinterface #3382: Error involving 'Digital filter critical frequencies must be 0 < Wn < fs/2 (fs=5.0 -> fs/2=2.5)' with kilosort3 docker container

Auxcy1ene, 2024-09-07:
I have been trying to use the docker image of kilosort3 to produce a sorter output that I can analyse for my MSci project but I keep getting the same result. Whether or not I use kilosort gui or the docker image from spike interface for kilosort3, I always end up with the same error message:

""INFO: The correct number of channels has been estimated to be 32.
INFO: Other possibilities could be [40 48 50 60]
Traceback (most recent call last):

File "C:\Users\sean_\anaconda3\envs\kilosort_gui\lib\site-packages\kilosort\gui\settings_box.py", line 326, in import_settings

with open(settings_file_name, 'r') as f:

FileNotFoundError
:
[Errno 2] No such file or directory: ''

Digital filter critical frequencies must be 0 < Wn < fs/2 (fs=5.0 -> fs/2=2.5)""

I am unsure what to do since I have not only tried to input my own raw/binary data but I have downloaded other example data from the internet and have seen the same results. Upon searching for similar errors I have not been able to find anyone in a similar situation so I hope someone here can help me fix this.
Many Thanks
From Sean.H

JoeZiminski, 2024-09-07:
Hi Sean, could you post the code you are using to run the sorting? I think from the trace that the file `settings_file_name` cannot be found. This is then possibly not erroring out immediately for some reason but causing a secondary error because the argument passed to the filter are in the correct format. Happy to help here but you might have more luck on the kilosort github as this seems like a problem finding the file on the kilosort side. Nonetheless please post the code, the problem might be clear from that.

alejoe91, 2024-09-08:
What's the recording sampling frequency?

Auxcy1ene, 2024-09-11:
Hi there, this is no longer an issue anymore and I have an entirely new problem so I'll make a new issue.
Thanks for your time.

closed 2024-09-11 by Auxcy1ene