SpikeInterface/spikeinterface #3385: about bug

bxy666666, 2024-09-07:
This is my code. Imet a serious problem.  I hope your help. When dealing different files,  the number of units are the same. That's not right.

import os
from pprint import pprint
import time
import spikeinterface
import spikeinterface as si  # import core only
import spikeinterface.extractors as se
import spikeinterface.preprocessing as spre
import spikeinterface.sorters as ss
import spikeinterface.postprocessing as spost
import spikeinterface.qualitymetrics as sqm
import spikeinterface.comparison as sc
import spikeinterface.core as score
import spikeinterface.exporters as sexp
import spikeinterface.curation as scur
import spikeinterface.widgets as sw
import spikeinterface.full as si
import matplotlib.pyplot as plt
import numpy as np
import os
import pandas as pd
from spikeinterface.widgets.isi_distribution import ISIDistributionWidget
from spikeinterface.sortingcomponents.peak_detection import detect_peaks
from spikeinterface.core.sortinganalyzer import create_sorting_analyzer
from spikeinterface.exporters import export_to_phy
from datetime import datetime, timedelta
global_job_kwargs = dict(n_jobs=1, chunk_duration="1s")
si.set_global_job_kwargs(**global_job_kwargs)

def process_recording(recording, data_location, suffix):
    # Preprocessing
    recording_f = si.bandpass_filter(recording, freq_min=300, freq_max=6000)
    recording_cmr = si.common_reference(recording_f, reference='global', operator='median')
    recording_preprocessed = recording_cmr.save(format='binary')
    kilosort2_params = ss.get_default_sorter_params('kilosort4')
    print("Updated Kilosort4 params:", kilosort2_params)
    # Run Kilosort4
    sorting_KS2 = ss.run_sorter(sorter_name="kilosort2", recording=recording_preprocessed,
                                extra_requirements=["numpy==1.26"], docker_image=True, verbose=True)

    # Extract waveforms
    we_KS2 = si.extract_waveforms(recording_preprocessed, sorting_KS2,
                                  os.path.join(data_location, f'waveforms_folder_{suffix}'), overwrite=None)

    # Save waveforms as .npy files
    npy_waveform_folder = os.path.join(data_location, f'waveform_npy_{suffix}')
    os.makedirs(npy_waveform_folder, exist_ok=True)
    for unit_id in we_KS2.unit_ids:
        waveforms = we_KS2.get_waveforms(unit_id)
        npy_filename = f"unit_{unit_id}_waveforms.npy"
        npy_filepath = os.path.join(npy_waveform_folder, npy_filename)
        np.save(npy_filepath, waveforms)
        print(f"Unit {unit_id} waveforms saved to {npy_filepath}")

    # Compute metrics
    amplitudes = spost.compute_spike_amplitudes(we_KS2)
    unit_locations = spost.compute_unit_locations(we_KS2)
    spike_locations = spost.compute_spike_locations(we_KS2)
    correlograms, bins = spost.compute_correlograms(we_KS2)
    similarity = spost.compute_template_similarity(we_KS2)
    ISI = spost.compute_isi_histograms(we_KS2, window_ms=100.0, bin_ms=2.0, method="auto")
    metric = spost.compute_template_metrics(we_KS2, include_multi_channel_metrics=True)
    metric_names = spost.get_template_metric_names()

    print(we_KS2.get_available_extension_names())
    waveform_folder = os.path.join(data_location, f'waveforms_folder_{suffix}')
    if not os.path.isdir(waveform_folder):
        print(f"Waveform folder does not exist: {waveform_folder}")
        return
[... 118 lines cut ...]
 
    cleaned_recording = process_with_artifact_removal(recording, stim_times, recording_start_time)

    cleaned_recording_first_half, cleaned_recording_second_half = split_recording_at_midpoint(cleaned_recording,
                                                                                              recording_start_time)
    process_recording(cleaned_recording_first_half, data_location, 'KS2-first_half')
    process_recording(cleaned_recording_second_half, data_location, 'KS2-second_half')

# List of file sets with paths and start times
file_sets = [
    {
        "data_location": ,
        "csv_file": ,
        "recording_start_time": 
    },

]

# Process each file set
for file_set in file_sets:
    process_files(
        data_location=file_set["data_location"],
        csv_file=file_set["csv_file"],
        recording_start_time=file_set["recording_start_time"]
    )

alejoe91, 2024-09-07:
Hi, are you sure you're processing different files? 

Can you `print(file_set["data_location"])`?

bxy666666, 2024-09-08:
> Hi, are you sure you're processing different files?
> 
> Can you `print(file_set["data_location"])`?

cleaned_recording_first_half and cleaned_recording_second_half are the different part of my files, so it is different recordings right?  I also have file_sets = [

    {
        "data_location": "D:\\sjwlab\\bxy\\0828shang\\24395\\1card-3interval-near\\game\\",
        "csv_file": r"D:\sjwlab\bxy\0828shang\24395\1card-3interval-near\game\stim_region_times_20240828_105130.csv",
        "recording_start_time": datetime.strptime("2024/8/28 10:51:32", "%Y/%m/%d %H:%M:%S")
    },
    {
        "data_location": "D:\\sjwlab\\bxy\\game\\24395\\01\\01-1\\",
        "csv_file": r"D:\sjwlab\bxy\game\24395\01\01-1\stim_region_times_20240830_112443.csv",
        "recording_start_time": datetime.strptime("2024/8/30 11:24:46", "%Y/%m/%d %H:%M:%S")
    },

    {
        "data_location": "D:\\sjwlab\\bxy\\game\\24395\\01\\4.4-1\\",
        "csv_file": r"D:\sjwlab\bxy\game\24395\01\01-2\stim_region_times_20240830_113653.csv",
        "recording_start_time": datetime.strptime("2024/8/30 11:36:56", "%Y/%m/%d %H:%M:%S")
    },

]

alejoe91, 2024-09-08:
They are 2 halves of the same recording! The neurons recorded will be the same so the sorter might find the same number of units.

bxy666666, 2024-09-08:
> They are 2 halves of the same recording! The neurons recorded will be the same so the sorter might find the same number of units.

however, my {
    "data_location": "D:\\sjwlab\\bxy\\0828shang\\24395\\1card-3interval-near\\game\\",
    "csv_file": r"D:\sjwlab\bxy\0828shang\24395\1card-3interval-near\game\stim_region_times_20240828_105130.csv",
    "recording_start_time": datetime.strptime("2024/8/28 10:51:32", "%Y/%m/%d %H:%M:%S")
},
{
    "data_location": "D:\\sjwlab\\bxy\\game\\24395\\01\\01-1\\",
    "csv_file": r"D:\sjwlab\bxy\game\24395\01\01-1\stim_region_times_20240830_112443.csv",
    "recording_start_time": datetime.strptime("2024/8/30 11:24:46", "%Y/%m/%d %H:%M:%S")
},

{
    "data_location": "D:\\sjwlab\\bxy\\game\\24395\\01\\4.4-1\\",
    "csv_file": r"D:\sjwlab\bxy\game\24395\01\01-2\stim_region_times_20240830_113653.csv",
    "recording_start_time": datetime.strptime("2024/8/30 11:36:56", "%Y/%m/%d %H:%M:%S")
},   also had the same units.

alejoe91, 2024-09-08:
Did you get the same spike train?

zm711, 2025-02-21:
I'll close for being stale. But feel free to reach back out if you have additional questions.

closed 2025-02-21 by zm711