SpikeInterface/spikeinterface #3387: redirecting output for logging of external sorter output in run_sorter

Djoels, 2024-09-09:
I'm looking to capture the output of the run_sorter function, in which each sorter is printing to stdout.

An example of this:
`
INFO:optuna.study.study:Trial 36 finished with value: 0.08451191145591755 and parameters: {'Th_universal': 7.658515340357731, 'Th_learned': 7.587632722927706, 'do_CAR': True, 'n_pcs': 9, 'Th_single_ch': 6.359439352001865, 'acg_threshold': 0.17898986877557815, 'ccg_threshold': 0.2146340760145186, 'do_correction': True}. Best is trial 2 with value: 0.271796815489171.
2024-09-09 12:13 - INFO - 
╔═════════════════════════════════════════════════════════════════════════════╗
║     running kilosort4 with output dir trial_20240909_kilosort4_093a4f46     ║
╚═════════════════════════════════════════════════════════════════════════════╝
INFO:kilosort.io:========================================
INFO:kilosort.io:Loading recording with SpikeInterface...
INFO:kilosort.io:number of samples: 18040000
INFO:kilosort.io:number of channels: 1023
INFO:kilosort.io:numbef of segments: 1
INFO:kilosort.io:sampling rate: 30000.0
INFO:kilosort.io:dtype: int16
INFO:kilosort.io:========================================
Skipping kilosort preprocessing.
INFO:kilosort.run_kilosort: 
INFO:kilosort.run_kilosort:Computing drift correction.
INFO:kilosort.run_kilosort:----------------------------------------
INFO:kilosort.spikedetect:Re-computing universal templates from data.
100%|█████████████████████████████████████████████████████████████████████████████| 301/301 [13:07<00:00,  2.61s/it]
INFO:kilosort.run_kilosort:drift computed in  824.21s; total  824.22s
INFO:kilosort.run_kilosort: 
INFO:kilosort.run_kilosort:Extracting spikes using templates
INFO:kilosort.run_kilosort:----------------------------------------
INFO:kilosort.spikedetect:Re-computing universal templates from data.
 38%|█████████████████████████████▏                                               | 114/301 [01:36<02:38,  1.18it/s]
`

Everything below the large square (starting with INFO:kilosort.io) is kilosort4 output, which I currently don't seem to be able to capture using Python code.

Is it known to anyone how you could either redirect the logs towards your own log handlers or where you could find the logs in case something went wrong?

JoeZiminski, 2024-09-12:
Hi @Djoels, can you post the code you are using to capture the logging output?

I think kilosort has it's own logger [e.g. here](https://github.com/search?q=repo%3AMouseLand%2FKilosort+logging&type=code). I forget the intricacies of logging, but I think it should be possible to instantiate or grab their logger and set another handler on it to output where you want it to go. Kilosort is using `logging.get_logger(__name__)` across different modules, I think each logger will have the name of the file that it is run in, worth double checking through by breaking into the kilosort code. Let me know if you have any questions about that. Unfortunately logging in python is always kind of confusing and finickity.

Djoels, 2024-09-12:
The current code I'm using has following fragments:

`logger = logging.getLogger(__name__)
    logger.setLevel(logging.INFO)
    console_handler = logging.StreamHandler()
    log_path = os.path.join(optuna_study_root_path, f"optuna_{dataset}_{sorter}_{date.today().isoformat()}.log")
    print(f"{log_path=}")
    file_handler = logging.FileHandler(log_path, mode="a", encoding="utf-8")
    formatter = logging.Formatter(
        "{asctime} - {levelname} - {message}",
        style="{",
        datefmt="%Y-%m-%d %H:%M",
    )
    console_handler.setFormatter(formatter)
    file_handler.setFormatter(formatter)
    logger.addHandler(console_handler)
    logger.addHandler(file_handler)`

and then the actual running of the sorter happens as such:

`
recording = se.BinaryRecordingExtractor(
    file_paths = "temp_wh.dat", 
    sampling_frequency=30000,
    num_channels=1023,
    dtype=int,
    time_axis='C',
)
logger.debug(recording)
# load probe file
probegroup = probeinterface.read_prb("probe.prb")
recording.set_probegroup(probegroup, in_place=True)
ground_truth = se.SHYBRIDSortingExtractor(gt_path, sampling_frequency=recording.get_sampling_frequency())

sorting = ss.run_sorter(sorter_name="kilosort4, 
        recording=recording,
        folder=output_folder,
        docker_image=False,
        verbose=True,
        raise_error=True,
        **params
    )`

as mentioned, the kilosort 4 logging is output to stdout (console) but it doesn't end up in the logfile that I set above

closed 2024-09-12 by Djoels